moveit/moveit #2512: gazebo.launch does not use xacro to load robot_description or respects 'load_robot_description' arg

simonbogh, 2021-02-09:
### Description
When running `demo_gazebo.launch` for an MSA generated moveit_config package, `gazebo.launch` is called. Here I experience two issues:
1. The robot model XACRO is not loaded using `xacro`but `textfile` hence a proper URDF is not generated on the parameter server: https://github.com/ros-planning/moveit/blob/6512d987b8379519146193267906ee957586c4e8/moveit_setup_assistant/templates/moveit_config_pkg_template/launch/gazebo.launch#L15
2. Whether to load the `robot_description` or not is not respected from the argument in `demo_gazebo.launch`, so even though it is set to false, then `gazebo.launch` will still load `robot_description` to the parameter server (wrongly as seen in point 1.): https://github.com/ros-planning/moveit/blob/6512d987b8379519146193267906ee957586c4e8/moveit_setup_assistant/templates/moveit_config_pkg_template/launch/demo_gazebo.launch#L12

### Your environment

ROS Distro: Melodic (but shouldn't matter)
OS Version: Ubuntu 18.04

### Expected behaviour

- If `<arg name="load_robot_description" default="false"/>`is set to false, then `robot_description` should not be loaded from `gazebo.launch`. 
- If it is set to true then `gazebo.launch` should load the `robot_description` using `xacro`.

MohmadAyman, 2021-09-24:
Thanks for the issue, I plan to fix it this weekend.

rhaschke, 2021-12-28:
Fixed as part of #2932.

closed 2021-12-28 by rhaschke