moveit/moveit #2514: libmoveit_kdl_kinematics_plugin.so: undefined symbol: _ZNK3KDL6Vector4NormEv

aravinho, 2021-02-11:
### Description

Hi!  Until recently, my moveit installation has been working fine.  I recently did a fresh build of ROS, and now notice problems when running the `move_group_python_interface` tutorial.  Particularly, there seems to be an issue with the KDL plugin.  I have tried several times, tried building moveit both from binary and source, and the problem persists.  Details below.  

Note that on another machine, I was able to get this working about a week ago.  The only difference is that on the successful machine, I seem to have an older version of the moveit binary, and I suspect this is causing the issue.
```
# Machine that works.  Very scared to apt upgrade to the 20120206 version now!
ros-noetic-moveit                               1.1.0-1focal.20210112.110506
# Machine that breaks
ros-noetic-moveit                               1.1.0-1focal.20210206.044149
```

Thank you for the help!

### Your environment
* ROS Distro: Noetic
* OS Version: Ubuntu 20.04
* Source or Binary build? Binary
* If binary, which release version? `ros-noetic-moveit 1.1.0-1focal.20210206.044149`
* If source, which branch? N/A

### Steps to reproduce
Tell us how to reproduce this issue. Attempt to provide a working demo, perhaps using Docker.
```
# Install ROS/catkin
sudo apt install ros-noetic-desktop-full
sudo apt install -y python3-rosdep \
	python3-rosinstall \
	python3-rosinstall-generator \
	python3-wstool \
	python3-catkin-lint \
	python3-osrf-pycommon \
	ros-noetic-catkin \
	python3-catkin-tools \
	clang-format-10 \

# Install moveit and dependencies
sudo apt-get install -y ros-$ROS_DISTRO-moveit \
	ros-$ROS_DISTRO-libfranka \
	ros-$ROS_DISTRO-moveit-resources-panda-moveit-config \
	ros-$ROS_DISTRO-rviz-visual-tools \
	ros-$ROS_DISTRO-moveit-visual-tools \
	ros-$ROS_DISTRO-combined-robot-hw

# Create catkin workspace and download `moveit_tutorials`
mkdir ~/ws_moveit/src
cd ~/ws_moveitsrc
git clone https://github.com/ros-planning/moveit_tutorials.git
git clone https://github.com/ros-planning/panda_moveit_config.git
git clone https://github.com/frankaemika/franka_ros

# build catkin ws
source /opt/ros/noetic/devel.bash
cd ~/ws_moveit
catkin build -DPYTHON_EXECUTABLE=/usr/bin/python3.7 -DPYBIND11_PYTHON_VERSION=3.7
source ~/ws_moveit/devel.setup.bash

# roslaunch and run `move_group_python_interface` tutorial

# in one terminal
roslaunch panda_moveit_config demo.launch  # in one terminal

# in another terminal
cd ~/ws_moveit/src/moveit_tutorials/doc/move_group_python_interface/scripts
python move_group_python_interface_tutorial.py

```

### Expected behaviour
I expect the tutorial should complete successfully, as it used to.

### Actual behaviour
Rviz successfully loads from the `roslaunch`.  The tutorial starts successfully, loads the robot model, is able to successfully execute a movement to the joint goal, but when trying to execute a movement to a pose goal, freezes.  The `roslaunch` terminal console reads:
```
/opt/ros/noetic/lib/moveit_ros_move_group/move_group: symbol lookup error: /opt/ros/noetic/lib//libmoveit_kdl_kinematics_plugin.so: undefined symbol: _ZNK3KDL6Vector4NormEv
[move_group-5] process has died [pid 194507, exit code 127, cmd /opt/ros/noetic/lib/moveit_ros_move_group/move_group --debug /joint_states:=/joint_states_desired __name:=move_group __log:=/home/aravind/.ros/log/bdaa8a56-6c08-11eb-8a79-eb941317d985/move_group-5.log].
log file: /home/aravind/.ros/log/bdaa8a56-6c08-11eb-8a79-eb941317d985/move_group-5*.log
```

### Backtrace or Console output

Use [gist.github.com](gist.github.com) to copy-paste the console output or segfault backtrace using gdb.

aravinho, 2021-02-11:
> Are you sure you've completely cleaned all your workspaces and rebuilt from scratch? I was rebasing one of our systems onto master today and had similar KDL/kinematics linker errors.

I've just spun up a fresh Ubuntu 20.04 Google Cloud Platform instance.  Installed ROS noetic totally from scratch, following [the instructions here](https://wiki.ros.org/noetic/Installation/Ubuntu).  Currently in the process of building `moveit` from source, following [the instructions here](https://ros-planning.github.io/moveit_tutorials/doc/getting_started/getting_started.html).

The moveit build is taking a while, and probably will be done in an hour or so.  I'll try it out and let you know whether it works.

aravinho, 2021-02-11:
Even with a fresh ROS install, the error persists.  Here is a detailed list of commands to reproduce:
```
###########################################################################
# Install ROS noetic
# based on https://wiki.ros.org/noetic/Installation/Ubuntu
###########################################################################

sudo sh -c 'echo "deb http://packages.ros.org/ros/ubuntu $(lsb_release -sc) main" > /etc/apt/sources.list.d/ros-latest.list'
sudo apt-key adv --keyserver 'hkp://keyserver.ubuntu.com:80' --recv-key C1CF6E31E6BADE8868B172B4F42ED6FBAB17C654
sudo apt update
sudo apt install ros-noetic-desktop-full
source /opt/ros/noetic/setup.bash

# setup rosdep
sudo apt install -y python3-rosdep python3-rosinstall python3-rosinstall-generator python3-wstool build-essential
sudo rosdep init
rosdep update

###########################################################################
# Install noetic and build from source
# Based on https://ros-planning.github.io/moveit_tutorials/doc/getting_started/getting_started.html
###########################################################################

# update everything for good measure
rosdep update
sudo apt update
sudo apt dist-upgrade

# Install catkin tools
sudo apt install -y ros-noetic-catkin python3-catkin-tools
sudo apt install -y python3-wstool
sudo apt install -y python3-osrf-pycommon  

# Set up catkin workspace and get code
mkdir -p ~/ws_moveit/src
cd ~/ws_moveit/src

wstool init .
wstool merge -t . https://raw.githubusercontent.com/ros-planning/moveit/master/moveit.rosinstall
wstool update -t .

# Install all dependencies
cd ~/ws_moveit/src
rosdep install -y --from-paths . --ignore-src --rosdistro noetic

# Update because it says to
sudo sh -c 'echo "deb http://packages.ros.org/ros-testing/ubuntu $(lsb_release -sc) main" > /etc/apt/sources.list.d/ros-latest.list'
sudo apt update

# Build the workspace
cd ~/ws_moveit
catkin config --extend /opt/ros/${ROS_DISTRO} --cmake-args -DCMAKE_BUILD_TYPE=Release
catkin build

##########################################################################
# run the tutorial
##########################################################################

# In one terminal
source /opt/ros/noetic/setup.bash
cd ~/ws_moveit
source ~/ws_moveit/devel/setup.bash
roslaunch panda_moveit_config demo.launch

# in another terminal, create a python virtualenv and run tutorial
source /opt/ros/noetic/setup.bash
cd ~/ws_moveit
source ~/ws_moveit/devel/setup.bash
rosrun moveit_tutorials move_group_python_interface_tutorial.py

# It runs the first few steps fine
# Then after Press `Enter` to execute a movement using a pose goal ...
# in the roslaunch terminal console, I get the following
/home/aravind/ws_moveit/devel/lib/moveit_ros_move_group/move_group: symbol lookup error: /home/aravind/ws_moveit/devel/lib//libmoveit_kdl_kinematics_plugin.so: undefined symbol: _ZNK3KDL6Vector4NormEv
[move_group-5] process has died [pid 302262, exit code 127, cmd /home/aravind/ws_moveit/devel/lib/moveit_ros_move_group/move_group --debug /joint_states:=/joint_states_desired __name:=move_group __log:=/home/aravind/.ros/log/59b7d598-6c81-11eb-9ca0-cff6b308d599/move_group-5.log].
log file: /home/aravind/.ros/log/59b7d598-6c81-11eb-9ca0-cff6b308d599/move_group-5*.log
```

gleichdick, 2021-02-12:
Hey, could you try to link the KDL library explicitly?
Like this
```diff
diff --git a/moveit_kinematics/kdl_kinematics_plugin/CMakeLists.txt b/moveit_kinematics/kdl_kinematics_plugin/CMakeLists.txt
index cd3ffc089..54fc610f8 100644
--- a/moveit_kinematics/kdl_kinematics_plugin/CMakeLists.txt
+++ b/moveit_kinematics/kdl_kinematics_plugin/CMakeLists.txt
@@ -5,7 +5,7 @@ add_library(${MOVEIT_LIB_NAME}
   src/chainiksolver_vel_mimic_svd.cpp)
 set_target_properties(${MOVEIT_LIB_NAME} PROPERTIES VERSION "${${PROJECT_NAME}_VERSION}")
 
-target_link_libraries(${MOVEIT_LIB_NAME} ${catkin_LIBRARIES})
+target_link_libraries(${MOVEIT_LIB_NAME} ${catkin_LIBRARIES} ${orocos_kdl_LIBRARIES})
 
 install(TARGETS ${MOVEIT_LIB_NAME} 
         LIBRARY DESTINATION ${CATKIN_PACKAGE_LIB_DESTINATION} 
```

aravinho, 2021-02-12:
I didnt' try that @gleichdick suggestion directly, but that suggestion led me to find what I believe to be the root cause of the error.

`moveit` seems to require `liborocos-kdl.so.1.4.0` at runtime.  I had two versions of the `.so` file installed, `1.4.0` and `1.4.1`.  I believe I got `1.4.1` when I built `orocos_kdl` and `python_orocos_kdl` [from source](https://github.com/orocos/orocos_kinematics_dynamics) a little while ago.

So my `/usr/lib` looked like as follows.  Notably, the symlinks pointed to version `1.4.1`.  So at runtime, I would get an undefined symbol error, because it was looking for a symbol that's present in `liborocos-kdl.so.1.4.0` but not `liborocos-kdl.so.1.4.1`.

```
$ ls -l /usr/lib/*orocos*
lrwxrwxrwx 1 root root     20 Apr  9  2020 /usr/lib/liborocos-kdl.so -> liborocos-kdl.so.1.4
lrwxrwxrwx 1 root root     22 Feb 11 22:24 /usr/lib/liborocos-kdl.so.1.4 -> liborocos-kdl.so.1.4.1
-rw-r--r-- 1 root root 780464 Apr  9  2020 /usr/lib/liborocos-kdl.so.1.4.0
-rw-r--r-- 1 root root 849024 Jan 25 16:08 /usr/lib/liborocos-kdl.so.1.4.1
```
I manually modified the symlink to point to `1.4.0` and it works.  My `/usr/lib` now looks like this, and the tutorial works.
```
$ cd /usr/lib
$ sudo rm liborocos-kdl.so.1.4  # remove existing symlink to 1.4.1
$ sudo ln -s liborocos-kdl.so.1.4.0 liborocos-kdl.so.1.4  # create a symlink to 1.4.0
$ ls -l /usr/lib/*orocos*
lrwxrwxrwx 1 root root     20 Apr  9  2020 /usr/lib/liborocos-kdl.so -> liborocos-kdl.so.1.4
lrwxrwxrwx 1 root root     22 Feb 12 18:27 /usr/lib/liborocos-kdl.so.1.4 -> liborocos-kdl.so.1.4.0
-rw-r--r-- 1 root root 780464 Apr  9  2020 /usr/lib/liborocos-kdl.so.1.4.0
-rw-r--r-- 1 root root 849024 Jan 25 16:08 /usr/lib/liborocos-kdl.so.1.4.1
```

This unblocks me for now since I'm not actively using `PyKDL` like I was, but this seems to mean that I can't use both `moveit` (depends on `1.4.0` at runtime) and `python_orocos_kdl` (depends on `1.4.1` at runtime) simultaneously.  (Unless there's some way to do so that I'm not aware of.)

gleichdick, 2021-02-13:
Ah, these infos helped me a lot. There was a [breaking change](https://github.com/orocos/orocos_kinematics_dynamics/commit/a6fbd38c509ed7d00db0110fb0bf69afa879ad24) in KDL: `Norm()` now has a parameter `eps`, that's why the name mangling changed. Unfortunately, they haven't bumped their SOVERSION yet. To solve this in general, remove the precompiled KDL (e.g. apt-get remove) and compile everything which depends on KDL from source (moveit, geometry2, etc.). Otherwise, you could try to increment the minor KDL version number **before compiling KDL** (e.g. to 1.5.0), in  so that the linker can resolve the different versions for the precompiled (moveit) and self-compiled (python_orocos_kdl) binaries.

When I install all dependencies in a Focal docker container, only one version of KDL gets installed. So I guess (haven't tested it, due to lack of time) that your MWE isn't complete, because the source install of KDL is missing. I think that there is no problem with moveit or KDL, but with different KDL versions on your local machine.
```
root@1aae0643bc62:~# ls -l /usr/lib/liborocos-kdl.so*
lrwxrwxrwx 1 root root     20 Apr  9  2020 /usr/lib/liborocos-kdl.so -> liborocos-kdl.so.1.4
lrwxrwxrwx 1 root root     22 Apr  9  2020 /usr/lib/liborocos-kdl.so.1.4 -> liborocos-kdl.so.1.4.0
-rw-r--r-- 1 root root 780464 Apr  9  2020 /usr/lib/liborocos-kdl.so.1.4.0
```

mfkenson, 2021-04-07:
Previosuly I expererienced the same error message 
`libmoveit_kdl_kinematics_plugin.so: undefined symbol: _....................`

And I  confirm that after I manually did a sudo apt-get purge ros-noetic-moveit* and get moveit compiled manually in my workspace, then it works.

I did not dig deep enough to find the root cause though. And there is always only 1 version of KDL.
```
$ ls -l /usr/lib/liborocos-kdl.so*
lrwxrwxrwx 1 root root     20 Apr 10  2020 /usr/lib/liborocos-kdl.so -> liborocos-kdl.so.1.4
lrwxrwxrwx 1 root root     22 Apr 10  2020 /usr/lib/liborocos-kdl.so.1.4 -> liborocos-kdl.so.1.4.0
-rw-r--r-- 1 root root 780464 Apr 10  2020 /usr/lib/liborocos-kdl.so.1.4.0
```

v4hn, 2021-04-07:
@mfkenson the problem here was clearly that KDL did not bump their sonames on breaking ABI in the `Norm` method. If you experience a concrete other bug at the moment, please open a new issue.

closed 2021-08-29 by v4hn