moveit/moveit #2517: Make error message in PositionIKRequest less confusing

felixvd, 2021-02-13:
### Description

As described in [this ROS Answers post](https://answers.ros.org/question/371571/found-empty-jointstate-message/), calling the `compute_ik` service with an empty `robot_state` field displays the error `Found empty JointState message` because the field is parsed in [_robotStateMsgToRobotStateHelper](http://docs.ros.org/en/api/moveit_core/html/conversions_8cpp_source.html#l00375) after passing through [here](https://docs.ros.org/en/kinetic/api/moveit_ros_move_group/html/kinematics__service__capability_8cpp_source.html#l00134). Something like "No hint for IK provided - starting from current robot state" should be displayed before the IK is called, imo.

I believe the error message normally appears when an empty message arrives on `/joint_states`, and that behavior should persist. But I can't find where exactly it happens (presumably in `planning_scene_monitor`/`current_state_monitor`), so I might just keep the error print in `utils` and add the extra information.

This would be a good first-timer issue since it only requires small changes in a few files.

### Steps to reproduce

Call the `compute_ik` service with an empty `robot_state` field in the `PositionIKRequest` message.

FabianSchuetze, 2021-03-04:
I have been toying with this issue and would happily try sending a patch at WMD. If anymore would also want to work on this during WMD, I'd be happy to do it together (I'm working at EU hours)!

v4hn, 2021-03-10:
Sorry @FabianSchuetze , I just saw this issue.

The mentioned error message is an artifact of #499 . Before this pr, robot state messages were *always* interpreted relative to the current robot state. I just filed #2544 to remove the error.

I believe it is perfectly fine behavior to have the service use the current state as seed, if nothing else is provided.
If anything should be changed here, it should probably be the weird long comment in [PositionIKRequest](https://github.com/ros-planning/moveit_msgs/blob/master/msg/PositionIKRequest.msg) that asks users to always fill in the field.

closed 2021-03-10 by v4hn

FabianSchuetze, 2021-03-10:
Thanks @v4hn  for your detailed an kind comment  - it provided very valuable background info for me!

Do we want to modify the comment for the PositionIKRequest message? We could say something such as: "If the robot state is set explicitly, the IKService will use this state as basis for the IK calculation. If not, the service uses the current state.". I think this could help users but is of course just a very minor thing.

v4hn, 2021-03-10:
> Do we want to modify the comment for the PositionIKRequest message?

Feel free to propose a pull-request. I believe the current comment is quite confusing and wrong.

PR #2544 (2021-03-10): Fix missing isEmpty check in compute_ik service
Everything works as expected here, if you specify an empty RobotState it is a common pattern in MoveIt to use the current state instead.

The error message is a leftover artifact from #499.

Closes #2517.