moveit/moveit #2522: Disabling and hiding parts of an robot ... implementing tool changers

simonschmeisser, 2021-02-17:
We have projects and requests coming up from time to time that plan to use multiple different end-effectors via a tool changer, let's say a suction cup and a parallel gripper. Given that `RobotModel` is intended to be static there is not really a way to notify anyone about changes in it(?).

One hacky way to implement this is to add all possible end-effectors to the URDF and connect them via linear joints to the actual robot. The detached ones could then be sent on a "planetary orbit" and would be both out of sight and out of collisions. So it could work ... but it's a hack of course

A slightly more proper approach for which I would like to gauge interest and opinions here is to add all options again but to implement a way to flag hidden/disabled links. This would need to be added to both collision checking and visualization (and planning scene msg).

It could also be a property of `JointModel` to be disabled (stuff behind this joint would then "disappear") 

@felixvd @rhaschke and others, looking forward to your opinions

(Or would #2037 and follow ups solve this, I'm missing out on the grand plan there)

gavanderhoorn, 2021-02-17:
#202?

simonschmeisser, 2021-02-17:
#202 would solve this as well, yes, but my "proposal" aims for something implementable in one or two weeks

AndyZe, 2021-02-17:
The general idea sounds nice to me but it seems like the type of large, breaking change that should go in MoveIt2. If we target MoveIt2, then maybe it makes sense to do the extra work to implement a scene graph.

felixvd, 2021-02-17:
Creative workaround. I might do that if you really need to be done with it within 1-2 weeks. Actually, instead of making it 

Exchanging end effectors is a necessity. I see a few ways to implement it:

1) A scene graph as discussed in #202
2) Forcing a reload of the `robot_description`
3) Enabling/disabling links/joints as you described

I haven't looked into the lower level classes enough to know if a `disable` field in the `JointModel` class would be simpler to realize than changing the `robot_description` and reloading the model. One concern I have is that disabling parts of the robot is somewhat hacky, too. Physical bodies generally don't disappear, and it might invite user errors (although that's not a strong argument).

If the `robot_description` could be reloaded, I imagine the model solution to "changing end effectors on an industrial robot arm" would be to define the possible configurations in different URDF/xacro files and to load another one when needed. Is that more or less intuitive than defining each end effector simultaneously on the robot and in its storage? I don't know, but reloading seems more sensible. I think I would encourage working in that direction, but like I said I am not familiar with the code.

For what it's worth, the "grand plan" for #2037 is to follow up with a scene graph, but it does not have to affect the kinematic model right away. #2037 also only affects CollisionObjects. Attaching one would only help if your end effector is not actuated, so that would exclude parallel grippers.