moveit/moveit #2523: Detaching object from the robot causes wrong transformation for the sub-frames

JafarAbdi, 2021-02-17:
### Description

I'm having a bug where if I do the following it causes the subframe transformation to be incorrect

1- Add an object with a subframe to the planning scene 
2- Move the robot
3- Detach the object

https://user-images.githubusercontent.com/16278108/108221320-b6007780-7148-11eb-90e3-ad4de0490fd4.mp4

I don't see the issue when using this [PR](https://github.com/ros-planning/moveit/pull/2037)

https://user-images.githubusercontent.com/16278108/108221389-cadd0b00-7148-11eb-90b6-147d8cab2413.mp4

### Your environment
* ROS Distro: [Melodic]
* OS Version: e.g. Ubuntu 18.04
* Source or Binary build? Source
* If source, which branch? master

### Steps to reproduce
1- `roslaunch moveit_resources_panda_moveit_config demo.launch`
2- Add an object to the planning scene with a subframe [example](https://gist.github.com/JafarAbdi/1fd133b149c326c3620b034a66043222)

gautz, 2021-09-15:
Is this solved by #2037 ?

JafarAbdi, 2021-09-15:
Yep, that's right

closed 2021-09-15 by JafarAbdi