moveit/moveit #2529: [noetic] ros-noetic-moveit-kinematics depends on python2

mikaelarguedas, 2021-02-26:
### Description

On Ubuntu Focal, the deb ros-noetic-moveit-kinematics depends on python2 due to lxml

### Your environment
* ROS Distro: [Noetic]
* OS Version: Ubuntu 18.04
* Binary build
* If binary, which release version? 1.1.0

### Steps to reproduce
```
docker run -it --rm ros:noetic
# apt update
# apt-cache show ros-noetic-moveit-kinematics | grep python-
Depends: libboost-filesystem1.71.0, libboost-program-options1.71.0, libc6 (>= 2.14), libconsole-bridge0.4, libgcc-s1 (>= 3.0), liborocos-kdl1.4, libstdc++6 (>= 9), liburdfdom-world, ros-noetic-moveit-core, ros-noetic-moveit-ros-planning, libeigen3-dev, liborocos-kdl-dev, liburdfdom-tools, python-lxml, ros-noetic-pluginlib (>= 1.11.2), ros-noetic-roscpp, ros-noetic-tf2, ros-noetic-tf2-kdl
```

### Expected behaviour
Should not list any python2 dependency

### Actual behaviour
Lists python-lxml as a dependency

### Additional information
It looks like the package.xml correctly lists dependency based on the python ROS version as of 1.1.1:
https://github.com/ros-planning/moveit/blob/7e1ccecde9b956c358d9d40630b13635171ec70d/moveit_kinematics/package.xml#L35-L38

But never made it to the main repo due to regressions.
Would it be possible to make a noetic release fixing the dependeny list ?

v4hn, 2021-02-26:
That's how old our last noetic release is?! :sob: 

@tylerjw said yesterday at the maintainer meeting that he is working on the `noetic` release.
Tyler, could you create an issue to track the release?
As I argued yesterday as well, I support switching to autogenerated changelogs if we also have a blog post/ short description in the repository summarizing important changes.
It's unacceptable to keep trivial bugs like this unreleased just because nobody has the time to edit lots of text files and people bikeshed over what the files should look like.

tylerjw, 2021-04-09:
It is happening, I'm working on preparing the blog post for ros discourse now to announce the new features.  https://github.com/ros-planning/moveit/issues/2593

mikaelarguedas, 2021-06-07:
forgot to close this issue, this has been fixed thanks all for the release

closed 2021-06-07 by mikaelarguedas