moveit/moveit #2531: Changing our CHANGELOG formatting requirements to support more frequent releases

tylerjw, 2021-02-26:
Some are wondering why there hasn't been a Noetic release since October of last year when we said we were going to do releases every 6mo.  The short answer is the reason is that I've been busy working on other things when I have time to work on moveit (specifically the mess that is ci and some cmake/ros2 issues in moveit2).  The largest blocker to doing a release is how much time it takes to curate the CHANGELOG files.

@v4hn pointed out here: https://github.com/ros-planning/moveit/issues/2529#issuecomment-786545110 and I agree that this current state is unacceptable.

I will admit that our current approach to CHANGELOG files is aesthetically nice, and I generally really like not breaking with whatever style already exists within a file.  However, the fact that it often takes about a full day of work for me to do text edits on these files seems unreasonable unless many people are actually getting value from reading them.  

First, I'd like to know if anyone reads our changelogs and what parts of our current practice provide them the most value?

Second, I'd like to propose we have a couple of simple rules we require for commit messages and then basically accept the auto-generated changelogs from those commit messages.  If that is unacceptable to someone, maybe they would have the time or know someone else who has the time to continue the current style?  If someone wants to take that up and I'll do the rest of the release that would work too.  The goal here is to enable us to have frequent releases.

I would still prepare a Discourse post highlighting the largest changes for each release as I believe that is the most likely thing for others in the ros community to read to find out what is new in moveit releases.

The proposed rules for commit messages are these (to enable the changelogs to be cleaner when auto-generated):

1. Separate subject from body with a blank line
2. Limit the subject line to 50 characters
3. Capitalize the subject line
4. Do not end the subject line with a period
5. Use the imperative mood in the subject line
6. Wrap the body at 72 characters
7. Use the body to explain what and why vs. how

I copied these rules from here: https://chris.beams.io/posts/git-commit/

If this seems acceptable to others I'll make a PR to the website to explain these changes and we can start enforcing this when we merge PRs.  If we do the squash merge in github we can change the commit message.  If the contributor wants a merge merge then they can rebase their change to have commit messages that follow this.  I feel like this is a small price to pay in order to make releases be much less painful to do and still have reasonably readable changelogs.  Nice sideafect of this is it will improve the readability of our git logs too.

rhaschke, 2021-02-28:
- I know from own experience that formatting CHANGELOG messages takes a lot of time, though not a full day (more something like an hour - of course this depends on the amount of commits in a new release).
- In case of issues, I'm first looking into the CHANGELOGS to search for potential reasons. Hence, I believe they are more important than a coarse discourse blog post.
- Although the proposed rules for commit messages are very reasonable, I think, they will not solve the formatting problem: I think, we already obey those rules - at least most of the time. The extra formatting work involves ordering of commit messages by type (feature, bugfix, maintanence) and usually removing the more detailed body to have a concise overview. Having the links to the PRs allows diving deeper if necessary...

felixvd, 2021-03-01:
> Some are wondering why there hasn't been a Noetic release since October of last year when we said we were going to do releases every 6mo.

Wasn't it weeks? ;)

Anyway, I agree with the rules and I also think we mostly follow them already. They probably won't solve the problem if we want a document with the same utility.

I would be happy with either:

A) having a summary of major changes for each version in the Changelog, and then auto-generated commit messages below
B) imposing a stronger message format (including the tag feature/bugfix/maintenance) when squash-merging

But I would add that I mostly look at the MIGRATING.md when I have issues, so my opinion shouldn't weigh much.

v4hn, 2021-03-01:
> though not a full day (more something like an hour ...).

It definitely took longer than a day before because people (myself included) started bikeshedding the details). :smiley:

> In case of issues, I'm first looking into the CHANGELOGS to search for potential reasons

Personally, I would never trust additional documentation when searching for a bug but simply go to the repository and look at `git log path/to/package` (or `git log --oneline ...` to avoid unnecessary details).
As long as the commit messages look reasonable that works out even better than a well-formated changelog for me.

I also think we mostly adhere to the rules Tyler quoted above, although the only rule I would really enforce from this list is
> Separate subject from body with a blank line

because tools depend on this to extract the subject.

@rhaschke from your post I think you do appreciate if you can check sorted Changelogs instead of `git log`, but also don't want to do the work on a regular bases anymore.
Over the past few years we had multiple "burned out" release maintainers that all complained about this step to sort all the changelogs. From my perspective this is simply not acceptable anymore as it blocks actual code changes and bug fixes to be distributed. If anything, the main bottleneck for releases should be backporting relevant changes to older distributions, but nobody's even mentioning that step.

From my perspective @tylerjw should simply use the autogenerated Changelog entries from now on and get the releases out (after an appropriate freeze period).

I would love to see the discourse posts in the repository as well! Maybe in a new top-level "RELEASES.md" file?

v4hn, 2021-03-01:
> > Some are wondering why there hasn't been a Noetic release since October of last year when we said we were going to do releases every 6mo.
> > 
> Wasn't it weeks? ;)

I knew something felt off! 
https://moveit.ros.org/documentation/contributing/releases/
MoveIt has a 6 week cadence for releases

Any statements on this Tyler? :rocket:

tylerjw, 2021-03-01:
> > > Some are wondering why there hasn't been a Noetic release since October of last year when we said we were going to do releases every 6mo.
> > 
> > 
> > Wasn't it weeks? ;)
> 
> I knew something felt off!

There I fixed it, It was supposed to be 6 weeks.

> > though not a full day (more something like an hour ...).
> 
> It definitely took longer than a day before because people (myself included) started bikeshedding the details).

With the re-sorting of them, fixing spelling errors, adding the tags, and fixing the link to PR formatting (if you do a normal merge the auto-generated changelog doesn't link to the PR and I had to do that normally) it easily took a days worth of time to get this done.  There are a ton of packages in moveit and each time I did this it was a really annoying manual process.  And then there was fixing things found in the review.

To be explicitly clear, what I'm suggesting is that the benefits of doing this is not worth the time cost or the delaying of the release.

As to the benefits, are they really that useful when looking for the reason for some bug wouldn't bisecting using git logs be much more effective?  Or, at least as effective as looking at CHANGELOGS.

If the tag is actually important we could get 80% of the way to the current changelog format with the autogenerated with this change:

1. All PRs are squash merged, and when they are the maintainer changes the commit message to match the format we require.

That would eliminate the manual step of fixing the linking to the PR (squash merges have the link included correctly).  It would also eliminate the step of adding the tag.  Lastly, I propose we give up on re-ordering them.  Chronological order seems more useful to me anyway.  I'll group them in the discourse post according to type.

The discourse post doesn't have to be granular.  I could include all changes in it.

Lastly, there is another solution.  If someone would like to write a new auto-generating changelog script for us to use that would meet our rules that would also fix this.  The problem isn't the current format, just the time it takes to manually create it.

v4hn, 2021-03-02:
You know, if you keep proposing alternatives we will never get a release :smile: 

> All PRs are squash merged

This is still as unacceptable as it was last time we discussed it. We would loose important history especially with bigger patch sets.

> Lastly, there is another solution. If someone would like to write a new auto-generating changelog script for us to use that would meet our rules that would also fix this. The problem isn't the current format, just the time it takes to manually create it.

Then you would still have to ensure extraction works as needed and it will break with the occasional "offending" commit.
My goal here would be to make the step a no-brainer to unblock developer time and not add more complexity to maintain. :-)

With two explicit supports (Tyler and me) and one abstention (? Robert) I guess you are hesitant to just do a release like that.
I would be interested to get a clear vote/opinion at least from some more maintainers, @gavanderhoorn, @jschleicher (I know you already +1'ed a post above), @davetcoleman @jrgnicho to get a statement from all independent parties.

Alternatively we could go into "feature freeze" soon and make the decision at the next maintainer meeting right before a scheduled release date.

gavanderhoorn, 2021-03-02:
While changelogs are nice to have, and especially in active projects can provide a nice summary of activity and changes in a particular release, the tooling we have doesn't scale too well with the many packages in MoveIt.

One approach I've seen used which isn't actually *that* bad would be to still go through the motions (ie: use `catkin_generate_changelogs` etc), but then remove all entries it created and replace it by something like this:

```rst
See `here <https://github.com/ros-planning/moveit/compare/1.1.0...1.1.1>`_ for the commit log on Github.
```

That would get you this for MoveIt release `1.1.1`: [link](https://github.com/ros-planning/moveit/compare/1.1.0...1.1.1).

Not as nice as an actual changelog, but for developers interested in seeing whether something was changed still provides similar/the same information.

And if commit messages were nicely written, should still provide a way for less involved users to find whether they are running into something which got touched between the previous and the current release.

rhaschke, 2021-03-02:
1. I agree with Michael, that squash-merging everything isn't an option at all.
2. Improving the changelog auto-generator to correctly handle merge commits shouldn't be that difficult. That's a good idea.
    However, the changelog generator cannot add semantic tags automatically. Adding them during the commit would be helpful.
3. I think the strongest delay was caused by the changelog review process in the past. 
    We should just drop the review process, but not the actual formatting effort.
4. If we do more frequent releases (aiming for 6 weeks again), the individual release effort isn't that big anymore. 
    It just piles up, if we wait too long. My biggest problem when I was doing releases, was to actually find one or two full days of free time to do all the merging of pending PRs and doing the pre-release tests. The changelogs took some time as well, but not the most.

jschleicher, 2021-03-03:
> changelogs are nice to have

Absolutely. If changelogs are the only blocker, just skip them, please! Migration.md and git history is a reasonable start - that was the consensus in the [maintainer meeting last thursday](https://discourse.ros.org/t/moveit-manipulation-wg-meeting-february-25th/18939/46058) as far as I remember. So I think we don't have to wait for the next meeting?!

My main request is to get a new apt release now with the PRs already merged. Not so much about outstanding PRs (it's usually good practice to open a merge window _after_ releases) or _formatting_ (= nice to have). If anyone is interested in good formatting, just volunteer and do the work, please ;-) I sadly do not have that time, although I'd appreciate good release notes as quality indication for good software.

davetcoleman, 2021-03-04:
I love @gavanderhoorn's proposal... that Github commit log is nice because it has direct hyperlinks to the actual code changes and PR notes. Seems much faster.

v4hn, 2021-03-04:
So basically every package's changelog entry should be replaced by the output of
```
git log --oneline --no-decorate $(git describe --abbrev=0 --tags)...HEAD path/to/package
```
?

@tylerjw could you possibly create a PR for this style of Changelog?
I guess we still want to have a freeze period and consider `reviewed-for-backport` before the release, but it would give anyone a chance to complain before the actual release. :sunglasses:

tylerjw, 2021-03-25:
@v4hn here is the first one: https://github.com/ros-planning/panda_moveit_config/pull/78

The guys at frank are blocked by me not releasing this package and holding it hostage for the giant PR of mine to it.  I'm trying to quickly get out a minor release to unblock them.

felixvd, 2021-04-09:
I am happy to see the release being prepared in #2591. I understand that this changelog is auto-generated as discussed in this thread. It seems to me that the [moveit meta-package changelog](https://github.com/ros-planning/moveit/pull/2591/files#diff-6c5c23d9eb24f74a3d0597249d2503eb3dd347c4c373adc50d038c359efd66a2) was used to list major new features/packages. Does this mean that MIGRATION.md is now the place to do that?

tylerjw, 2021-04-09:
> I am happy to see the release being prepared in #2591. I understand that this changelog is auto-generated as discussed in this thread. It seems to me that the [moveit meta-package [...]

We could put things in that file if they are actual migration issues.  I'm compiling a list of the changes now into a discourse post that'll announce the largest changes.  I'll post it in the issue: https://github.com/ros-planning/moveit/issues/2593 for comment before posting to discourse.