moveit/moveit #2534: best way to perform grasping/picking in moveit

christinaionescu, 2021-03-01:
Hi!

I am currently trying to accomplish grasping with the Franka Emika Panda. Therefore, I oriented myself at the pick&place-tutorial. 

In the simulation/visualization of the plan everything works fine but when I execute on the real robot, the gripper movement is not properly synchronized with the arm movement, meaning: it closes too early, I assume already in the pre_grasp_pose. Is this an issue you already know?

Is there a way to perform grasping in another way than using the pick function? Could I somehow just command a closing of the gripper without defining the other poses as in the pick function? 

Something else I was wondering about: why is the Planning Group in the pick&place tutorial defined as PLANNING_GROUP = "panda_arm" and not "panda_arm_hand"?

Thank you!
Best regards

yurirocha15, 2021-03-04:
I haven't had the chance to use it yet, but I believe the MTC (https://github.com/ros-planning/moveit_task_constructor)  is supposed to replace the pick&place tutorial. Maybe would be worth giving it a try.

YannickRiou, 2021-12-16:
For info, the MTC isn't fully usable as the grasp fails because the gripper can't reach the goal position (when grasping, it tries to be fully closed but can't because of the object it is actually grasping !). Some people found that it is better to use the action client "grasp" of Franka.

v4hn, 2021-12-17:
> For info, the MTC isn't fully usable as the grasp fails because the gripper can't reach the goal position

Please create a proper issue for this in MTC. I know what you are talking about, but it should not result in a problem anywhere in the pipelines. We did not have problems with this on a variety of robots in the past.