moveit/moveit #2538: TOTG assumes zero for initial velocities/accelerations

AndyZe, 2021-03-04:
I'm trying to use TOTG for motion smoothing in a closed-loop control situation. The robot starts with non-zero velocity/acceleration.

Unfortunately it seems like TOTG assumes zero for initial velocity/acceleration. Here's a print of the current joint velocities. You can see that some of them (for the active joints) are nonzero. After that is the output from TOTG. The velocities at the first waypoint are all zero.

```
ros.closed_loop_control: current_state velocities: 
ros.closed_loop_control: 0
ros.closed_loop_control: 0
ros.closed_loop_control: 0
ros.closed_loop_control: 0
ros.closed_loop_control: 0
ros.closed_loop_control: 0
ros.closed_loop_control: 0
ros.closed_loop_control: -0.08
ros.closed_loop_control: 0.0372958
ros.closed_loop_control: -0.0135544
ros.closed_loop_control: -0.0549201
ros.closed_loop_control: 0.0453485
ros.closed_loop_control: -0.00605703
ros.closed_loop_control: 0
ros.closed_loop_control: 0
ros.closed_loop_control: 0
ros.closed_loop_control: joint_trajectory: 
  header: 
    seq: 0
    stamp: 0.000000000
    frame_id: tractor_base
  joint_names[]
    joint_names[0]: seg_0_shoulder_pan_joint
    joint_names[1]: seg_0_shoulder_lift_joint
    joint_names[2]: seg_0_elbow_joint
    joint_names[3]: seg_0_wrist_1_joint
    joint_names[4]: seg_0_wrist_2_joint
    joint_names[5]: seg_0_wrist_3_joint
  points[]
    points[0]: 
        positions[]
        positions[0]: -0.120561
        positions[1]: -3.42849
        positions[2]: 1.37549
        positions[3]: -1.63605
        positions[4]: -0.769146
        positions[5]: 0.26796
      velocities[]
        velocities[0]: -0
        velocities[1]: 0
        velocities[2]: -0
        velocities[3]: -0
        velocities[4]: 0
        velocities[5]: -0
      accelerations[]
        accelerations[0]: 0
        accelerations[1]: 0
        accelerations[2]: 0
        accelerations[3]: 0
        accelerations[4]: 0
        accelerations[5]: 0
      effort[]
      time_from_start: 0.000000000
    points[1]: 
        positions[]
        positions[0]: -0.120961
        positions[1]: -3.4283
        positions[2]: 1.37542
        positions[3]: -1.63632
        positions[4]: -0.768919
        positions[5]: 0.26793
      velocities[]
        velocities[0]: -0.08
        velocities[1]: 0.0372957
        velocities[2]: -0.0135614
        velocities[3]: -0.0549256
        velocities[4]: 0.0453552
        velocities[5]: -0.00605587
      accelerations[]
        accelerations[0]: -8
        accelerations[1]: 3.72957
        accelerations[2]: -1.35614
        accelerations[3]: -5.49256
        accelerations[4]: 4.53552
        accelerations[5]: -0.605587
      effort[]
      time_from_start: 0.010000000
```

avoelz, 2021-03-05:
Hello Andy,

the time-optimal trajectory generation assumes that the robot can follow the path exactly. This is in general only possible if the initial and the final velocity are zero. Otherwise, the initial velocity must be directed tangentially to the path and must not exceed the limits imposed by the curvature of the path. One can modify the TOTG code to allow an initial path velocity > 0, but that does not allow arbitrary initial velocities. See e.g. https://doi.org/10.1109/LRA.2019.2929990 for an alternative approach that is more robust to varying initial states.

AndyZe, 2021-03-05:
Thanks! I will probably read your paper. You don't happen to have a C++ implementation available, do you?   ;)

I was thinking of how to spoof TOTG by adding a fake initial waypoint but, like you said, that approach would be seriously limited.

closed 2021-03-05 by AndyZe

buckleytoby, 2025-04-04:
@AndyZe any update to this? I'm facing the same problem.

AndyZe, 2025-04-04:
You could give this alternative a shot:

https://github.com/moveit/moveit/blob/master/moveit_core/trajectory_processing/include/moveit/trajectory_processing/ruckig_traj_smoothing.h

buckleytoby, 2025-04-04:
Aha. Funny enough I planned on smoothing with Ruckig but didn't realize it works for any arbitrary initial state. Great.