moveit/moveit #2546: [maintainers] Why don't we use mergify bot?

davetcoleman, 2021-03-10:
I just saw @clalancette use this Github bot [Mergify](https://mergify.io/) to backport a tutorial change to an older version of ROS 2 [here](https://github.com/ros2/ros2_documentation/pull/1205). 

This seems so easy and useful for maintaining multiple version of MoveIt. Should we use it??

v4hn, 2021-03-12:
That approach seems worth looking into for semi-automatic backports.

Still, there will be a lot of pull-requests that need minor fixes to apply to older branches, so the `review-for-backports` approach would still be necessary to grab those.

rhaschke, 2021-03-12:
I agree. Mergify will simplify trivial backports, but I'm not sure how it will handle conflicts.

clalancette, 2021-03-18:
> I agree. Mergify will simplify trivial backports, but I'm not sure how it will handle conflicts.

So the way that Mergify works is that it opens a pull request for a backport regardless of whether there is a merge conflict or not.  If the patch applies cleanly, then you just have a new pull request to review and merge.  If there are conflicts, then it applies and commits the patches with the conflict markers (`<<<<<<<`, `>>>>>>>`, etc).  When it opens the PR, in the description it lists which files had conflicts from which patch.

Overall, I feel it is a pretty nice system.  It has a few annoying warts, but it does take a lot of drudgery out of doing backports.