moveit/moveit #2548: Building moveit on Windows

Tobias-Fischer, 2021-03-10:
**Is your feature request related to a problem? Please describe.**
Currently moveit does not build on Windows. The main problem is that many symbols are not being properly exported.

**Describe the solution you'd like**
I would like to be able to build moveit on Windows without having to modify/patch any code.

**Describe alternatives you've considered**
---

**Additional context**
@seanyen has been building moveit for Windows for a while, and has a set of patches that need to be applied before the build can succeed. The latest set of patches can be found [here](https://github.com/ms-iot/moveit/tree/windows/1.1.1).

In the [RoboStack project](https://github.com/RoboStack/ros-noetic) we (mainly @wolfv @traversaro and myself in the case of moveit) also package moveit for ROS on Conda. We mostly apply @seanyen's patches, with a few modifications. The patches can be found [here](https://github.com/RoboStack/ros-noetic/tree/master/patch) (look for ros-noetic-moveit*).

It would be great if we could reduce the patches to a minimum, and instead have most of the Windows build fixes here upstream. Would the upstream maintainers here be interested in merging these patches?

simonschmeisser, 2021-04-14:
Just open a PR and then we can have a detailed look at the issues more easily. You probably also want to add a ci for windows?

Tobias-Fischer, 2021-04-19:
See https://github.com/ros-planning/moveit/pull/2604 for the PR

closed 2022-03-03 by Tobias-Fischer