moveit/moveit #2551: Moveit automatically sets the planar virtual joint limits to be -5.0 and 5.0

lyf44, 2021-03-12:
### Description

It seems that Moveit automatically sets the planar virtual joint's position limit to be from -5 to 5. 

### Your environment
* ROS Distro: Melodic
* OS Version: Ubuntu 18.04
* Binary: 1.0.7

### Steps to reproduce
I am trying to plan the whole-body motion of a Fetch robot. Therefore, I manually added a virtual planar joint between Fetch base_link and the world frame in fetch.srdf
```
<virtual_joint name="joint0" type="planar" parent_frame="world" child_link="base_link" />
```
When I plan for that virtual joint with the goal position that is beyond [-5.0, 5.0], moveit will complain "state out of bound" as shown below.

### Expected behaviour
I expect the virtual joint to have no position limits.

### Actual behaviour
It seems moveit sets the joint limit for that virtual joint to be -5.0 and 5.0 automatically. I have confirmed this by printing active joint bounds during planning.

### Backtrace or Console output
```
 INFO] [1615545015.166870423]: Planning request received for MoveGroup action. Forwarding to planning pipeline.
[ INFO] [1615545015.167365531]: Planner configuration 'whole_body[RRTConnectkConfigDefault]' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1615545015.167490738]: whole_body[RRTConnectkConfigDefault]: Starting planning with 1 states already in datastructure
[ INFO] [1615545015.178120039]: Constraint satisfied:: Joint name: 'joint0/x', actual value: -5.690468, desired value: -5.690497, tolerance_above: 0.000100, tolerance_below: 0.000100
[ INFO] [1615545015.178142380]: Constraint satisfied:: Joint name: 'joint0/y', actual value: 1.060118, desired value: 1.060115, tolerance_above: 0.000100, tolerance_below: 0.000100
[ INFO] [1615545015.178152685]: Constraint satisfied:: Joint name: 'joint0/theta', actual value: 0.000089, desired value: 0.000000, tolerance_above: 0.000100, tolerance_below: 0.000100
[ INFO] [1615545015.178174437]: Constraint satisfied:: Joint name: 'torso_lift_joint', actual value: 0.258856, desired value: 0.258764, tolerance_above: 0.000100, tolerance_below: 0.000100
[ INFO] [1615545015.178203446]: Constraint satisfied:: Joint name: 'shoulder_pan_joint', actual value: -1.451008, desired value: -1.450954, tolerance_above: 0.000100, tolerance_below: 0.000100
[ INFO] [1615545015.178212644]: Constraint satisfied:: Joint name: 'shoulder_lift_joint', actual value: -0.028459, desired value: -0.028413, tolerance_above: 0.000100, tolerance_below: 0.000100
[ INFO] [1615545015.178234342]: Constraint satisfied:: Joint name: 'upperarm_roll_joint', actual value: 1.351271, desired value: 1.351236, tolerance_above: 0.000100, tolerance_below: 0.000100
[ INFO] [1615545015.178246325]: Constraint satisfied:: Joint name: 'elbow_flex_joint', actual value: 0.937888, desired value: 0.937884, tolerance_above: 0.000100, tolerance_below: 0.000100
[ INFO] [1615545015.178259555]: Constraint satisfied:: Joint name: 'forearm_roll_joint', actual value: -0.545811, desired value: -0.545808, tolerance_above: 0.000100, tolerance_below: 0.000100
[ INFO] [1615545015.178273639]: Constraint satisfied:: Joint name: 'wrist_flex_joint', actual value: -0.154727, desired value: -0.154665, tolerance_above: 0.000100, tolerance_below: 0.000100
[ INFO] [1615545015.178285774]: Constraint satisfied:: Joint name: 'wrist_roll_joint', actual value: 2.721544, desired value: 2.721573, tolerance_above: 0.000100, tolerance_below: 0.000100
[ INFO] [1615545015.178332211]: State outside bounds
[ERROR] [1615545015.187693497]: whole_body[RRTConnectkConfigDefault]: Unable to sample any valid states for goal tree
[ INFO] [1615545015.187757678]: whole_body[RRTConnectkConfigDefault]: Created 1 states (1 start + 0 goal)
[ INFO] [1615545015.187810072]: No solution found after 0.020334 seconds
[ INFO] [1615545015.187828466]: Unable to solve the planning problem
```

felixvd, 2021-03-14:
Is the FixWorkspaceBounds adapter setting it to this value because you didn't specify the size of your workspace?

lyf44, 2021-04-12:
Thanks, the issue fixed by changing "default_workspace_bounds" param.

closed 2021-04-12 by lyf44