moveit/moveit #2556: Continues Movement

MBaranPeker, 2021-03-19:
Hi, I  have a question for a application similar to jogging. Robot start at position A to move B while that it calculate waypoints then send it robot when the robot execute the action it waits for the new waypoints to execute. What I want is while robot executing the first action, I want to sent robot to the waypoints between position B and C then robot can add up the waypoints for continues movement. I do wonder is it possible to do that without changing the velocity. 

Thanks,

Position A = (0, 0, 0)  Position B = (0, 0, 10)  Position C = (0, 0, 20) Position D = (10, 0, 20) .

[image]

felixvd, 2021-03-24:
I think this would be better suited for ROS Answers. Updating goals while the robot is moving is not generally supported in MoveIt outside of servoing, but it's not impossible. You can calculate a new trajectory and [replace your current one with it](http://wiki.ros.org/joint_trajectory_controller/UnderstandingTrajectoryReplacement), but you need to ensure that the trajectory you just calculated does not move your robot backwards, which I have done in the past by removing part of the new trajectory's start points. 

It sounds like you are expecting linear motions in cartesian motions, but when you replace the trajectory, your joints will move to the next target in joint space, which generally does not result in a linear motion. If you can't accept that, you might want to reduce speed or look into [cartesian controllers like this](https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers).

MBaranPeker, 2021-03-25:
Hi, @felixvd thank you so much for your detailed answer. I will dive into that and let you know the result.