moveit/moveit #2558: moveit_msgs::MotionPlanRequest has no member named ‘pipeline_id'

Sunny478, 2021-03-22:
### Description
While building MoveIt from source (I have done so multiple times successfully), 'moveit_ros_planning_interface' packages poses an error while building viz. 
`‘moveit_msgs::MotionPlanRequest {aka struct Moveit_msgs::MotionPlanRequest_<std::allocator<void> >}’ has no member named ‘pipeline_id’`
Since I had previously built all the packages, so I narrowed it down to the particular commit as in from which commit does the error starts and here are the results...
The build runs fine till this particular commit - 

[image]
gives : 
[image]

but the right next commit i.e.
[image]
gives : 
[image]

On inspecting this particular commit, it does add the line `request.pipeline_id = planning_pipeline_id_;` and in  http://docs.ros.org/en/jade/api/moveit_msgs/html/msg/MotionPlanRequest.html , there is no reference about `planning_id`. 

### Your environment
* ROS Distro: Melodic
* OS Version: Ubuntu 18.04
* Source build

### Steps to reproduce
Build MoveIt from source.

### Expected behaviour
MoveIt successfully built

### Actual behaviour
Error while building 'moveit_ros_planning_interface' package

### Backtrace or Console output
../src/moveit/moveit_ros/planning_interface/move_group_interface/src/move_group_interface.cpp:1085:13: error: ‘moveit_msgs::MotionPlanRequest {aka struct moveit_msgs::MotionPlanRequest_<std::allocator<void> >}’ has no member named ‘pipeline_id’
     request.pipeline_id = planning_pipeline_id_;
             ^~~~~~~~~~~
make[2]: *** [move_group_interface/CMakeFiles/moveit_move_group_interface.dir/src/move_group_interface.cpp.o] Error 1
make[1]: *** [move_group_interface/CMakeFiles/moveit_move_group_interface.dir/all] Error 2
make: *** [all] Error 2

Failed     << moveit_ros_planning_interface:make                             [ Exited with code 2 ]                                                  
Failed    <<< moveit_ros_planning_interface                                  [ 39.4 seconds ]

felixvd, 2021-03-22:
1. You are looking at outdated API documentation for Jade (Google results tricked you) instead of the [master branch](http://docs.ros.org/en/api/moveit_msgs/html/msg/MotionPlanRequest.html).
2. It looks like you didn't follow the [build instructions](https://moveit.ros.org/install/source/), specifically these steps:

```
wstool merge -t src https://raw.githubusercontent.com/ros-planning/moveit/master/moveit.rosinstall
wstool update -t src
rosdep install -y --from-paths src --ignore-src --rosdistro ${ROS_DISTRO}
```

Sunny478, 2021-03-22:
Thanks, this solved the issue. I thought it won't be necessary to download the source code every time I build moveit.

felixvd, 2021-03-23:
When you build from source, you generally need the full source code, yes. I assume this was meant to be closed.

closed 2021-03-23 by felixvd