moveit/moveit #2559: Migrating CI runs to GitHub Actions

tylerjw, 2021-03-23:
This is an issue to show my progress on migrating moveit/moveit2 to GitHub actions.

Here is a current copy of my `industrial_ci_action.yml` for moveit:

```yaml
# This config uses industrial_ci (https://github.com/ros-industrial/industrial_ci.git).
# For troubleshooting, see readme (https://github.com/ros-industrial/industrial_ci/blob/master/README.rst)

name: BuildAndTest

on: [push, pull_request]

jobs:
  industrial_ci:
    strategy:
      matrix:
        env:
          - {
            DOCKER_IMAGE: 'moveit/moveit:master-ci',
            TARGET_CMAKE_ARGS: "-DCMAKE_BUILD_TYPE=RelWithDebInfo -DCMAKE_C_FLAGS='--coverage' -DCMAKE_CXX_FLAGS='--coverage' --no-warn-unused-cli",
            AFTER_RUN_TARGET_TEST: './.ci.prepare_codecov',
            ADDITIONAL_DEBS: 'lcov',
            CCOV_UPLOAD: true
          }
          - {
            DOCKER_IMAGE: 'moveit/moveit:master-ci-shadow-fixed',
            TARGET_CMAKE_ARGS: "-DCMAKE_BUILD_TYPE=Release",
            CCOV_UPLOAD: false,
            CATKIN_LINT: true
          }
          - {
            DOCKER_IMAGE: 'moveit/moveit:melodic-ci',
            TARGET_CMAKE_ARGS: "-DCMAKE_BUILD_TYPE=Release",
            CCOV_UPLOAD: false,
            BEFORE_TARGET_TEST_EMBED: 'set +u && source moveit_kinematics/test/test_ikfast_plugins.sh && set -u'
          }
    env:
      UPSTREAM_WORKSPACE: moveit.rosinstall
      AFTER_SETUP_UPSTREAM_WORKSPACE: 'vcs pull $BASEDIR/upstream_ws/src'
      CCACHE_DIR: "${{ github.workspace }}/.ccache"
      BASEDIR: ${{ github.workspace }}/.work
      CACHE_PREFIX: "${{ matrix.env.DOCKER_IMAGE }}${{ matrix.env.CCOV_UPLOAD && '-ccov' || '' }}"
    runs-on: ubuntu-latest
    steps:
      # Workaround for https://github.com/actions/virtual-environments/issues/709
      - name: free up disk space
        run: |
          sudo rm -rf "/usr/local/share/boost"
          sudo rm -rf "$AGENT_TOOLSDIRECTORY"
          sudo apt-get clean
          df -h
      - uses: actions/checkout@v2
      - name: cache upstream_ws
        uses: pat-s/always-upload-cache@v2.1.3
        with:
          path: ${{ env.BASEDIR }}/upstream_ws
          key: upstream_ws-${{ env.CACHE_PREFIX }}-${{ hashFiles('moveit.rosinstall') }}-${{ github.run_id }}
          restore-keys: |
            upstream_ws-${{ env.CACHE_PREFIX }}-${{ hashFiles('moveit.rosinstall') }}
      - name: cache ccache
        uses: pat-s/always-upload-cache@v2.1.3
        with:
          path: ${{ env.CCACHE_DIR }}
          key: ccache-${{ env.CACHE_PREFIX }}-${{ github.sha }}-${{ github.run_id }}
          restore-keys: |
            ccache-${{ env.CACHE_PREFIX }}-${{ github.sha }}
            ccache-${{ env.CACHE_PREFIX }}
      - name: industrial_ci
        uses: 'ros-industrial/industrial_ci@master'
        env: ${{ matrix.env }}
      - name: upload test artifacts (on failure)
        uses: actions/upload-artifact@v2
        if: failure()
        with:
          name: test-results
          path: ${{ env.BASEDIR }}/target_ws/**/test_results/**/*.xml
      - name: upload codecov report
        uses: codecov/codecov-action@v1
        if: ${{ matrix.env.CCOV_UPLOAD }}
        with:
          files: ${{ env.BASEDIR }}/coverage.info
```

Here is where I'm testing (on my forks): https://github.com/tylerjw/moveit/actions https://github.com/tylerjw/moveit2/actions

Current performance on moveit is about 22 minutes for the second two runs (shadow-fixed, melodic).  Here is a link to see that: https://github.com/tylerjw/moveit/actions/runs/678379698

### Running out of disk space (on code cov run)

You'll notice I'm trying to delete stuff out of the image we are started into.  That's because the code coverage run is running out of disk space during testing.  The next thing I'm going to explore is doing things to reduce the size of the docker images in hope that'll help.

### clang-tidy

There is an implementation of clang-tidy in industrial_ci, however it doesn't filter packages based on if they have changes like moveit_ci does.  My plan right now is to create a script similar to .ci.prepare_codecov that contains the code for doing that test.  Testing the implementation in industrial_ci moveit2 takes ~2hrs to complete.  I plan on working on this once I solve the disk space issue for the codecov run.

rhaschke, 2021-03-23:
> Running out of disk space (on code cov run)

Did you verified that the working folder and the folders you are cleaning up (/usr/local) are on the same disk? (Use `mount`)
If not, it doesn't help to cleanup. Using `df` you can figure out real disk usage.

> clang-tidy

If you take the effort to migrate our `clang-tidy` optimizations, directly push those changes upstream to `industrial_ci`. They will be happy to merge this, I guess. Meanwhile you could run `industrial_ci` from your fork.
Thanks for all your great work on this, @tylerjw!

tylerjw, 2021-03-23:
Doing this:

```
sudo apt-get -qq purge build-essential libboost* mongodb* ghc* azure* google* libgl* hhvm dotnet* firefox adoptopenjdk* llvm* powershell mysql* python*
```

I was able to free up 10gb.  What I'm worried about is that it is actually in the working directory /mnt that is where docker is or something like that and making / have more space isn't helping.

Here are the before and after df -h runs:

```
Filesystem      Size  Used Avail Use% Mounted on
/dev/root        84G   67G   17G  81% /
devtmpfs        3.4G     0  3.4G   0% /dev
tmpfs           3.4G   12K  3.4G   1% /dev/shm
tmpfs           696M  1.1M  695M   1% /run
tmpfs           5.0M     0  5.0M   0% /run/lock
tmpfs           3.4G     0  3.4G   0% /sys/fs/cgroup
/dev/loop0      139M  139M     0 100% /snap/chromium/1514
/dev/loop2       56M   56M     0 100% /snap/core18/1988
/dev/loop1      139M  139M     0 100% /snap/chromium/1523
/dev/loop3      163M  163M     0 100% /snap/gnome-3-28-1804/145
/dev/loop4       70M   70M     0 100% /snap/lxd/19188
/dev/loop5       71M   71M     0 100% /snap/lxd/19647
/dev/loop6       33M   33M     0 100% /snap/snapd/11107
/dev/sda15      105M  7.8M   97M   8% /boot/efi
/dev/loop7       65M   65M     0 100% /snap/gtk-common-themes/1514
/dev/sdb1        14G  4.1G  9.0G  32% /mnt
/dev/loop9       33M   33M     0 100% /snap/snapd/11402
```

```
Filesystem      Size  Used Avail Use% Mounted on
/dev/root        84G   57G   28G  68% /
devtmpfs        3.4G     0  3.4G   0% /dev
tmpfs           3.4G   12K  3.4G   1% /dev/shm
tmpfs           696M  1.1M  695M   1% /run
tmpfs           5.0M     0  5.0M   0% /run/lock
tmpfs           3.4G     0  3.4G   0% /sys/fs/cgroup
/dev/loop0      139M  139M     0 100% /snap/chromium/1514
/dev/loop2       56M   56M     0 100% /snap/core18/1988
/dev/loop1      139M  139M     0 100% /snap/chromium/1523
/dev/loop3      163M  163M     0 100% /snap/gnome-3-28-1804/145
/dev/loop4       70M   70M     0 100% /snap/lxd/19188
/dev/loop5       71M   71M     0 100% /snap/lxd/19647
/dev/loop6       33M   33M     0 100% /snap/snapd/11107
/dev/sda15      105M  7.8M   97M   8% /boot/efi
/dev/loop7       65M   65M     0 100% /snap/gtk-common-themes/1514
/dev/sdb1        14G  4.1G  9.0G  32% /mnt
/dev/loop9       33M   33M     0 100% /snap/snapd/11402
```

rhaschke, 2021-03-23:
> What I'm worried about is that it is actually in the working directory /mnt that is where docker is or something like that and making / have more space isn't helping.

In this case, my suggestion in https://github.com/tylerjw/moveit/pull/5 to move docker to /mnt will not help. It's already there :cry:
We could try the other way around: move docker to / and .ccache to /mnt?

tylerjw, 2021-03-23:
Update on caching target directories.  This does work technically except for the issue that the object files themselves are massive when you do a coverage build.  For example just the `robot_state.cpp.o` file is 47mb.  This means that even after deleting the coverage files, test results, and src directory the target_ws is 7.2GB for this repo.  There is a limit of 5gb per repo for the cache so that won't work.  I think we can cache the target workspace for each other job type though.

tylerjw, 2021-03-23:
Code coverage seems to now be solved, here is the current working state of the yaml:

```yaml
# This config uses industrial_ci (https://github.com/ros-industrial/industrial_ci.git).
# For troubleshooting, see readme (https://github.com/ros-industrial/industrial_ci/blob/master/README.rst)

name: BuildAndTest

on:
  workflow_dispatch:
  pull_request:
  push:
    branches:
      - master

jobs:
  industrial_ci:
    strategy:
      matrix:
        env:
          - {
            IMAGE: 'master-ci',
            CCOV: true,
            AFTER_RUN_TARGET_TEST: './.ci.prepare_codecov'
           }
          - {
            IMAGE: 'master-ci-shadow-fixed',
            CATKIN_LINT: true
          }
          - {
            IMAGE: 'melodic-ci',
            BEFORE_TARGET_TEST_EMBED: 'set +u && source moveit_kinematics/test/test_ikfast_plugins.sh && set -u'
          }
    env:
      DOCKER_IMAGE: moveit/moveit:${{ matrix.env.IMAGE }}
      UPSTREAM_WORKSPACE: moveit.rosinstall
      CCOV: ${{ matrix.env.CCOV }}
      AFTER_SETUP_UPSTREAM_WORKSPACE: 'vcs pull $BASEDIR/upstream_ws/src'
      TARGET_CMAKE_ARGS: >
        -DCMAKE_BUILD_TYPE=${{ matrix.env.CCOV && 'RelWithDebInfo' || 'Release'}}
        ${{ matrix.env.CCOV && '-DCMAKE_CXX_FLAGS="--coverage" --no-warn-unused-cli' || '' }}
      CCACHE_DIR: "${{ github.workspace }}/.ccache"
      BASEDIR: ${{ github.workspace }}/.work
      CACHE_PREFIX: "${{ matrix.env.IMAGE }}${{ matrix.env.CCOV && '-ccov' || '' }}"

    name: "${{ matrix.env.IMAGE }}${{ matrix.env.CATKIN_LINT && ' + catkin_lint' || ''}}${{ matrix.env.CCOV && ' + ccov' || ''}}${{ contains(matrix.env.BEFORE_TARGET_TEST_EMBED, 'ikfast') && ' + ikfast' || ''}}"
    runs-on: ubuntu-latest
    steps:
      - name: "Free up disk space"
        if: ${{ env.CCOV }}
        run: |
          sudo apt-get -qq purge build-essential ghc*
          sudo apt-get clean
          docker system prune -af
          sudo mkdir /mnt/ccache
          mkdir ${{ env.CCACHE_DIR }}
          sudo mount --bind ${{ env.CCACHE_DIR }} /mnt/ccache
          sudo rm -rf /usr/local/*
          df -h
      - uses: actions/checkout@v2
      - name: cache upstream_ws
        uses: pat-s/always-upload-cache@v2.1.3
        with:
          path: ${{ env.BASEDIR }}/upstream_ws
          key: upstream_ws-${{ env.CACHE_PREFIX }}-${{ hashFiles('moveit2.repos') }}-${{ github.run_id }}
          restore-keys: |
            upstream_ws-${{ env.CACHE_PREFIX }}-${{ hashFiles('moveit2.repos') }}
      - name: cache target_ws
        if: ${{ ! env.CCOV }}
        uses: pat-s/always-upload-cache@v2.1.3
        with:
          path: ${{ env.BASEDIR }}/target_ws
          key: target_ws-${{ env.CACHE_PREFIX }}-${{ hashFiles('**/CMakeLists.txt') }}-${{ hashFiles('**/package.xml') }}-${{ github.run_id }}
          restore-keys: |
            target_ws-${{ env.CACHE_PREFIX }}-${{ hashFiles('**/CMakeLists.txt') }}-${{ hashFiles('**/package.xml') }}
      - name: cache ccache
        uses: pat-s/always-upload-cache@v2.1.3
        with:
          path: ${{ env.CCACHE_DIR }}
          key: ccache-${{ env.CACHE_PREFIX }}-${{ github.sha }}-${{ github.run_id }}
          restore-keys: |
            ccache-${{ env.CACHE_PREFIX }}-${{ github.sha }}
            ccache-${{ env.CACHE_PREFIX }}
      - name: industrial_ci
        uses: 'ros-industrial/industrial_ci@master'
        env: ${{ matrix.env }}
      - name: upload test artifacts (on failure)
        uses: actions/upload-artifact@v2
        if: failure()
        with:
          name: test-results
          path: ${{ env.BASEDIR }}/target_ws/**/test_results/**/*.xml
      - name: upload codecov report
        uses: codecov/codecov-action@v1
        if: ${{ env.CCOV }}
        with:
          files: ${{ env.BASEDIR }}/coverage.info
      - name: prepare target_ws for cache
        if: ${{ always() && ! env.CCOV }}
        run:
          du -sh ${{ env.BASEDIR }}/target_ws
          sudo find ${{ env.BASEDIR }}/target_ws -wholename '*/test_results/*' -delete
          sudo rm -rf ${{ env.BASEDIR }}/target_ws/src
          du -sh ${{ env.BASEDIR }}/target_ws
```

tylerjw, 2021-03-25:
We have migrated master.  Closing this for now, will open PRs to backport the migration to other branches.

closed 2021-03-25 by tylerjw