moveit/moveit #2561: Rviz elements not drawn correctly on Noetic (broken-looking windows)

felixvd, 2021-03-25:
In our docker container, the Rviz window looks broken like this. @simonschmeisser confirmed this:

https://user-images.githubusercontent.com/4535737/112409012-1728f580-8d5c-11eb-8b41-4a9f95f17bd1.mp4

I don't know that this is a MoveIt or even Rviz problem at all, instead of a Qt or driver issue. The same thing happens in a different Ubuntu 20.04 image I use. Anyway, it's easy to reproduce:

`cd` to a moveit repo and execute this:  
`.docker/gui-docker --rm -it moveit/moveit:noetic-source /bin/bash`  
(to enter the container. Then, in the container:)  
`source ../install/setup.bash`  
`roslaunch panda_moveit_config demo.launch`  

### Your environment
* OS Version: Ubuntu 20.04
* ROS Distro: Noetic

simonschmeisser, 2021-03-25:
https://stackoverflow.com/questions/52501594/qt-app-ui-elements-randomly-rendered-as-blank-black-in-docker  proposes some things to try

Also note that this issue is limited to docker, running "natively" I have no rendering issues

gavanderhoorn, 2021-03-25:
> broken-looking windows

the technical term you're looking for I believe is *cheese*.

GUI redraw problems -- especially on resize -- have been referred to like that.

rhaschke, 2021-03-25:
This is neither a MoveIt nor rviz issue. I don't believe it's Qt either. I would guess X11 or the display driver.