moveit/moveit #2570: Quick question about fake_execution flag

ranjitkathiriya, 2021-03-29:
Hello there,

Just a quick question, if we are configuring a real robot with move it what will be the flag of fake_execution flag? it will be true or false?

```
<include file="$(find kuka_moveit_roof)/launch/move_group.launch">
      <arg name="publish_monitored_planning_scene" value="true" />
       <arg name="allow_trajectory_execution" value="true"/>
      <arg name="fake_execution" value="true"/>
    <arg name="info" value="true"/>
    <arg name="debug" value="$(arg debug)"/>
    <arg name="pipeline" value="$(arg pipeline)"/>
    <arg name="load_robot_description" value="$(arg load_robot_description)"/>
  </include>
```

and what args is important when we are configuring a real robot? In repository, I only find 'publish_monitored_planning_scene' can you please put a light on this question?

Overview of your issue here.

### Your environment
* ROS Distro: Noetic
* OS Version: e.g. Ubuntu 20.04

ranjitkathiriya, 2021-03-30:
Finally, Solve this issue:

> if we are configuring a real robot with moveit what will be the flag of `fake_execution` flag? it will be true or false?

Answer: It should be always set to false. You can set it by making false here as below or ignore writing this fake_execution line by default it is set to false. It is very important to set `publish_monitored_planning_scene` as true.

```
<include file="$(find kuka_moveit_roof)/launch/move_group.launch">
      <arg name="publish_monitored_planning_scene" value="true" />
       <arg name="allow_trajectory_execution" value="true"/>
      <arg name="fake_execution" value="true"/>
    <arg name="info" value="true"/>
    <arg name="debug" value="$(arg debug)"/>
    <arg name="pipeline" value="$(arg pipeline)"/>
    <arg name="load_robot_description" value="$(arg load_robot_description)"/>
  </include>
```
> If I am making the fake_execution flag as false I am getting an error as unused args.

Answer: In `trajectory_execution.launch.xml` uncomment below line. The reason of this bug was the `moveit_controller_manager.launch.xml` file has no argument name `execution_type`.

```
<include file="$(find kuka_moveit_roof)/launch/$(arg moveit_controller_manager)_moveit_controller_manager.launch.xml">
<!--    <arg name="execution_type" value="$(arg execution_type)" />-->
  </include>
```

closed 2021-03-30 by ranjitkathiriya

valmil2000, 2021-07-12:
Hey how exactly did you solve this bug?

ranjitkathiriya, 2021-07-12:
I have provided my solution over here. 

https://github.com/ros-planning/moveit/issues/2570#issuecomment-810305129 (Just above your comment)

ranjitkathiriya, 2021-07-12:
Or you can remove all args and keep this: 

```
<include file="$(find kuka_kr20_movit_config)/launch/move_group.launch">
      <arg name="publish_monitored_planning_scene" value="true" />
  </include>
```

Dont forget to do this: Answer: In trajectory_execution.launch.xml uncomment below line. The reason of this bug was the moveit_controller_manager.launch.xml file has no argument name execution_type.

as in above https://github.com/ros-planning/moveit/issues/2570#issuecomment-810305129 comment.

I think it should work.

valmil2000, 2021-07-12:
Thanks for your help. I actually had to comment the line instead of uncommenting it.