moveit/moveit #2574: private `setToIKSolverFrame()` breaks workflow

bsaund, 2021-03-31:
https://github.com/ros-planning/moveit/pull/2461 makes `setToIKSolverFrame()` private. A public `setToIKSolverFrame()` is necessary (afaik) to effectively use other parts of the moveit API.
For example, my code uses the kinematics solver to get multiple IK solutions for a robot arm with redundant DOFs.
```
const kinematics::KinematicsBaseConstPtr &solver = jmg->getSolverInstance();
auto kinematic_state = std::make_shared<robot_state::RobotState>(model_);

kinematic_state->setToIKSolverFrame(solverTrobot, solver);

Eigen::Affine3d pose_in_solver_frame = solverTrobot * ee_target_pose_in_robot_frame;
... (conversion to target_poses) ...
solver->getPositionIK(target_poses, ...)
```
[Link to my code](https://github.com/UM-ARM-Lab/arm_robots/blob/3c6f42e622330cca31da5d722dcb855cc25b2c48/jacobian_follower/src/jacobian_follower/jacobian_follower.cpp#L170)
[Moveit docs](http://docs.ros.org/en/api/moveit_core/html/classkinematics_1_1KinematicsBase.html#ab874f2e103378473f93e1cd1d3e2e347)

It is necessary to transform the target pose(s) in the robot frame into the solver frame. With `setToIKSolverFrame` now private, I have to duplicate the `setToIKSolverFrame()` in my own code.

@rhaschke @AndyZe (were on the PR mentioned above)

Some possible resolutions are:
1) Make `setToIKSolverFrame()` public again
2) Duplicate `setToIKSolverFrame()` logic in my own code
3) Find a different pathway to get multiple IK solutions from a pose in robot (not solver) frame. 

I appreciate any time and comments w.r.t. this issue, thanks!

rhaschke, 2021-03-31:
Sorry. Obviously, we didn't expect this function to be used externally. @AndyZe can you make those functions public again, please?

v4hn, 2021-07-09:
Fixed by #2580

closed 2021-07-09 by v4hn