moveit/moveit #2576: Falling back to using the the move_group node namespace (deprecated behavior).

stevedanomodolor, 2021-04-02:
### Description

I am currently running the abb experimental package, using the irb1200 5 90 robots and I get this warning when I run the demo.launch, how do i fix it
`Falling back to using the the move_group node namespace (deprecated behavior).`

* ROS Distro: Noetic
* OS Version: e.g. Ubuntu 20.04

roslaunch abb_irb1200_5_90_moveit_config demo.launch

closed 2021-04-02 by stevedanomodolor

felixvd, 2021-04-03:
Please add the solution to your problem for future visitors.

stevedanomodolor, 2021-04-03:
The issue was with the moveit package for the abb robot. They were old ros style implementation that I had to change to stop getting this warning. This is the final move_group.launch file. The previous one included a group ns.  

```
<launch>

  <include file="$(find abb_irb1200_5_90_moveit_config)/launch/planning_context.launch" />

  <!-- GDB Debug Option -->
  <arg name="debug" default="false" />
  <arg unless="$(arg debug)" name="launch_prefix" value="" />
  <arg     if="$(arg debug)" name="launch_prefix"
	   value="gdb -x $(find abb_irb1200_5_90_moveit_config)/launch/gdb_settings.gdb --ex run --args" />

  <!-- Verbose Mode Option -->
  <arg name="info" default="$(arg debug)" />
  <arg unless="$(arg info)" name="command_args" value="" />
  <arg     if="$(arg info)" name="command_args" value="--debug" />

  <!-- move_group settings -->
  <arg name="pipeline" default="ompl" />
  <arg name="allow_trajectory_execution" default="true"/>
  <arg name="fake_execution" default="false"/>
  <arg name="execution_type" default="interpolate"/> <!-- set to 'last point' to skip intermediate trajectory in fake execution -->
  <arg name="max_safe_path_cost" default="1"/>
  <arg name="jiggle_fraction" default="0.05" />
  <arg name="publish_monitored_planning_scene" default="true"/>

  <!-- Planning Pipelines -->
  <group ns="move_group/planning_pipelines">

    <!-- OMPL -->
    <include ns="ompl" file="$(find abb_irb1200_5_90_moveit_config)/launch/planning_pipeline.launch.xml">
      <arg name="pipeline" value="$(arg pipeline)" />
    </include>

    <!-- Load non-default planner if passed as parameter -->
    <include if="$(eval arg('pipeline') not in ['ompl', 'chomp', 'pilz_industrial_motion_planner'])" ns="$(arg pipeline)"
         file="$(find abb_irb1200_5_90_moveit_config)/launch/planning_pipeline.launch.xml">
         <arg name="pipeline" value="$(arg pipeline)" />
    </include>
</group>

  <!-- Planning Functionality -->
  <!-- <include ns="move_group" file="$(find abb_irb1200_5_90_moveit_config)/launch/planning_pipeline.launch.xml">
    <arg name="pipeline" value="ompl" />
  </include> -->

  <!-- Trajectory Execution Functionality -->
  <include ns="move_group" file="$(find abb_irb1200_5_90_moveit_config)/launch/trajectory_execution.launch.xml" if="$(arg allow_trajectory_execution)">
    <arg name="moveit_manage_controllers" value="true" />
    <arg name="moveit_controller_manager" value="abb_irb1200_5_90" unless="$(arg fake_execution)"/>
    <arg name="moveit_controller_manager" value="fake" if="$(arg fake_execution)"/>
  </include>

  <!-- Sensors Functionality -->
  <include ns="move_group" file="$(find abb_irb1200_5_90_moveit_config)/launch/sensor_manager.launch.xml" if="$(arg allow_trajectory_execution)">
    <arg name="moveit_sensor_manager" value="abb_irb1200_5_90" />
  </include>

  <!-- Start the actual move_group node/action server -->
  <node name="move_group" launch-prefix="$(arg launch_prefix)" pkg="moveit_ros_move_group" type="move_group" respawn="false" output="screen" args="$(arg command_args)">
    <!-- Set the display variable, in case OpenGL code is used internally -->
    <env name="DISPLAY" value="$(optenv DISPLAY :0)" />

    <param name="allow_trajectory_execution" value="$(arg allow_trajectory_execution)"/>
    <param name="max_safe_path_cost" value="$(arg max_safe_path_cost)"/>
    <param name="jiggle_fraction" value="$(arg jiggle_fraction)" />
    <param name="default_planning_pipeline" value="$(arg pipeline)"/>

    <!-- load these non-default MoveGroup capabilities -->
    <!--
    <param name="capabilities" value="
                  a_package/AwsomeMotionPlanningCapability
                  another_package/GraspPlanningPipeline
                  " />
    -->

    <!-- inhibit these default MoveGroup capabilities -->
    <!--
    <param name="disable_capabilities" value="
                  move_group/MoveGroupKinematicsService
                  move_group/ClearOctomapService
                  " />
    -->

    <!-- Publish the planning scene of the physical robot so that rviz plugin can know actual robot -->
    <param name="planning_scene_monitor/publish_planning_scene" value="$(arg publish_monitored_planning_scene)" />
    <param name="planning_scene_monitor/publish_geometry_updates" value="$(arg publish_monitored_planning_scene)" />
    <param name="planning_scene_monitor/publish_state_updates" value="$(arg publish_monitored_planning_scene)" />
    <param name="planning_scene_monitor/publish_transforms_updates" value="$(arg publish_monitored_planning_scene)" />
  </node>

</launch>
```

felixvd, 2021-04-03:
Oh right. It's a good idea to regenerate the moveit_config package with the setup assistant when moving between distros or major versions.

stevedanomodolor, 2021-04-03:
Ok thank you, i will do that

robertjbush, 2021-05-05:
This was helpful. @felixvd Is there a pre-built reference configuration which is more recent than the panda [repo here](https://github.com/ros-planning/panda_moveit_config/tree/melodic-devel/launch). The above repo is referenced on the MoveIt! tutorials page. I would rather correct my known and reviewed configuration files 'manually' instead of starting afresh using the setup assistant.