moveit/moveit #2577: Collision detection plugins managment in PlanningScene

enricovillagrossi, 2021-04-02:
Hi,
I'm currently looking to create a test a moveit collision detection plugin. The current collision detectors available in moveit (based on FCL and Bullet) are implemented as plugins. If I'm not wrong, the managment of collision detector as plugin is available only in the class PlanningSceneMonitor while it is not available in the class PlanningScene where the default collision detector is FCL and its laoding is done during the initialization through an "hard coded" line with the method setActiveCollisionDetector. If the user wants to change the default collision detector it is anyway neccssary to use the setActiveCollisionDetector method losing the benefits of the plugin.
Is there any reason for this choice? Do I miss something?
Thank you.
Enrico

simonschmeisser, 2021-04-06:
Support for plugins seems unfinished in some places. It is quite possible that those missing functions where not concious design choices but just never needed. Please open a PR with what you need added and then we can see how that fits in.

:+1: for adding more tests

v4hn, 2021-04-06:
You are right, the plugin is only applied automatically if you use a PlanningSceneMonitor.

However, it's straight-forward to instantiate a [`CollisionPluginLoader` and call `setupScene(nh, scene)` on it](https://github.com/ros-planning/moveit/blob/master/moveit_ros/planning/collision_plugin_loader/include/moveit/collision_plugin_loader/collision_plugin_loader.h#L53).
The main source for this two-step mechanism is that core data structures should not rely on ROS parameter interfaces, so their setup in a ROS ecosystem is often split out to some loader class in `moveit_ros`.

I'm not sure what to improve here (aside from better documentation).
You could probably add another constructor to avoid setting up the FCL plugin for no reason when you know what other plugin to use.