moveit/moveit #2578: Planned paths are not displayed in MotionPlanning plugin by default anymore

felixvd, 2021-04-02:
With #2127 merged, the topic `/move_group/display_planned_path` that the Rviz plugin subscribes to does not receive plans to display anymore. The plans now arrive on topics that are namespaced to each planning plugin (e.g. `/move_group/planning_pipelines/ompl/display_planned_path`).

v4hn, 2021-04-06:
> `/move_group/planning_pipelines/ompl/display_planned_path`

Sounds like we should make sure the pipelines publish to `/move_group/display_planned_path` instead.
It does not make much sense to keep these topics separate for different pipelines.

felixvd, 2021-05-02:
For sure. Sadly, the `PlanningPipeline` only has a child nodehandle in its own namespace (`.../move_group/planning_pipelines/ompl/`. Handing in the absolute path wouldn't work if the `move_group` node itself is namespace, so I guess the `PlanningPipeline` needs to prepend `/planning_pipelines/ompl` to its topics and parameters.

v4hn, 2021-05-02:
If we are ok with such an approach, the cheapest solution is probably to keep the interface and do something like
```
std::string ns = nh_.getNamespace();
ns.resize(ns.rfind("planning_pipelines/"));
NodeHandle parent_ns { ns };
```

Passing in only the parent NodeHandle won't work because the pipeline does not seem to know its own namespace on the parameter server.

Also I strictly prefer not to change the semantics of parameters without actually breaking API so that users will notice at compile time.

henningkayser, 2021-05-05:
Can't we just use a private NodeHandle for creating the publisher?

felixvd, 2021-05-06:
> Also I strictly prefer not to change the semantics of parameters without actually breaking API so that users will notice at compile time.

Definitely a concern. We could make another constructor with a different argument order?

> If we are ok with such an approach, the cheapest solution is probably to keep the interface and do something like
> 
> ```
> std::string ns = nh_.getNamespace();
> size_t parent_pos = ns.rfind("planning_pipelines/")
> if(parent_pos != std::string::npos) ns.resize(parent_pos);
> NodeHandle parent_ns(ns);
> ```

Won't this get expensive if every MTC stage creates a PlanningPipeline?

> Passing in only the parent NodeHandle won't work because the pipeline does not seem to know its own namespace on the parameter server.

You mean the child namespace? That's systematically generated though, and the pipeline should know its own name.

> Can't we just use a private NodeHandle for creating the publisher?

Can you explain?

henningkayser, 2021-05-06:
> > Can't we just use a private NodeHandle for creating the publisher?
> 
> Can you explain?

See #2652

PR #2652 (2021-05-10): Use private NodeHandle instead of child for PlanningPipeline topics
Fixes #2578.

#2127 introduced a child NodeHandle for initializing the PlanningPipeline. Since the display topics were generated from that one, the MotionPlanning panel wasn't able to receive any display markers from the "move_group" namespace anymore. By advertising the topics from a private NodeHandle (`"~"`) all markers and display messages are again published from the node's namespace, which in case of MoveGroup is "move_group". -> All fine!
closed 2021-05-10 by v4hn