moveit/moveit #2581: moveitcpp vs movegroup

stevedanomodolor, 2021-04-06:
This is out of curiosity. When is important to use moveitcpp instead of movegroup(advantages and inconvenience). If the goal was a commercial application, would moveitcpp be better and why.

Thank you very much for your response

AndyZe, 2021-04-06:
One advantage is decreased latency.

Some disadvantages are:

-  Can only be used with C++
- Less tutorial documentation.

I'd be interested to hear more pros/cons, too!

v4hn, 2021-04-06:
`movegroup` does not exist, you are probably talking about the `MoveGroupInterface`.
This is a very thin class wrapping ROS actions that provides very easy access to MoveIt. Nothing more.

If you want detailed control over the C++ datastructures that do the actual planning, you need MoveItCpp instead or have to do the whole class setup that MoveItCpp does for you by yourself. That's pretty much the whole story.

As this is no issue at all, I'll close it now (sorry Andy ;)). Feel free to discuss such question at https://answers.ros.org

closed 2021-04-06 by v4hn

AndyZe, 2021-04-06:
Well, you have reminded me of another advantage of moveit_cpp:

You can do things like combine trajectories more easily, or run your own post-processign with `moveit_cpp`. That's not really an option with `MGI`.

stevedanomodolor, 2021-04-06:
Thanks for your response