moveit/moveit #2582: MotionPlanning Display re-enables after jump in time

christian-rauch, 2021-04-06:
### Description

The `MotionPlanning` Display type is re-enabled in RViz when a jump in time is detected. This is distracting when replaying bag files and visualising a recorded robot state and sensor data in RViz.

The only workaround is to remove the display type.

### Your environment
* ROS Distro: Noetic
* OS Version: Ubuntu 20.04
* Binary: rviz version 1.14.5

### Steps to reproduce

1. add `MotionPlanning` Display type to RViz
2. play a bag file in loop (`-l`)
3. disable the `MotionPlanning` by unchecking the checkmark
4. the `MotionPlanning` is re-enabled when the bag file repeats, the checkmark is added again

### Expected behaviour
The `MotionPlanning` Display should stay disabled

### Actual behaviour
`MotionPlanning` is enabled again after each reset of time.

### Backtrace or Console output

Use [gist.github.com](gist.github.com) to copy-paste the console output or segfault backtrace using gdb.

rhaschke, 2021-04-06:
I will have a look at this. Can you tell which kind of topics you recorded in the bag file? Does that matter for the bug?

rhaschke, 2021-04-06:
I can confirm this issue. It is caused by a `Display::reset()` triggered by rviz on a time reset, which then disables and re-enables the frame (and display). However, a resolution seems not trivial: Not re-enabling the frame here:
https://github.com/ros-planning/moveit/blob/d8e77c2d4b78b73268f01039b56f083f4a97d9ed/moveit_ros/visualization/motion_planning_rviz_plugin/src/motion_planning_display.cpp#L303
causes rviz to freeze on shutdown, because there are pending background / mainloop jobs for some reason.
Reenabling the display/frame before shutdown resolves the issue. Not seeing the reset trigger from rviz, doesn't cause the problem as well. I suspect a dangling loadRobotModel()... Need to investigate further.

christian-rauch, 2021-04-07:
@rhaschke Thank you very much for providing a fix for this issue that quick.

PR #2588 (2021-04-07): Fix PlanningScene / MotionPlanning displays
This fixes several issues with PSD/MPD, see individual commit messages for details. Please, keep them during merging.
Fixes #2582.