moveit/moveit #2583: Invalid target measured dimensions

lorepieri8, 2021-04-06:
### Description

I cannot complete the Hand-Eye Calibration tutorial, as rviz freezes when selecting the Image Topic.

### Your environment
* ROS Distro: Noetic
* OS Version: e.g. Ubuntu 20.04
* Binary build

### Backtrace or Console output
`[ERROR] [1617740758.023112738]: Invalid target measured dimensions: marker_size 0,000000, marker_seperation 0,000000
`

lorepieri8, 2021-04-06:
The solution is in https://github.com/ros-visualization/rviz/issues/1249: 

`export LC_NUMERIC="en_US.UTF-8"`

v4hn, 2021-04-07:
> ROS Distro: Noetic
> OS Version: e.g. Ubuntu 20.04

> The solution is in ros-visualization/rviz#1249:
> export LC_NUMERIC="en_US.UTF-8"

The thread there talks about this being a problem in ROS *melodic*. Is this really still a problem in Ubuntu 20.04 / noetic?

gavanderhoorn, 2021-04-07:
Reminds me of https://github.com/ros-planning/moveit_calibration/issues/68.

lorepieri8, 2021-04-07:
> Reminds me of [ros-planning/moveit_calibration#68](https://github.com/ros-planning/moveit_calibration/issues/68).

Yes, it is indeed the exact error.

> The thread there talks about this being a problem in ROS _melodic_. Is this really still a problem in Ubuntu 20.04 / noetic?

I can confirm that I encountered the problem in Ubuntu 20.04 / noetic.

gavanderhoorn, 2021-04-07:
Sounds like the Calibration display/plugin is not doing the right thing.