moveit/moveit #2587: Only joint-space goals are supported

mpatalberta, 2021-04-06:
### Descriptioin
Trying to use the python api with chomp does not work.
RVIS works with chomp however

The following error:
Overview of your issue here.Only joint-space goals are supported

### Your environment
* ROS Distro: [melodic]
* OS Version: e.g. Ubuntu 18.04
* Source or Binary build? Source
* If binary, which release version?
* If source, which branch?master

### Steps to reproduce
Tell us how to reproduce this issue. Attempt to provide a working demo, perhaps using Docker.
Create a valid chomp configuration using rviz//moveit
Check that the RVIZ/moveit works
Using cartiesian poses in python  application try to plan  a pathe using chomp
### Expected behaviour
Tell us what should happen

### Actual behaviour
Tell us what happens instead

### Backtrace or Console output

Use [gist.github.com](gist.github.com) to copy-paste the console output or segfault backtrace using gdb.

mpatalberta, 2021-04-06:
From chomp_planner.cpp
  if (req.goal_constraints[0].joint_constraints.empty() || !req.goal_constraints[0].position_constraints.empty() ||
      !req.goal_constraints[0].orientation_constraints.empty())
  {
    ROS_ERROR_STREAM("Only joint-space goals are supported");
    res.error_code_.val = moveit_msgs::MoveItErrorCodes::INVALID_GOAL_CONSTRAINTS;
    return false;
  }

mpatalberta, 2021-04-06:
https://answers.ros.org/question/316245/chomp-can-only-support-joint-space-goals/
The link above is the same issue that I am having

mpatalberta, 2021-04-06:
https://github.com/ros-planning/moveit/blob/master/moveit_planners/chomp/chomp_motion_planner/src/chomp_planner.cpp

felixvd, 2021-04-06:
I’ve had the same issue before using C++.

mpatalberta, 2021-04-06:
Going back over history i can track to at least 2019 when one item was tested
From chomp_planner.cpp
if (req.goal_constraints[0].joint_constraints.empty() || !req.goal_constraints[0].position_constraints.empty() ||
!req.goal_constraints[0].orientation_constraints.empty())
{
ROS_ERROR_STREAM("Only joint-space goals are supported");
res.error_code_.val = moveit_msgs::MoveItErrorCodes::INVALID_GOAL_CONSTRAINTS;
return false;
}
to see if this causes problem;
if (req.goal_constraints[0].joint_constraints.empty() )
{
ROS_ERROR_STREAM("Only joint-space goals are supported");
res.error_code_.val = moveit_msgs::MoveItErrorCodes::INVALID_GOAL_CONSTRAINTS;
return false;
}

mpatalberta, 2021-04-07:
Felix what am saying the original code back in 2019 only did a single test
if (req.goal_constraints[0].joint_constraints.empty() )
{
ROS_ERROR_STREAM("Only joint-space goals are supported");
res.error_code_.val = moveit_msgs::MoveItErrorCodes::INVALID_GOAL_CONSTRAINTS;
return false;
}
I am unsure when this code was changed. I know some time back in 2019 it was as above

v4hn, 2021-04-07:
As far as I know the Chomp planner, by design, can't handle pose targets. It needs to know the joint configuration you are trying to reach to optimize the trajectory for this one goal.
If you are happy with an implementation that computes IK based on the previous state and feeds the result (if IK succeeded) into Chomp, you can simply use the [`PoseStamped` variant of `set_joint_value_target`](https://github.com/ros-planning/moveit/blob/master/moveit_commander/src/moveit_commander/move_group.py#L206).
(Note to other maintainers: that method is the most unpythonic way to implement such a mechanism I've ever seen and it should really just die...)

mpatalberta, 2021-04-07:
We removed the test below from chomp_planning.cpp and recompiled moveit using noetic. We could never git to with melodic.
if (req.goal_constraints[0].joint_constraints.empty() )
{
ROS_ERROR_STREAM("Only joint-space goals are supported");
res.error_code_.val = moveit_msgs::MoveItErrorCodes::INVALID_GOAL_CONSTRAINTS;
return false;
}

mpatalberta, 2021-04-07:
The change of removing the test with noetic would allow us to use the python api to run omlp  post process chomp planners

v4hn, 2021-04-08:
> The change of removing the test with noetic would allow us to use the python api to run omlp post process chomp planners

Feel free to add a pull-request with a proper explanation of what works/doesn't work.
It's hard to understand your problem/solution because of broken grammar or fuzzy statements.
- what planning pipeline fails
- why is this test (which seems legitimate, as I stated before) a problem?