moveit/moveit #2590: Unexpected commanded velocity upon direction change

mrunaljsarvaiya, 2021-04-08:
### Description
Last end effector velocity doesn't seem to get set to zero once a 0 twist command is sent. 

* ROS Distro: [Noetic]
* OS Version: Ubuntu 20.04.2
* Source: noetic-devel

### Steps to reproduce
I'm running moveit_servo on my custom robot arm (2dof planar arm with a speherical wrist). I've noticed that when switching the direction of a `delta_twist_cmd`, eg:
- Step 1: `twist:
  linear:
    x: 0.01
    y: 0.00
    z: 0.00
  angular:
    x: 0.0
    y: 0.0
    z: 0.0"`
- Step 2: Stop the eff by commanding 0 velocities
- Step 3: `twist:
  linear:
    x: -0.01
    y: 0.0
    z: 0.0
  angular:
    x: 0.0
    y: 0.0
    z: 0.0"`

### Expected behaviour
At step 3, I would have expected the eff's velocity to decrease from 0 to -0.05

### Actual behaviour
At step 3, eff's velocity reduces from 0.05 to -0.05

Is there some code that tries to gradually reduce the velocity from the last recorded velocity? Perhaps that last_velocity isn't getting updated?

closed 2021-04-09 by mrunaljsarvaiya

AndyZe, 2021-04-09:
I'm curious if you closed the issue because you found a mistake on your end?

Because of the low-pass filter, it can take some time for any command to converge to expected.

mrunaljsarvaiya, 2021-04-09:
> I'm curious if you closed the issue because you found a mistake on your end?
> 
> Because of the low-pass filter, it can take some time for any command to converge to expected.

So it's odd but if I install moveit_servo (1.1.0) via the binary build (apt-get install ros-noetic-moveit-servo) I see the issue I described above, but when I install the same 1.1.0 release from source, I don't see this behavior anymore. 

Regarding the action of the low lass filter, I see the convergence issue if I'm instantaneously changing the direction of the commanded velocity. However, in this case I'll stop the eff by commanding a 0 velocity, wait until its stopped and then command a negative velocity