moveit/moveit #2593: Noetic Release 1.1.2

tylerjw, 2021-04-09:
It is happening: https://github.com/ros/rosdistro/pull/29063

tylerjw, 2021-04-09:
We are proud to announce a release of Noetic is on the way.
There were 60 separate commits that are new since the last release of moveit.
Here is a summary of what new changes there are:

## Features

* [moveit_chomp_optimizer_adapter] Sanitize CHOMP initialization method parameter (#2540 <https://github.com/ros-planning/moveit/issues/2540>)
* [moveit_commander] Add utility functions to Python PSI: add/detach objects, apply_planning_scene() (#2532 <https://github.com/ros-planning/moveit/issues/2532>)
* [moveit_commander] add get_active_joint_names (#2533 <https://github.com/ros-planning/moveit/issues/2533>)
* [moveit_commander] Add get_current_state_bounded(), get_current_state(), enforce_bounds() to MoveGroupCommander (#2356 <https://github.com/ros-planning/moveit/issues/2356>)
* [moveit_core] Make setToIKSolverFrame accessible again (#2580 <https://github.com/ros-planning/moveit/issues/2580>)
* [moveit_core] Python bindings for moveit_core (#2547 <https://github.com/ros-planning/moveit/issues/2547>)
* [moveit_core] add get_active_joint_names (#2533 <https://github.com/ros-planning/moveit/issues/2533>)
* [moveit_core] Provide a function to set the position of active joints in a JointModelGroup (#2456 <https://github.com/ros-planning/moveit/issues/2456>)
* [moveit_core] RobotModelBuilder: Add parameter to specify the joint rotation axis
* [moveit_core] RobotModelBuilder: Allow adding end effectors (#2454 <https://github.com/ros-planning/moveit/issues/2454>)
* [moveit_planners_chomp] Sanitize CHOMP initialization method parameter (#2540 <https://github.com/ros-planning/moveit/issues/2540>)
* [moveit_ros_manipulation] Order PlaceLocations by quality during planning (#2378 <https://github.com/ros-planning/moveit/issues/2378>)
* [moveit_ros_move_group] Support multiple planning pipelines with MoveGroup via MoveItCpp (#2127 <https://github.com/ros-planning/moveit/issues/2127>)
* [moveit_ros_planning] Support multiple planning pipelines with MoveGroup via MoveItCpp (#2127 <https://github.com/ros-planning/moveit/issues/2127>)
* [moveit_ros_planning] FixStartStateBounds: Copy attached bodies when adapting the start state (#2398 <https://github.com/ros-planning/moveit/issues/2398>)
* [moveit_ros_planning_interface] Add get_active_joint_names (#2533 <https://github.com/ros-planning/moveit/issues/2533>)
* [moveit_ros_visualization] Allow selecting planning pipeline in RViz MotionPlanningDisplay
* [moveit_servo] Halt Servo command on Pose Tracking stop (#2501 <https://github.com/ros-planning/moveit/issues/2501>)
* [moveit_servo] Low latency mode (#2401 <https://github.com/ros-planning/moveit/issues/2401>)
* [moveit_setup_assistant] Update MSA launch templates for multi-pipeline support
* [pilz_industrial_motion_planner] Use kinematics solver timeout if not specified otherwise (#2489 <https://github.com/ros-planning/moveit/issues/2489>)
* [pilz_industrial_motion_planner] Add Pilz industrial motion planner (#1893 <https://github.com/ros-planning/moveit/issues/1893>)

## Testing

* [moveit_planners_ompl] Add test to ompl interface for StateValidityChecker (#2247 <https://github.com/ros-planning/moveit/issues/2247>)
* [moveit_planners_ompl] Add test for PlanningContextManager in ompl interface (#2248 <https://github.com/ros-planning/moveit/issues/2248>)
* [moveit_ros_perception] Enable mesh filter (#2448 <https://github.com/ros-planning/moveit/issues/2448>)
* [moveit_ros_planning_interface] Improve robustness of subframes test (#2488 <https://github.com/ros-planning/moveit/issues/2488>)
* [moveit_ros_planning_interface] Improve robustness of move group interface test (#2484 <https://github.com/ros-planning/moveit/issues/2484>)
* [moveit_ros_planning_interface] Unit Test for ByteString-based ROS msg conversion (#2362 <https://github.com/ros-planning/moveit/issues/2362>)

## Documentaion

* [moveit_core] Update doxygen comments for distance() and interpolate() (#2528 <https://github.com/ros-planning/moveit/issues/2528>)
* [moveit_core] Clean up collision-related log statements (#2480 <https://github.com/ros-planning/moveit/issues/2480>)
* [moveit_core] Fix doxygen documentation for setToIKSolverFrame (#2461 <https://github.com/ros-planning/moveit/issues/2461>)
* [moveit_core] Update collision-related comments (#2382 <https://github.com/ros-planning/moveit/issues/2382>) (#2388 <https://github.com/ros-planning/moveit/issues/2388>)
* [moveit_ros_planning] Fix some typos in comments (#2466 <https://github.com/ros-planning/moveit/issues/2466>)
* [moveit_ros_planning_interface] Add debugging log statement for a common error (#2509 <https://github.com/ros-planning/moveit/issues/2509>)

## Maintnence

* [moveit_core] Replaced eigen+kdl conversions with tf2_eigen + tf2_kdl (#2472 <https://github.com/ros-planning/moveit/issues/2472>)
* [moveit_core] Delete CollisionRequest min_cost_density
* [moveit_core] Changed processing_thread_ spin to use std::make_unique instead of new (#2412 <https://github.com/ros-planning/moveit/issues/2412>)
* [moveit_kinematics] Python3 compatibility for ikfast's round_collada_numbers.py (#2473 <https://github.com/ros-planning/moveit/issues/2473>)
* [moveit_ros_planning_interface] Fix scaling factor parameter names (#2452 <https://github.com/ros-planning/moveit/issues/2452>)
* [moveit_ros_robot_interaction] Suppress warnings "mesh_use_embedded_materials is ignored"
* [moveit_servo] Refactor Servo velocity bounds enforcement (#2471 <https://github.com/ros-planning/moveit/issues/2471>)
[... cut ...]
* [moveit_ros_occupancy_map_monitor] It's not an error not to define a plugin (#2521 <https://github.com/ros-planning/moveit/issues/2521>)
[... 34 lines cut ...]
* Felix von Drigalski
* Henning Kayser
* Immanuel Martini
* Jafar Abdi
* Jere Liukkonen
* Jeroen
* John Stechschulte
* Markus Vieth
* Michael Görner
* Nathan Brooks
* parunapu
* pe
* Peter Mitrano
* petkovich
* Pilz GmbH and Co. KG
* Robert Haschke
* sevangelatos
* Shota Aoki
* Simon Schmeisser
* Stuart Anderson
* Thomas G
* tkovich
* Tobias Fischer
* Tyler Weaver
* Udbhavbisarya23

tylerjw, 2021-04-09:
:point_up: that's the summary I've prepared for discourse, please let me know if you see anything you'd like changed.

AndyZe, 2021-04-09:
Maintnence --> Maintenance

felixvd, 2021-04-09:
I actually thought that was tongue-in-cheek and on purpose

Looks good to me though

jschleicher, 2021-04-12:
> * **[pilz_industrial_motion_planner]** Use kinematics solver timeout if not specified otherwise [#2489](https://github.com/ros-planning/moveit/issues/2489)

Maybe move 2489 down to Maintenance as well, since it was a follow-up fix after integration of the pilz planner.

Overall some features touching multiple packages are duplicated. For example 2533 is listed three times. Imho the style of [previous discourse posts](https://discourse.ros.org/t/announcing-moveit-1-1-1-release-for-ros-noetic/16802) without individual package names was neater (you could still add pr numbers).

tylerjw, 2021-04-12:
We are proud to announce a release of Noetic is on the way.
There were 60 separate commits that are new since the last release of moveit.
Here is a summary of what new changes there are:

## Features

* Add utility functions to Python PSI: add/detach objects, apply_planning_scene() [#2532](https://github.com/ros-planning/moveit/issues/2532)
* Make setToIKSolverFrame accessible again [#2580](https://github.com/ros-planning/moveit/issues/2580)
* Python bindings for moveit_core [#2547](https://github.com/ros-planning/moveit/issues/2547)
* add get_active_joint_names [#2533](https://github.com/ros-planning/moveit/issues/2533)
* Provide a function to set the position of active joints in a JointModelGroup [#2456](https://github.com/ros-planning/moveit/issues/2456)
* RobotModelBuilder: Add parameter to specify the joint rotation axis
* RobotModelBuilder: Allow adding end effectors [#2454](https://github.com/ros-planning/moveit/issues/2454)
* Sanitize CHOMP initialization method parameter [#2540](https://github.com/ros-planning/moveit/issues/2540)
* Order PlaceLocations by quality during planning [#2378](https://github.com/ros-planning/moveit/issues/2378)
* Support multiple planning pipelines with MoveGroup via MoveItCpp [#2127](https://github.com/ros-planning/moveit/issues/2127)
* FixStartStateBounds: Copy attached bodies when adapting the start state [#2398](https://github.com/ros-planning/moveit/issues/2398)
* Allow selecting planning pipeline in RViz MotionPlanningDisplay
* Low latency mode [#2401](https://github.com/ros-planning/moveit/issues/2401)
* Update MSA launch templates for multi-pipeline support
* Add Pilz industrial motion planner [#1893](https://github.com/ros-planning/moveit/issues/1893)

## Testing

* Add test to ompl interface for StateValidityChecker [#2247](https://github.com/ros-planning/moveit/issues/2247)
* Add test for PlanningContextManager in ompl interface [#2248](https://github.com/ros-planning/moveit/issues/2248)
* Enable mesh filter [#2448](https://github.com/ros-planning/moveit/issues/2448)
* Improve robustness of subframes test [#2488](https://github.com/ros-planning/moveit/issues/2488)
* Improve robustness of move group interface test [#2484](https://github.com/ros-planning/moveit/issues/2484)
* Unit Test for ByteString-based ROS msg conversion [#2362](https://github.com/ros-planning/moveit/issues/2362)

## Documentaion

* Update doxygen comments for distance() and interpolate() [#2528](https://github.com/ros-planning/moveit/issues/2528)
* Clean up collision-related log statements [#2480](https://github.com/ros-planning/moveit/issues/2480)
* Fix doxygen documentation for setToIKSolverFrame [#2461](https://github.com/ros-planning/moveit/issues/2461)
* Update collision-related comments [#2382](https://github.com/ros-planning/moveit/issues/2382) [#2388](https://github.com/ros-planning/moveit/issues/2388)
* Fix some typos in comments [#2466](https://github.com/ros-planning/moveit/issues/2466)
* Add debugging log statement for a common error [#2509](https://github.com/ros-planning/moveit/issues/2509)

## Maintenance

* Replaced eigen+kdl conversions with tf2_eigen + tf2_kdl [#2472](https://github.com/ros-planning/moveit/issues/2472)
* Delete CollisionRequest min_cost_density
* Changed processing_thread_ spin to use std::make_unique instead of new [#2412](https://github.com/ros-planning/moveit/issues/2412)
* Python3 compatibility for ikfast's round_collada_numbers.py [#2473](https://github.com/ros-planning/moveit/issues/2473)
* Fix scaling factor parameter names [#2452](https://github.com/ros-planning/moveit/issues/2452)
* Suppress warnings "mesh_use_embedded_materials is ignored"
* Refactor Servo velocity bounds enforcement [#2471](https://github.com/ros-planning/moveit/issues/2471)
* Cleanup current state handling in servo [#2372](https://github.com/ros-planning/moveit/issues/2372)
* Change servo namespacing logic [#2354](https://github.com/ros-planning/moveit/issues/2354)
* MSA launch files: fix indentation [#2371](https://github.com/ros-planning/moveit/issues/2371)
* pilz planner: add string includes [#2483](https://github.com/ros-planning/moveit/issues/2483)
* Use kinematics solver timeout if not specified otherwise [#2489](https://github.com/ros-planning/moveit/issues/2489)
* Upgrade cmake_minimum_required to 3.1 [#2453](https://github.com/ros-planning/moveit/issues/2453)
* Halt Servo command on Pose Tracking stop [#2501](https://github.com/ros-planning/moveit/issues/2501)

## Bug Fixes

* Fix logic, improve function comment for clearDiffs() [#2497](https://github.com/ros-planning/moveit/issues/2497)
* Fix RobotState::dropAccelerations/dropEffort to not drop velocities [#2478](https://github.com/ros-planning/moveit/issues/2478)
* Fix validation of orientation constraints [#2434](https://github.com/ros-planning/moveit/issues/2434)
* Fix OrientationConstraint::decide [#2414](https://github.com/ros-planning/moveit/issues/2414)
* Fix missing isEmpty check in compute_ik service [#2544](https://github.com/ros-planning/moveit/issues/2544)
* It's not an error not to define a plugin [#2521](https://github.com/ros-planning/moveit/issues/2521)
* Fixed flood of errors on startup for mesh_filter [#2550](https://github.com/ros-planning/moveit/issues/2550)
* PlanExecution: Correctly handle preempt-requested flag [#2554](https://github.com/ros-planning/moveit/issues/2554)
* thread safety in clear octomap & only update geometry [#2500](https://github.com/ros-planning/moveit/issues/2500)
[... cut ...]
* Fix various issues in PlanningScene / MotionPlanning displays [#2588](https://github.com/ros-planning/moveit/issues/2588)
[... 28 lines cut ...]
* David V. Lu!!
* Felix von Drigalski
* Henning Kayser
* Immanuel Martini
* Jafar Abdi
* Jere Liukkonen
* Jeroen
* John Stechschulte
* Markus Vieth
* Michael Görner
* Nathan Brooks
* parunapu
* Peter Mitrano
* petkovich
* Pilz GmbH and Co. KG
* Robert Haschke
* sevangelatos
* Shota Aoki
* Simon Schmeisser
* Stuart Anderson
* Thomas G
* tkovich
* Tobias Fischer
* Tyler Weaver
* Udbhavbisarya23

k-okada, 2021-04-16:
moveit 1.1.2 have some trouble when installing with tf-conversion package both try to install `'/opt/ros/noetic/lib/python3/dist-packages/libgtest.so'`.

```
  2270 'Preparing to unpack .../497-ros-noetic-tf_1.13.2-1focal.20210414.184309_amd64.deb ...'
  2271 'Unpacking ros-noetic-tf (1.13.2-1focal.20210414.184309) ...'
  2272 'dpkg: error processing archive /tmp/apt-dpkg-install-rMeCvu/497-ros-noetic-tf_1.13.2-1focal.20210414.184309_amd64.deb (--unpack):'
  2273 ' trying to overwrite '/opt/ros/noetic/lib/python3/dist-packages/libgtest.so', which is also in package ros-noetic-moveit-core 1.1.2-1focal.20210414.202258'
  2274 'dpkg-deb: error: paste subprocess was killed by signal (Broken pipe)'
  2275 'Selecting previously unselected package ros-noetic-tf-conversions.'
  2276 'Preparing to unpack .../498-ros-noetic-tf-conversions_1.13.2-1focal.20210414.190640_amd64.deb ...'
  2277 'Unpacking ros-noetic-tf-conversions (1.13.2-1focal.20210414.190640) ...'
  2278 'Errors were encountered while processing:'
  2279 ' /tmp/apt-dpkg-install-rMeCvu/497-ros-noetic-tf_1.13.2-1focal.20210414.184309_amd64.deb'
  2280 'E: Sub-process /usr/bin/dpkg returned an error code (1)'
```

tylerjw, 2021-04-16:
@v4hn responding here because I can't find the thread where we were talking about a minor followup release.  Until this issue: https://github.com/ros/rosdistro/issues/29173  with rosdistro is fixed it doesn't seem like there can be a Noetic release.  As a result I'm going to wait till next week to do another minor release.

peci1, 2021-04-20:
The catkin PR resolving ros/rosdistro#29173 has been merged, could you trigger a new moveit_core CI build to see if it fixed the issue?

tylerjw, 2021-04-20:
I think it has to be triggered on the buildfarm after catkin is released.

closed 2021-05-04 by tylerjw