moveit/moveit #2599: ROS-Moveit ERROR：Group 'arm' is not a chain

fekix9, 2021-04-12:
ROS:neotic
Moveit：1.1.0-1focal.20210304.173046

When I run the demo, the model in rviz cannot be dragged and planned.

<pre>PARAMETERS
 * /joint_state_publisher/source_list: [&apos;move_group/fake...
 * /move_group/allow_trajectory_execution: True
 * /move_group/arm/default_planner_config: None
 * /move_group/arm/longest_valid_segment_fraction: 0.005
 * /move_group/arm/planner_configs: [&apos;AnytimePathShor...
 * /move_group/arm/projection_evaluator: joints(shoulder_p...
 * /move_group/capabilities: 
 * /move_group/controller_list: [{&apos;name&apos;: &apos;fake_a...
 * /move_group/disable_capabilities: 
 * /move_group/gripper/default_planner_config: None
 * /move_group/gripper/longest_valid_segment_fraction: 0.005
 * /move_group/gripper/planner_configs: [&apos;AnytimePathShor...
 * /move_group/gripper/projection_evaluator: joints(finger_joi...
 * /move_group/initial: [{&apos;group&apos;: &apos;arm&apos;,...
 * /move_group/jiggle_fraction: 0.05
 * /move_group/max_range: 5.0
 * /move_group/max_safe_path_cost: 1
 * /move_group/moveit_controller_manager: moveit_fake_contr...
 * /move_group/moveit_manage_controllers: True
 * /move_group/octomap_resolution: 0.025
 * /move_group/planner_configs/AnytimePathShortening/hybridize: True
 * /move_group/planner_configs/AnytimePathShortening/max_hybrid_paths: 24
 * /move_group/planner_configs/AnytimePathShortening/num_planners: 4
 * /move_group/planner_configs/AnytimePathShortening/planners: 
 * /move_group/planner_configs/AnytimePathShortening/shortcut: True
 * /move_group/planner_configs/AnytimePathShortening/type: geometric::Anytim...
 * /move_group/planner_configs/BFMT/balanced: 0
 * /move_group/planner_configs/BFMT/cache_cc: 1
 * /move_group/planner_configs/BFMT/extended_fmt: 1
 * /move_group/planner_configs/BFMT/heuristics: 1
 * /move_group/planner_configs/BFMT/nearest_k: 1
 * /move_group/planner_configs/BFMT/num_samples: 1000
 * /move_group/planner_configs/BFMT/optimality: 1
 * /move_group/planner_configs/BFMT/radius_multiplier: 1.0
 * /move_group/planner_configs/BFMT/type: geometric::BFMT
 * /move_group/planner_configs/BKPIECE/border_fraction: 0.9
 * /move_group/planner_configs/BKPIECE/failed_expansion_score_factor: 0.5
 * /move_group/planner_configs/BKPIECE/min_valid_path_fraction: 0.5
 * /move_group/planner_configs/BKPIECE/range: 0.0
 * /move_group/planner_configs/BKPIECE/type: geometric::BKPIECE
 * /move_group/planner_configs/BiEST/range: 0.0
 * /move_group/planner_configs/BiEST/type: geometric::BiEST
 * /move_group/planner_configs/BiTRRT/cost_threshold: 1e300
 * /move_group/planner_configs/BiTRRT/frountier_node_ratio: 0.1
 * /move_group/planner_configs/BiTRRT/frountier_threshold: 0.0
 * /move_group/planner_configs/BiTRRT/init_temperature: 100
 * /move_group/planner_configs/BiTRRT/range: 0.0
 * /move_group/planner_configs/BiTRRT/temp_change_factor: 0.1
 * /move_group/planner_configs/BiTRRT/type: geometric::BiTRRT
 * /move_group/planner_configs/EST/goal_bias: 0.05
 * /move_group/planner_configs/EST/range: 0.0
 * /move_group/planner_configs/EST/type: geometric::EST
 * /move_group/planner_configs/FMT/cache_cc: 1
 * /move_group/planner_configs/FMT/extended_fmt: 1
 * /move_group/planner_configs/FMT/heuristics: 0
 * /move_group/planner_configs/FMT/nearest_k: 1
 * /move_group/planner_configs/FMT/num_samples: 1000
 * /move_group/planner_configs/FMT/radius_multiplier: 1.1
 * /move_group/planner_configs/FMT/type: geometric::FMT
 * /move_group/planner_configs/KPIECE/border_fraction: 0.9
 * /move_group/planner_configs/KPIECE/failed_expansion_score_factor: 0.5
 * /move_group/planner_configs/KPIECE/goal_bias: 0.05
 * /move_group/planner_configs/KPIECE/min_valid_path_fraction: 0.5
 * /move_group/planner_configs/KPIECE/range: 0.0
[... 204 lines cut ...]

[ INFO] [1618240889.215375193]: Stereo is NOT SUPPORTED
[ INFO] [1618240889.215442760]: OpenGL device: Mesa Intel(R) HD Graphics 630 (KBL GT2)
[ INFO] [1618240889.215467995]: OpenGl version: 4.6 (GLSL 4.6) limited to GLSL 1.4 on Mesa system.
[ INFO] [1618240892.538014245]: Loading robot model &apos;myarm1&apos;...
[ INFO] [1618240892.538069791]: No root/virtual joint specified in SRDF. Assuming fixed joint
<font color="#C4A000">[ WARN] [1618240892.538354618]: Link gripper_finger_link1 has visual geometry but no collision geometry. Collision geometry will be left empty. Fix your URDF file by explicitly specifying collision geometry.</font>
<font color="#C4A000">[ WARN] [1618240892.538385975]: Link gripper_finger_link2 has visual geometry but no collision geometry. Collision geometry will be left empty. Fix your URDF file by explicitly specifying collision geometry.</font>
<font color="#C4A000">[ WARN] [1618240892.538402547]: Link grasping_frame has visual geometry but no collision geometry. Collision geometry will be left empty. Fix your URDF file by explicitly specifying collision geometry.</font>
<font color="#CC0000">[ERROR] [1618240892.574731094]: Group &apos;arm&apos; is not a chain</font>
<font color="#CC0000">[ERROR] [1618240892.575428985]: Kinematics solver of type &apos;kdl_kinematics_plugin/KDLKinematicsPlugin&apos; could not be initialized for group &apos;arm&apos;</font>
<font color="#CC0000">[ERROR] [1618240892.575765806]: Kinematics solver could not be instantiated for joint group arm.</font>
[ INFO] [1618240892.616883299]: Starting planning scene monitor
[ INFO] [1618240892.619470256]: Listening to &apos;/move_group/monitored_planning_scene&apos;
[ INFO] [1618240892.641831406]: No active joints or end effectors found for group &apos;arm&apos;. Make sure that kinematics.yaml is loaded in this node&apos;s namespace.
[ INFO] [1618240892.643472983]: No active joints or end effectors found for group &apos;arm&apos;. Make sure that kinematics.yaml is loaded in this node&apos;s namespace.
[ INFO] [1618240892.648170398]: Constructing new MoveGroup connection for group &apos;arm&apos; in namespace &apos;&apos;
<font color="#CC0000">[ERROR] [1618240892.951707093]: Client [/move_group] wants topic /move_group/goal to have datatype/md5sum [moveit_msgs/MoveGroupActionGoal/689f0d7213d5960a2a9e9b117a0fdb73], but our version has [moveit_msgs/MoveGroupActionGoal/6514728724a6f4ce3f2f5a26db255bc9]. Dropping connection.</font>
<font color="#CC0000">[ERROR] [1618240922.662611634]: Unable to connect to move_group action server &apos;move_group&apos; within allotted time (30s)</font>
[ INFO] [1618240922.662944787]: Constructing new MoveGroup connection for group &apos;arm&apos; in namespace &apos;&apos;
<font color="#CC0000">[ERROR] [1618240922.919773550]: Client [/move_group] wants topic /move_group/goal to have datatype/md5sum [moveit_msgs/MoveGroupActionGoal/689f0d7213d5960a2a9e9b117a0fdb73], but our version has [moveit_msgs/MoveGroupActionGoal/6514728724a6f4ce3f2f5a26db255bc9]. Dropping connection.</font>
<font color="#CC0000">[ERROR] [1618240952.678639832]: Unable to connect to move_group action server &apos;move_group&apos; within allotted time (30s)</font>
</pre>

Can someone help me? What caused this? Is it a version issue?

fekix9, 2021-04-12:
My xacro and srdf files are [here](https://github.com/fekix9/moveit_myarm_version1.0)  . I used moveit_setup_assistant to generate   
 demo.launch.

felixvd, 2021-04-13:
Duplicate of [this ROS Answers question](https://answers.ros.org/question/375513/error-group-arm-is-not-a-chain/)

fekix9, 2021-04-13:
The problem is not solved

v4hn, 2021-04-13:
Nobody claimed that, but let's keep the discussion in one place. Sorry for not responding there. You didn't address @felixvd's questions there yet.

> Did you try removing the inertia from the base link and fixing the other errors? It seems like your versions don't match or you are building parts of moveit from source but not others. I don't see anything obviously wrong in the SRDF/URDF, but a picture would help. I might also try removing the gripper chain and end effector from the SRDF for debugging.

As long as you use inconsistent versions of MoveIt in the same workspace(s) - as indicated by the topic type discrepancy - that is the most likely source of your problem. I don't see anything obviously wrong with the description files.

closed 2021-04-13 by v4hn

fekix9, 2021-04-13:
May be you are right.But I don't know how to fix it.