moveit/moveit #2601: Make TrajectoryBlend a move_group_capability

felixvd, 2021-04-13:
Currently, trajectory blending is supported for MotionSequenceRequest message as described in [this tutorial](https://ros-planning.github.io/moveit_tutorials/doc/pilz_industrial_motion_planner/pilz_industrial_motion_planner.html#user-interface-sequence-capability)

However, there are use cases where one would want to combine trajectories directly, such as when you don't want to replan all of the trajectories, or simply when you don't want to change the API of an entire existing pipeline to the new `MotionSequenceRequest`.

@jschleicher @ct2034 At first sight, the structure of the [trajectory blender](https://github.com/ros-planning/moveit/blob/master/moveit_planners/pilz_industrial_motion_planner/include/pilz_industrial_motion_planner/trajectory_blender.h) looks very much like a move_group_capability already. Is there a reason why this shouldn't be wrapped by one? Would this make a good first issue?

v4hn, 2021-04-13:
Why would you wrap that in a ros service? It doesn't need any information from the `move_group` node.

It would be nice though to have easy access to this functionality from core MoveIt as well as the Python wrappers.

felixvd, 2021-04-13:
Only for the easy access via the move_group_interface, that's all. I thought the capabilities are the intended way to extend it.

It seems like it could be useful in lower level interfaces, but that seems like a bigger change. This one seems as small and useful as the `compute_ik` capability.

v4hn, 2021-04-13:
> Only for the easy access via the move_group_interface, that's all.
> I thought the capabilities are the intended way to extend it.

Not everything has to be part of the MGI. :D
And even there not everything is connected to `move_group`.
For example `getCurrentState` subscribes to `/joint_states`
and `setJointValueTarget`/`getJointValueTarget` actually provides a wrapper for `RobotState::setFromIK`.

> It seems like it could be useful in lower level interfaces, but that seems like a bigger change.

I don't see why it would be big to move some files around and add a method to a python wrapper. :)

> This one seems as small and useful as the `compute_ik` capability.

`compute_ik` actually needs to know the PlanningScene maintained by `move_group` to generate collision-free IK solutions (if they are requested).
Personally I think all other use of this service should actually be implemented in local code instead of generating RPC overhead.
We had sufficiently many users complaining that online control loops are so slow when they use such services...

jschleicher, 2021-04-14:
@felixvd It would sound great, to see further usage of the blending. But to be honest, I'm not sure, how generic the implementation is. Please take a look at https://github.com/ros-planning/moveit/blob/master/moveit_planners/pilz_industrial_motion_planner/doc/MotionBlendAlgorithmDescription.pdf to find out, if it would be feasible for your use case.