moveit/moveit #2605: building with install-space and DESTDIR set fails

simonschmeisser, 2021-04-15:
### Description

I'm trying to build a newer version of our "lib pack" or "platform" deb like this
```
	catkin config --build-space ../catkinworkspace-tmp/build
	catkin config --devel-space ../catkinworkspace-tmp/devel
	catkin config --log-space ../catkinworkspace-tmp/log
	catkin config --install-space /opt/isys-vision/mikado --install
	catkin config --source-space .
	mkdir -p $(shell pwd)/debian/mikado-ros/opt/isys-vision/mikado/include
	DESTDIR=$(shell pwd)/debian/mikado-ros/ catkin build --no-status --summary --parallel-packages 1 --cmake-args -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCATKIN_ENABLE_TESTING="OFF" -DCMAKE_CXX_COMPILER_LAUNCHER=ccache
```

but unfortunately it fails with the following message:
```
Errors << chomp_motion_planner:cmake /home/jenkins/workspace/mikado_mikado_ros_version-2.1/Entwicklung/../catkinworkspace-tmp/log/chomp_motion_planner/build.cmake.000.log
CMake Error at /home/jenkins/workspace/mikado_mikado_ros_version-2.1/Entwicklung/debian/mikado-ros/opt/isys-vision/mikado/share/moveit_core/cmake/moveit_coreConfig.cmake:113 (message):
  Project 'moveit_core' specifies '/opt/isys-vision/mikado/include' as an
  include dir, which is not found.  It does neither exist as an absolute
  directory nor in '${{prefix}}//opt/isys-vision/mikado/include'.  Check the
  issue tracker 'https://github.com/ros-planning/moveit/issues' and consider
  creating a ticket if the problem has not been reported yet.
Call Stack (most recent call first):
  /opt/ros/melodic/share/catkin/cmake/catkinConfig.cmake:76 (find_package)
  CMakeLists.txt:10 (find_package)

cd /home/jenkins/workspace/mikado_mikado_ros_version-2.1/Entwicklung/../catkinworkspace-tmp/build/chomp_motion_planner; catkin build --get-env chomp_motion_planner | catkin env -si  /usr/bin/cmake /home/jenkins/workspace/mikado_mikado_ros_version-2.1/Entwicklung/./extern/moveit/moveit_planners/chomp/chomp_motion_planner --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/jenkins/workspace/mikado_mikado_ros_version-2.1/Entwicklung/../catkinworkspace-tmp/devel/.private/chomp_motion_planner -DCMAKE_INSTALL_PREFIX=/opt/isys-vision/mikado -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCATKIN_ENABLE_TESTING=OFF -DCMAKE_CXX_COMPILER_LAUNCHER=ccache; cd -

...............................................................................
Failed << chomp_motion_planner:cmake                               [ Exited with code 1 ]
```
If I re-enable building packages in parallel I get this multiple times in packages depending on moveit_core

Has anyone seen this before? Is there another way of building bundles of packages that I'm missing?

### Your environment
* ROS Distro: Melodic
* OS Version: Ubuntu 18.04
* Source build
* If source, which branch? current master

v4hn, 2021-04-15:
> Is there another way of building bundles of packages that I'm missing?

chroot is an obvious alternative, but a bit more work. Is this really a moveit-specific problem?

simonschmeisser, 2021-04-15:
well, it used to work until recently (with melodic-devel) and the errors occur in MoveIt while there are other packages with dependencies as well 
```
[build] Successful packages:

 [Successful] ensenso_profiler                        
 [Successful] fcl                                     
 [Successful] geometric_shapes                        
 [   Ignored] industrial_core                         
 [Successful] industrial_deprecated                   
 [Successful] industrial_io_client                    
 [Successful] industrial_msgs                         
 [Successful] industrial_robot_client                 
 [Successful] industrial_utils                        
 [   Ignored] mikado_ros                              
 [   Ignored] moveit                                  
 [Successful] moveit_core                             
 [Successful] moveit_msgs                             
 [   Ignored] moveit_planners                         
 [   Ignored] moveit_plugins                          
 [   Ignored] moveit_ros                              
 [   Ignored] moveit_runtime                          
 [Successful] octomap_rviz_plugins                    
 [Successful] ompl                                    
 [Successful] rviz                                    
 [Successful] rviz_visual_tools                       
 [Successful] simple_message                          

[build] Failed packages:

 [    Failed] chomp_motion_planner                    

[build] Abandoned packages:

 [ Abandoned] industrial_robot_simulator              
 [ Abandoned] industrial_trajectory_filters           
 [ Abandoned] mikado_visualization                    
 [ Abandoned] moveit_chomp_optimizer_adapter          
 [ Abandoned] moveit_commander                        
 [ Abandoned] moveit_fake_controller_manager          
 [ Abandoned] moveit_kinematics                       
 [ Abandoned] moveit_opw_kinematics_plugin            
 [ Abandoned] moveit_planners_chomp                   
 [ Abandoned] moveit_planners_ompl                    
 [ Abandoned] moveit_ros_benchmarks                   
 [ Abandoned] moveit_ros_control_interface            
 [ Abandoned] moveit_ros_manipulation                 
 [ Abandoned] moveit_ros_move_group                   
 [ Abandoned] moveit_ros_occupancy_map_monitor        
 [ Abandoned] moveit_ros_perception                   
 [ Abandoned] moveit_ros_planning                     
 [ Abandoned] moveit_ros_planning_interface           
 [ Abandoned] moveit_ros_robot_interaction            
 [ Abandoned] moveit_ros_visualization                
 [ Abandoned] moveit_ros_warehouse                    
 [ Abandoned] moveit_servo                            
 [ Abandoned] moveit_setup_assistant                  
 [ Abandoned] moveit_simple_controller_manager        
 [ Abandoned] moveit_visual_tools                     
 [ Abandoned] pilz_industrial_motion_planner          
 [ Abandoned] pilz_industrial_motion_planner_testutils
 [ Abandoned] urdf_preview_plugin
```

simonschmeisser, 2021-04-15:
it seems that creating that folder (/opt/isys-vision/mikado/include) in the docker image beforehand makes it compile. I should probably file this as a catkin or catkin tools issue.

closed 2021-04-15 by simonschmeisser