moveit/moveit #2606: new moveit_core python documentation

PeterMitrano, 2021-04-15:
This is sort of a meta-issue for tracking documentation of the new python bindings. I'm worried things might not be working, but maybe it's just taking time for things to sync.

First of all, I see that http://docs.ros.org/en/noetic/api/moveit_core/html/ now has python and c++ which suggests my docs changes in #2547 are happening, however the python docs are not showing up. 

I'm also seeing a failed build here

https://travis-ci.com/github/ros-planning/moveit_tutorials/jobs/497517586

which is weird because for the C++ external links it's complaining don't exist, _do_ exist, for example:

http://docs.ros.org/noetic/api/moveit_core/html/cpp/classplanning__scene_1_1PlanningScene.html

PeterMitrano, 2021-05-03:
I'm looking on the build farm to try to find why it's not happening, but I can't find any docs jobs for moveit_core. If anyone has suggestions where to look or what might be wrong I'd appreciate it

rhaschke, 2021-05-03:
> I'm also seeing a failed build here https://travis-ci.com/github/ros-planning/moveit_tutorials/jobs/497517586

This was due to the cpp subfolder introduced with your cpp + python doc config. It was fixed in https://github.com/ros-planning/moveit_tutorials/pull/628

> I'm looking on the build farm to try to find why it's not happening, but I can't find any docs jobs for moveit_core.

The doc job is here: https://build.ros.org/view/Ndoc/job/Ndoc__moveit__ubuntu_focal_amd64/
Could you compare the output [there](https://build.ros.org/view/Ndoc/job/Ndoc__moveit__ubuntu_focal_amd64/15/consoleFull#console-section-24) with your local one?

PeterMitrano, 2021-05-03:
Thanks! I see an error:

```
06:25:49 Configuration error:
06:25:49 There is a programmable error in your configuration file:
06:25:49 
06:25:49 Traceback (most recent call last):
06:25:49   File "/usr/lib/python3/dist-packages/sphinx/config.py", line 368, in eval_config_file
06:25:49     execfile_(filename, namespace)
06:25:49   File "/usr/lib/python3/dist-packages/sphinx/util/pycompat.py", line 150, in execfile_
06:25:49     exec_(code, _globals)
06:25:49   File "/tmp/ws/src/moveit/moveit_core/doc/conf.py", line 17, in <module>
06:25:49     import sphinx_rtd_theme
06:25:49 ModuleNotFoundError: No module named 'sphinx_rtd_theme'
```

Seems like that's not supported in whatever build environment this job is running in. Do you know how to change that? I.e. something like `pip install sphinx-rtd-theme` ? The default docs theme is really difficult to read and navigate, so I'd like to at least try to make the RTD theme work.

rhaschke, 2021-05-03:
I guess you need to pull in the dependencies explicitly via
[`<doc_depend>python3-sphinx-rtd-theme</doc_depend>` in package.xml](http://wiki.ros.org/catkin/package.xml#Dependencies).

PeterMitrano, 2021-05-03:
oh ok, I'll make a PR and try that. I noticed that locally using `doc_depend` doesn't work with rosdep (https://github.com/ros-infrastructure/rosdep/issues/498), so if that's what the build farm uses then we'll have to make it a regular `depend`

rhaschke, 2021-05-03:
Hm. I hope that the build farm uses a different mechanism then...

PeterMitrano, 2021-05-07:
Progress! The install worked, and http://docs.ros.org/en/noetic/api/moveit_core/html/python/ is live. But there are still errors

Here's the log:
https://build.ros.org/view/Ndoc/job/Ndoc__moveit__ubuntu_focal_amd64/16/consoleFull#console-section-24

Here are the errors/warnings
```
[autosummary] failed to import 'moveit.core': no module named moveit.core
api.rst:1: WARNING: Error in "autosummary" directive:
html_static_path entry '/tmp/ws/src/moveit/moveit_core/doc/_static' does not exist
```
So when rosdoc_lite calls the docs, perhaps it isn't running from a correctly setup environment in terms of python path or something? Because `import moveit.core` should work.

The second warning could be because sphinx is out of date. Not sure what version it uses, perhaps https://github.com/ros-infrastructure/catkin-sphinx ? which is definitely out of date...

The last error might be caused by the first two. Not sure.

Also not sure how to proceed with debugging, because these are issues that depend on me knowing things about the build farm config/infrastructure.

rhaschke, 2021-05-20:
Let's shift the discussion from https://github.com/ros-planning/moveit/pull/2645#issuecomment-834912867 to this open issue here.
The remaining issue is that we need a _full build_ to generate our `pybind11` bindings from which we then can auto-generate the python docs via rosdoc_lite / sphinx.

The solution suggested by @PeterMitrano in https://github.com/ros-planning/moveit/pull/2645#issuecomment-835455764 is to commit the auto-generated python docs into the source repository. This also could be automated by our CI process.
However, I don't like the idea to include auto-generated stuff into the git repository.

Alternatively, we could run `rosdoc_lite` as part of our GHA workflow and [rsync the generated docs ourselves](https://github.com/marketplace/actions/rsync-deployments-action) to http://docs.ros.org.
@tfoote, would that be possible, i.e. would you be willing to [grant us the corresponding permission](https://github.com/marketplace/actions/rsync-deployments-action#required-secret)?

gavanderhoorn, 2021-05-20:
What about the reverse? `rosdoc_lite` in CI, push results to some known place, have the buildfarm *import* the artefacts (perhaps as part of a custom doc job) and put them on `docs.ros.org`.

rhaschke, 2021-05-20:
Would work as well, but requires an intermediate "known place" to push/pull from.

PeterMitrano, 2021-05-20:
Perhaps a separate GitHub repo would be an ok intermediate place to have
that?

PeterMitrano, 2021-06-13:
Is anyone familiar with the build/CI infra willing to look into this at some point? This issue is blocking the tutorials (https://github.com/ros-planning/moveit_tutorials/pull/624) and more generally the expansion of the python bindings.

PeterMitrano, 2021-06-17:
so i started by trying to get rosdoc_lite installed on the CI server but I can't figure it out. Here's my attempt

https://github.com/UM-ARM-Lab/moveit/tree/build-docs-in-ci

I see in the [logs](https://pipelines.actions.githubusercontent.com/PuOsH1FFkLqv6NoW0lIDKMqZGeZNvv0f5bfWXXPEd4ISaeClRU/_apis/pipelines/1/runs/49/signedlogcontent/3?urlExpires=2021-06-17T23%3A12%3A05.8095877Z&urlSigningMethod=HMACV1&urlSignature=2pLog81uKMX6Y4OQSlzFx9Sg2J7s1bBtxLww5QXPnMQ%3D) that it installed it in the `install_target_dependencies` that it installed it, but I get `rosdoc_lite, command not found`. Perhaps it needs to be installed during the `install_downstream_dependencies` step but idk how to make that happen.

PeterMitrano, 2021-10-18:
I also like the second option, but I don't get why we'd piggyback on moveit_tutorials?

Regarding this page `https://moveit.ros.org/documentation/source-code-api/` I don't think people use this, fwiw.

I can spend a bit of time on this, I'm more motivated if I know you two are available for help/review. I did a bit of poking into the GHAs for moveit when I was trying to generate docs, and got stuck with rosdoc_lite not being available, so I feel like we stlil have that problem?

where should I look to start? Here? https://github.com/ros-planning/moveit_tutorials/runs/3886517242?check_suite_focus=true
seems like that's generating docs. Maybe we can set up an analogous action in the moveit repo that builds docs and deploys to `moveit.ros.org` ?

gavanderhoorn, 2021-10-18:
> Both of you, Peter and Michael, pointed out an important issue I didn't yet consider: If we use `moveit_tutorials` CI to generate the API docs, we won't get the latest updates from the MoveIt core repo.

can't you use something like [peter-evans/repository-dispatch](https://github.com/peter-evans/repository-dispatch) to trigger a workflow run from `moveit`?

rhaschke, 2021-10-31:
@nbbrooks and @tylerjw, this is neither related to ROS2 nor is it a chicken-and-egg problem. I have developed a procedure to build the API docs as part of our CI. You just have to decide **where** to deploy them, see https://github.com/ros-planning/moveit/pull/2914.

PeterMitrano, 2022-02-08:
Where do we stand on this? is there anything I should do to keep it moving?

addy1997, 2022-02-18:
@PeterMitrano, I would like to help with this issue too.

v4hn, 2022-07-28:
So it looks like [moveit.ros.org directly maps to github pages](https://dnschecker.org/all-dns-records-of-domain.php?query=moveit.ros.org&rtype=ALL&dns=google), so the whole domain content is handled there. From some very preliminary resource browsing it looks like all the website has to be deployed in one job.

I just looked at the [moveit.ros.org repo](https://github.com/ros-planning/moveit.ros.org) a bit and it seems the build-and-deploy workflow is automatically setup for the `main` branch in the "pages" config. This *could* be changed to an explicit workflow that involves [the deploy action](https://github.com/marketplace/actions/deploy-to-github-pages).

So @rhaschke, it looks like nobody is really responsible for the website which is why there was no response here.
There's just the (at the moment trivial) github pages config and a registered subdomain that refers to a github server.

I believe the best way forward would be to implement a custom deploy action that generates the documentation from the `moveit` repository and trigger the workflow in `moveit.ros.org` from this repo here whenever the relevant files change.
In theory I guess there could be some caching that does not require the docs to be rebuilt every time someone changes the website.

Thanks for mentioning this effort in the maintainer meeting @PeterMitrano .

vatanaksoytezer, 2022-07-28:
We used https://github.com/ros-planning/moveit2_tutorials/commit/f680cbe8533f6049986b09165806c8ae019fe31d for generating API for MoveIt2, does it look reasonable to do something similar for MoveIt?