moveit/moveit #2611: PointCloudOctomapUpdater removes too many points with convex collision geometry

v4hn, 2021-04-16:
As mentioned in https://github.com/ros-planning/moveit_tutorials/pull/627 , @lianghongzhuo found an issue in the updater plugin.

It seems at least in some circumstances the shape filter can remove points inside the convex hull of a non-convex collision mesh.

Has anyone observed such an issue before?

We can reproduce this in the example below. The whole rack was modeled as a single collision mesh (I know it could be better if we model it as primitives, but anyway...) and the pointcloud inside does not fill voxels in the octomap - they are filtered when points inside the (padded) mesh are filtered out. Padding of non-convex meshes does not work as expected anyway and is an open problem in `geometric_shapes` (@peci1 pointed some of them out somewhere and there were other discussions too), but the problem also exists when `padding_scale: 1.0 ; padding_offset : 0.0` (so no padding at all).

[image]

I will not debug this right now, but wanted to document it at least.

peci1, 2021-04-17:
The [shapes library](https://github.com/ros-planning/geometric_shapes/blob/noetic-devel/include/geometric_shapes/bodies.h) from `geometric_shapes` doesn't have a notion of non-convex meshes. So the filtering can't work if it uses this library (IMO). I'm not sure if this is documented somewhere, though...

v4hn, 2021-04-17:
Thanks @peci1 I only recalled that non-convex meshes will misbehave, not that they are not at all implemented.

The problematic lines are [here](https://github.com/ros-planning/moveit/blob/master/moveit_ros/perception/point_containment_filter/src/shape_mask.cpp#L72) and [here](https://github.com/ros-planning/geometric_shapes/blob/noetic-devel/src/body_operations.cpp#L58)
where any mesh is treated as convex.

It would be nice to trigger an error message in this case, so we should probably check for convexity on either side?

peci1, 2021-04-17:
To be more precise: shapes.h defines a library of shapes, and this one should support non-convex meshes. bodies.h defines a library of bodies (i.e. shapes with 3D world position), and this one knows only convex meshes.

I think for filtering, you need bodies, because you need to filter out points from shapes located at particular locations.