moveit/moveit #2612: Rviz displays reload robot on reset

v4hn, 2021-04-16:
I'm not sure whether I should label this as a bug or as a feature request ;)

This mostly concerns using moveit with `rosbag play --loop`. Because the MoveIt displays reset the robot model on every reset and each jump back in time triggers a full reset in rviz, robots models never stop loading with short recordings.
My group has some models around that take several seconds to load (mostly because they include bigger meshes), so using them with short bags containing only a few pointclouds is basically impossible.

It would be nice to keep the robot models around on reset and possibly - if we really need to handle the case only reload them if the situation changed after the reset.

What do you think @rhaschke ? Other opinions are of course very welcome as well :-)

rhaschke, 2021-04-17:
Hm. I understand the problem. On the other hand, one _wants_ to reload the robot model by resetting the display.
Of course, the clean approach would introduce a new virtual method `Display::onTimeReset()`, which then could implement different behavior from `reset()`. However, this API+ABI change isn't really feasible.
Maybe one could use `sender()` equals (or not) `nullptr` to distinguish both situations in the `PlanningSceneDisplay`.

rhaschke, 2021-04-25:
I tried my proposal, but doesn't work either, because the manually pushing the Reset button, simply calls `reset()` as well. `QObject::sender()` isn't defined in both cases.

rhaschke, 2021-04-25:
We need to fall back to @v4hn's proposal comparing the new robot model description against the old one. However, that's not so trivial, because the `RobotModel` loading is done by the `PlanningSceneMonitor`...