moveit/moveit #2615: crash on reload of MotionPlanning widget

christian-rauch, 2021-04-22:
### Description

RViz crashed because of `MotionPlanning` when the scene is reloaded without a time signal

### Your environment
Ubuntu 20.04, noetic binary

### Steps to reproduce

1. start RViz with MotionPlanning widget
2. play bag file (with `--clock` and `/use_sim_time` set)
3. pause the bag file, so that no time is published on `/clock`
4. click the "Reset" button in RViz
5. disable the MotionPlanning widget again
6. goto 4.

### Expected behaviour
The scene is cleared.

### Actual behaviour
RViz crashed.

### Backtrace or Console output

gdb backtracked:
```
#0  moveit_rviz_plugin::PlanningSceneRender::clear() (this=this@entry=0x0) at /usr/include/c++/9/bits/shared_ptr_base.h:1020
#1  0x00007fff7192f9cf in moveit_rviz_plugin::PlanningSceneRender::renderPlanningScene(std::shared_ptr<planning_scene::PlanningScene const> const&, rviz::Color const&, rviz::Color const&, moveit_rviz_plugin::OctreeVoxelRenderMode, moveit_rviz_plugin::OctreeVoxelColorMode, float) (this=this@entry=0x0, scene=warning: RTTI symbol not found for class 'std::_Sp_counted_ptr<planning_scene::PlanningScene*, (__gnu_cxx::_Lock_policy)2>'
warning: RTTI symbol not found for class 'std::_Sp_counted_ptr<planning_scene::PlanningScene*, (__gnu_cxx::_Lock_policy)2>'

std::shared_ptr<const planning_scene::PlanningScene> (use count 2, weak count 1) = {...}, default_env_color=..., default_attached_color=..., octree_voxel_rendering=moveit_rviz_plugin::OCTOMAP_OCCUPIED_VOXELS, octree_color_mode=octree_color_mode@entry=moveit_rviz_plugin::OCTOMAP_Z_AXIS_COLOR, default_scene_alpha=default_scene_alpha@entry=0.899999976)
    at ./rviz_plugin_render_tools/src/planning_scene_render.cpp:83
#2  0x00007fff71fe4ec5 in moveit_rviz_plugin::PlanningSceneDisplay::renderPlanningScene() (this=0x555556636900) at /opt/ros/noetic/include/moveit/planning_scene_monitor/planning_scene_monitor.h:633
#3  0x00007fff71fe5072 in moveit_rviz_plugin::PlanningSceneDisplay::updateInternal(float, float) (this=0x555556636900, wall_dt=<optimised out>)
    at ./planning_scene_rviz_plugin/src/planning_scene_display.cpp:651
#4  0x00007fff78156dad in moveit_rviz_plugin::MotionPlanningDisplay::updateInternal(float, float) (this=0x555556636900, wall_dt=0.132994667, ros_dt=0)
    at ./motion_planning_rviz_plugin/src/motion_planning_display.cpp:1302
#5  0x00007ffff7e75e7b in rviz::DisplayGroup::update(float, float) (this=0x5555556e1800, wall_dt=0.132994667, ros_dt=0) at ./src/rviz/display_group.cpp:242
#6  0x00007ffff7f3e979 in rviz::VisualizationManager::onUpdate() (this=0x555556534370) at ./src/rviz/visualization_manager.cpp:339
#7  0x00007ffff73d7300 in QMetaObject::activate(QObject*, int, int, void**) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#8  0x00007ffff73e451e in QTimer::timeout(QTimer::QPrivateSignal) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#9  0x00007ffff73d7cf5 in QObject::event(QEvent*) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#10 0x00007ffff77d8a66 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#11 0x00007ffff77e20f0 in QApplication::notify(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#12 0x00007ffff73ab93a in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#13 0x00007ffff74028b0 in QTimerInfoList::activateTimers() () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#14 0x00007ffff740319c in  () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#15 0x00007ffff45fe17d in g_main_context_dispatch () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#16 0x00007ffff45fe400 in  () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#17 0x00007ffff45fe4a3 in g_main_context_iteration () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#18 0x00007ffff7403565 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#19 0x00007ffff73aa4db in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#20 0x00007ffff73b2246 in QCoreApplication::exec() () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#21 0x000055555555860d in main(int, char**) (argc=<optimised out>, argv=<optimised out>) at ./src/rviz/main.cpp:42
```

Use [gist.github.com](gist.github.com) to copy-paste the console output or segfault backtrace using gdb.

v4hn, 2021-04-22:
I guess reverting https://github.com/ros-planning/moveit/pull/2588 (or a specific commit thereof) fixes the issue?

christian-rauch, 2021-04-22:
Did this PR already make it into the binary release? I am using the binary packages from the ROS repo (`ros-noetic-moveit-ros-visualization` is on `1.1.0-1focal.20210304.170022`).

v4hn, 2021-04-22:
> Did this PR already make it into the binary release?

Not into a deployed one.

> I am using the binary packages from the ROS repo (`ros-noetic-moveit-ros-visualization` is on `1.1.0-1focal.20210304.170022`).

In that case please verify that you can still reproduce this from the master branch first.

rhaschke, 2021-04-23:
How often do you need to repeat step 4? This seems to be related to https://github.com/ros-visualization/rviz/issues/1622, which doesn't have a solution.

christian-rauch, 2021-04-23:
I only had to repeat this 2~3 times. Yes, these issues might be related, but the plugin should not make RViz crash.

I don't know exactly why it crashes, but it might be related to a shared pointer access in `planning_scene_render.cpp:83`. This issue should be resolved (e.g. check for NULL pointer, etc.) so that RViz does not crash.

lianghongzhuo, 2021-04-24:
I also saw a rviz crash when using this pull request: https://github.com/ros-planning/moveit/pull/2596/files
I documented the way to reproduce there.
Maybe it related? But I cannot reproduce it in the current moveit master branch.

rhaschke, 2021-04-25:
@christian-rauch, I think this issue should have been fixed by #2588. At least, I cannot reproduce it on master. Please give it a try and report (or close this issue).
@lianghongzhuo, I think your issue is unrelated to this.