moveit/moveit #2616: distanceRobot not taking into account disabled collisions from SRDF

mikramarc, 2021-04-23:
### Description

The `distanceRobot` method from `collision_detection` does not take into account the `disable_collisions` sections from SRDF when computing distance.

### Your environment
* ROS Distro: Melodic
* OS Version: Ubuntu 18.04
* Source or Binary build? Source
* If source, which branch? melodic-devel

### Steps to reproduce and behaviour
Having in my srdf present:
```
  <disable_collisions link1="link_1" link2="link_2" reason="Adjacent"/>
```
When doing:

```
  robot_model_loader_.reset(new robot_model_loader::RobotModelLoader());
  kinematic_model_ = robot_model_loader_->getModel();
  planning_scene_.reset(new planning_scene::PlanningScene(kinematic_model_));
  auto robot_model_ptr = planning_scene_->getRobotModel();
  collision_detection::CollisionRobotFCL crobot(robot_model_ptr);

  auto dreq = collision_detection::DistanceRequest();
  auto dres = collision_detection::DistanceResult();

 crobot.distanceSelf(dreq, dres, current_state);
```

the `dreq` object is giving distance value of 0 and giving `link_names` values of `link_1` and `link_2`, even though they are present in the disable collisions section in srdf.

I can confirm that srdf is read correctly since when I do 
```
  auto dc = robot_model_ptr->getSRDF()->getDisabledCollisionPairs();
```
the `dc` object does indeed have the `link_1` and `link_2` in it.

I can work around that doing the following:
```
  for (auto const& element : dc)
  {
    collision_detection::AllowedCollisionMatrix acm;
    acm.setEntry(element.link1_, element.link2_, true);
    dreq.acm = &acm;
  }
```
and only then calling the `distanceSelf`, however that significantly slows down the execution of that method (which I don't understand either since this would actual reduce the number of queries as far as I know).

I'm not sure weather this is a bug, an actual desired behaviour for some reason or if maybe I am using the code in a wrong way. @Levi-Armstrong or anyone else with experience with the collision_common library, any ideas? I could provide the full SRDF file if necessary. Thanks in advance!

v4hn, 2021-04-23:
Could you try to reproduce the bug with MoveIt's `master` branch? [You can setup this current version on 18.04 as well](https://moveit.ros.org/install/source/). If it's there, I would appreciate a new test case in [`moveit_core/collision_detection`](https://github.com/ros-planning/moveit/tree/master/moveit_core/collision_detection/test)

v4hn, 2021-04-23:
From what I can see [the FCL plugin checks for the allowed collision entry and will skip the link pair](https://github.com/ros-planning/moveit/blob/master/moveit_core/collision_detection_fcl/src/collision_common.cpp#L453)

v4hn, 2021-04-23:
Ok, sorry, I looked at it in more detail now.

The API is intentionally designed that way. The collision robot (or collision environment in `master`) does not know the allowed collision matrix, though you can obviously reconstruct a version of it there the way you pointed out above.
The official ACM, which can be updated through ROS interfaces and not just read from the SRDF, lives with the `PlanningScene` that contains the collision environment.
Any kind of acm-aware checking should use this one and not generate its own version.

The API in the collision env is inconsistent though. most likely for historical reasons:
The [`CollisionRequest`](https://github.com/ros-planning/moveit/blob/master/moveit_core/collision_detection/include/moveit/collision_detection/collision_common.h#L184) struct does not contain the ACM and thus the [collision checking API offers overloads where you can also pass the ACM that should be used](https://github.com/ros-planning/moveit/blob/master/moveit_core/collision_detection/include/moveit/collision_detection/collision_env.h#L94).
On the other hand, [the `DistanceRequest` **does** include the ACM](https://github.com/ros-planning/moveit/blob/master/moveit_core/collision_detection/include/moveit/collision_detection/collision_common.h#L287), so there is no need for the overload. Instead the API expects that you set this member yourself if you want to use an ACM.

So you are basically doing the correct thing above with `dreq.acm = &acm;` (you shouldn't assign the pointer in the loop for sure though ;)).
Most likely, your approach is slower because you construct the ACM from scratch in each distance request and you have to allocate and fill in [multiple `std::map<string,...>` structures](https://github.com/ros-planning/moveit/blob/master/moveit_core/collision_detection/include/moveit/collision_detection/collision_matrix.h#L272).
Instead, you should fill in `dreq.acm = &planning_scene.getAllowedCollisionMatrix()`, so you need the planning scene instance (or its acm) in scope to create the distance request.

I will close this already because I consider the explanation to suffice, but feel free to follow up if you still think there might be a problem.

closed 2021-04-23 by v4hn

mikramarc, 2021-04-23:
Hi @v4hn, thanks for looking into this!
1) regarding the pointer, yes, I accidentally put it in the loop in my post when pasting bits of code from my files, definitely doesn't belong there, fixed above :)
2)
> Most likely, your approach is slower because you construct the ACM from scratch in each distance request and you have to allocate and fill in multiple std::map<string,...> structures.

Hm, thats the thing, thought, I don't. The `dreq.acm = &acm` is only done once in the programs lifecycle and than the dreq is just used by distanceRobot. Either way, this is still fair bit of information you provided, I will keep digging and post here if I find anything interesting, in the meantime, happy for this issue to stay closed. Cheers!