moveit/moveit #2617: [moveit_servo] PoseTracking/Servo example implementation details

fzoric8, 2021-04-24:
### Description

Hi everyone, for last few days, I've been trying to enable `moveit_servo`, e.g. `pose_tracking_example.cpp` for my robotic arm. 
I'm using schunk lwa4p robotic arm which has 6DOF and I'm using the same arm in GAZEBO simulation. 
I'm quite sure that I don't quite understand some parameters in `ur_servo_config` file. To be clear, I've managed to launch `pose_tracking_example.cpp` and I've managed to start `pose_tracking_example` but my arm moves very slowly, and eventually if I give it enough time, ends up in ground (position error is increasing instead of decreasing)

Parameters that confuse me are mostly related to topic names: 
```
cartesian_command_in_topic: delta_twist_cmds  # Topic for incoming Cartesian twist commands
joint_command_in_topic: delta_joint_cmds # Topic for incoming joint angle commands
joint_topic: /lwa4p/joint_states
status_topic: status # Publish status to this topic
command_out_topic: /lwa4p/joint_group_position_controller/command 
```
When I'm using `JointGroupPositionController` `delta_twist_cmds` are published, but `delta_joint_cmds` are missing and I'm getting `stale_joint_command=1` when debugging in `servo_calcs`. **Maybe that's where I'm missing something.**

I've tried to use `JointTrajectoryController` as for UR, but I'm getting: 
```
[ERROR] [1619176604.362611307]: Trajectory message contains waypoints that are not strictly increasing in time.
```
My roslaunch for running pose_tracking_example looks as follows: 
```
<node name="servo_server" pkg="moveit_servo" type="pose_tracking_example" output="screen" >
            <rosparam command="load" file="$(find schunk_lwa4p_control)/config/schunk_servo_config.yaml" />
            <rosparam command="load" file="$(find schunk_lwa4p_control)/config/pose_tracking_settings.yaml" />
</node>
```

I've slightly changed main run loop of pose_tracking_example.cpp to change all 3 axis simultaneously, but I've kept those changes quite small, so there should be no problems.  
I've tried almost all different combinations for modifying pose (only changing one axis, changing multiple axis etc..., larger change e.g. 0.0004, smaller change e.g. 0.0001 but behavior is almost always the same. 
Arm starts moving a little bit but after some time it starts jerking almost uncontrollably. 
I've thought that maybe `JointGroupPositionController` has some `PID `params for each case, but [this](https://answers.ros.org/question/245802/why-joint-position-controller-is-not-sensitive-to-changing-pid-gains/) should mean that it's not necessary if using position reference for position joint controllers. 

My current PID params are: 
```
# Maximum value of error integral for all PID controllers
windup_limit: 0.1

# PID gains
x_proportional_gain: 1.5
y_proportional_gain: 1.5
z_proportional_gain: 1.5
x_integral_gain: 0
y_integral_gain: 0 
z_integral_gain: 0
x_derivative_gain: 0.00
y_derivative_gain: 0.00
z_derivative_gain: 0.00

angular_proportional_gain: 0.5
angular_integral_gain: 0.0
angular_derivative_gain: 0.0
```
I didn't try to turn on "drift dimensions" because my robot is not **underactuated** as stated [here](https://github.com/ros-planning/moveit/issues/2436). 

### Used environment
* ROS Distro: Melodic
* OS Version: e.g. Ubuntu 18.
* Moveit build from source on `melodic-devel` branch 

### Steps to reproduce
I can provide Docker with ROS Distro and MoveIt built from source with corresponding git repository for schunk lwa4p control but I'm quite sure that I'm not understanding or using something correctly, but if someone has time and will, let me know I can provide instructions and everything else needed for recreation.  

### Expected behaviour

As far as I understand, pose tracker uses 4 PIDs (3 for x,y,z and 1 for orientation) to calculate twist command for end effector. After that, twist command is used by `servo` and `servo_calcs` to output joint commands. PID gains for pose tracker are defined in `pose_tracking.yaml` and rest of the relevant configuration parameters are defined in (in my case) in: `schunk_servo_config.yaml`. Idea is to use twist end effector command and Jacobian to determine joint velocities command. From commanded joint velocities and our control rate period we can basically integrate to get wanted joint position command, and that should theoretically work. 

I'm using `JointGroupPositionController` and position commands for each joints are published, but no matter what PID value I choose for `PoseTracker` (they are loaded correctly on parameter server and in node, I've checked) but position error continuously increases (more precisely, error is not corrected with joint commands, and my PID gains are not relevant, as if twist command doesn't propagate correctly, to be more precise, start position of end effector in world frame is`x=0.0, y=0.0, z=1.25` and my end position of end effector in world frame should be: `x=0.25, y=0.25, z=1.00`), position errors are: `x_error=0.26, y_error=0.249, z_error=-0.248` for proportional gains of `1.5, 15, 500`.

Now as I'm writing this, I'm thinking that I'm getting joint_velocities as input commands and publishing them as position commands which could explain slow moving that happens and also could explain why is pose tracker error continuously increasing no matter which proportional gain I'm using. 

Is it possible to command joint positions for servoing? 
What am I missing? 
How can I be sure in my `robot_link_command_frame` or how can I check it? 
What would you recommend for smoother servoing, position joint control or velocity joint control? 

**If you could help me with some hints or answers or questions which could help me with this, it would be highly appreciated.** 

### Actual behaviour

Decrease of pose tracker error and different behaviour of robotic arm for different PID params of a pose tracker. 

#### P.S. I'm really sorry for listing this kind of problem as a bug because it's more likely that I'm not using it correctly. I'm not sure where could I get an appropriate answer or explanation, so I'm writing it here. Thank you for your time and understanding.

AndyZe, 2021-04-26:
> When I'm using JointGroupPositionController delta_twist_cmds are published, but delta_joint_cmds are missing

That's what should be happening. The Pose Tracker publishes twist commands for Servo to follow. Sounds good so far.

Please do `rostopic info delta_twist_cmds` to verify that Servo is subscribed to `delta_twist_cmds`? I wonder if you need a `/` here:

cartesian_command_in_topic: /delta_twist_cmds

To make sure `delta_twist_cmds` is published at a fixed rate, you can do this to echo the average publish frequency:

`rostopic hz -w 100 /delta_twist_cmds`

> I've tried to use JointTrajectoryController as for UR

I'm pretty sure Gazebo works with a JointTrajectoryController or JointGroupPositionController. (Backed up by the links [here](https://answers.ros.org/question/345721/position-controller-in-gazebo-for-panda-arm/) and [here](https://answers.gazebosim.org//question/25578/what-does-jointgroupposition-controller-do-when-it-does-not-have-pid-gains-set/).)

If you do want to use a JointTrajectoryController, make sure `use_gazebo=true` in the Servo config.yaml.

> My current PID params are:

Those look fine. Possibly they should all be negative -- maybe they are pushing the robot AWAY from the target position.

> What would you recommend for smoother servoing, position joint control or velocity joint control?

Velocity control with Gazebo is typically awful -- I'd avoid that. Although it works great in real life!

> Now as I'm writing this, I'm thinking that I'm getting joint_velocities as input commands and publishing them as position commands which could explain slow moving that happens and also could explain why is pose tracker error continuously increasing no matter which proportional gain I'm using.

That could make sense! Or, maybe you need to negate all PID gains like I already said.

> How can I be sure in my robot_link_command_frame or how can I check it?

My understanding is, it's the root link of the URDF by default. Probably `base_link` or something like that.

^I hope something in this wall of text is helpful!

AndyZe, 2021-04-26:
It would probably help if you attached your two config files to look at

fzoric8, 2021-04-26:
Hi Andy, thank you for your help, my configuration files look like following. 

`schunk_servo_config`: 
```
###############################################
# Modify all parameters related to servoing here
###############################################
use_gazebo: true # Whether the robot is started in a Gazebo simulation environment

target_control_rate: 50 # [Hz] Control rate

## Properties of incoming commands
command_in_type: "speed_units" # "unitless"> in the range [-1:1], as if from joystick. "speed_units"> cmds are in m/s and rad/s
scale:
  # Scale parameters are only used if command_in_type=="unitless"
  linear:  0.6  # Max linear velocity. Meters per publish_period. Unit is [m/s]. Only used for Cartesian commands.
  rotational:  0.5 # Max angular velocity. Rads per publish_period. Unit is [rad/s]. Only used for Cartesian commands.
  # Max joint angular/linear velocity. Rads or Meters per publish period. Only used for joint commands on joint_command_in_topic.
  joint: 0.1
low_pass_filter_coeff: 2.  # Larger --> trust the filtered data more, trust the measurements less.

## Properties of outgoing commands
publish_period: 0.008  # 1/Nominal publish rate [seconds]
low_latency_mode: false  # Set this to true to publish as soon as an incoming Twist command is received (publish_period is ignored)

# What type of topic does your robot driver expect?
# Currently supported are std_msgs/Float64MultiArray (for ros_control JointGroupVelocityController or JointGroupPositionController)
# or trajectory_msgs/JointTrajectory (for Universal Robots and other non-ros_control robots)
command_out_type: std_msgs/Float64MultiArray
#command_out_type: trajectory_msgs/JointTrajectory

# What to publish? Can save some bandwidth as most robots only require positions or velocities
publish_joint_positions: true
publish_joint_velocities: false
publish_joint_accelerations: false

## MoveIt properties
move_group_name: arm  # Often 'manipulator' or 'arm'
planning_frame: world  # The MoveIt planning frame. Often 'base_link' or 'world'

## Other frames
ee_frame_name: lwa4p_link6
# The name of the end effector link, used to return the EE pose
robot_link_command_frame:  lwa4p_link6  # commands must be given in the frame of a robot link. Usually either the base or end effector

## Stopping behaviour
incoming_command_timeout: 15 # Stop servoing if X seconds elapse without a new command
# If 0, republish commands forever even if the robot is stationary. Otherwise, specify num. to publish.
# Important because ROS may drop some messages and we need the robot to halt reliably.
num_outgoing_halt_msgs_to_publish: 4

## Configure handling of singularities and joint limits
lower_singularity_threshold:  50  # Start decelerating when the condition number hits this (close to singularity)
hard_stop_singularity_threshold: 300 # Stop when the condition number hits this
joint_limit_margin: 0.1 # added as a buffer to joint limits [radians]. If moving quickly, make this larger.

## Topic names
cartesian_command_in_topic: delta_twist_cmds  # Topic for incoming Cartesian twist commands
joint_command_in_topic: delta_joint_cmds # Topic for incoming joint angle commands
joint_topic: /lwa4p/joint_states
status_topic: status # Publish status to this topic
command_out_topic: /lwa4p/joint_group_position_controller/command # Publish outgoing commands here
#command_out_topic: /lwa4p/arm_controller/command

## Collision checking for the entire robot body
check_collisions: true # Check collisions?
collision_check_rate: 40 # [Hz] Collision-checking can easily bog down a CPU if done too often.
# Two collision check algorithms are available:
# "threshold_distance" begins slowing down when nearer than a specified distance. Good if you want to tune collision thresholds manually.
# "stop_distance" stops if a collision is nearer than the worst-case stopping distance and the distance is decreasing. Requires joint acceleration limits
collision_check_type: threshold_distance
# Parameters for "threshold_distance"-type collision checking
self_collision_proximity_threshold: 0.01 # Start decelerating when a self-collision is this far [m]
scene_collision_proximity_threshold: 0.05 # Start decelerating when a scene collision is this far [m]
# Parameters for "stop_distance"-type collision checking
collision_distance_safety_factor: 100 # Must be >= 1. A large safety factor is recommended to account for latency
min_allowable_collision_distance: 0.005 # Stop if a collision is closer than this [m]
```

`pose_tracking_settings.yaml`: 
```
#################################
# PID parameters for pose seeking
#################################

# Maximum value of error integral for all PID controllers
windup_limit: 0.1

# PID gains --> Increase gains even more, test it in simulation, CHECK REAL ROBOT SAFETY
x_proportional_gain: 1.5
y_proportional_gain: 1.5
z_proportional_gain: 1.5
x_integral_gain: 0.0
y_integral_gain: 0.0
z_integral_gain: 0.0
x_derivative_gain: 0.00
y_derivative_gain: 0.00
z_derivative_gain: 0.00

angular_proportional_gain: 50
angular_integral_gain: 0.0
angular_derivative_gain: 0.0
```

I didn't even thought of negative gains, therefore I'll try that. 

Output of `rostopic info delta_twist_cmds`: 
```
rostopic info /servo_server/delta_twist_cmds 
Type: geometry_msgs/TwistStamped

Publishers: 
 * /servo_server (http://zozan-pc:40443/)

Subscribers: 
 * /servo_server (http://zozan-pc:40443/)
```

Output of `rostopiz hz /servo_server/delta_twist_cmds` is: 
```
average rate: 125.011
	min: 0.004s max: 0.012s std dev: 0.00072s window: 100
average rate: 125.005
	min: 0.004s max: 0.012s std dev: 0.00073s window: 100
average rate: 125.008
	min: 0.004s max: 0.011s std dev: 0.00050s window: 100
```

I have custom launch file with `servo_server` namespace for `pose_tracking_example` but that shouldn't be a problem. 
I'll try negative gains, I don't know how I forgot that, thank you for your response, I'll let you know if anything helped :) 
Thanks

fzoric8, 2021-04-27:
Hi, I've managed to figure it out. It seems that I've changed my PID params for joint position controller of each joint for my robot arm.  were good enough to execute moveIt trajectories, but not good enough to handle small joint position changes that were published by moveit_servo. After a little bit of tuning of my joint position controller params, `moveit_servo` works quite fine for small distances. I've even managed to wrap pose_tracker to action server which enables smooth transition between trajectory control mode and servo control mode.

Thank you for your help, I'm closing this issue, cause everything works as expected. 
**If you have similar problems as I had, please check parameters of your joint position controllers which are loaded on param 
server.**

fzoric8, 2021-08-18:
Hi,

fzoric8, 2021-08-18:
Hi, 

as I've digged deeper in whole implementation of `moveit_servo` and I'm having troubles regarding conceptual understading of some stuff. 

So, as seen in implementation, there are 4 PIDs that convert position/orientation tracking errors to end effector linear and angular velocities.

Then we use those velocities and jacobian matrix to get joint velocities. From joint velocities and sampling time, we get joint position increments which are then added to current joint position. Such implementation enables us to track certain position.  

However, I'm having some troubles conceptually understanding some stuff. To cut it short, I'm curios about singularities. Robot is in singularity when Jacobian loses rank. And I'm wondering, Jacobian is a matrix that describes relationship between linear/angular end effector velocities to joint velocities. However, if we reach certain orientation before reaching our wanted position, that would imply that our Jacobian is not of a full rank, because there's no angular velocity of end effector. Does that affect `applyVelocityScaling` method and should I enable drift of certain dimensions to mitigate this. I'm trying to grasp what's happening conceptually in a moment where position or orientation is reached and row of the jacobian is set to zero because there's no need to generate movement alongside or around that axis? 

I hope I've explained it well enough, thank you for your time :)

AndyZe, 2021-08-18:
What you wrote mostly sounds good up to this point:

> I'm trying to grasp what's happening conceptually in a moment where position or orientation is reached and row of the Jacobian is set to zero because there's no need to generate movement alongside or around that axis?

The Jacobian doesn't become singular because you no longer want to generate motion along an axis. Requesting zero velocity for an axis is fine during normal operation.

It only becomes singular if the robot is actually at a singularity.

AndyZe, 2021-08-18:
Maybe what you could do is monitor the `status` topic and enable drift dimensions if the status topic is publishing 1 or 2. (status codes are defined [here](https://github.com/ros-planning/moveit/blob/master/moveit_ros/moveit_servo/include/moveit_servo/status_codes.h)).

@nbbrooks has kinda been contemplating damped least squares to help around singularities, but nothing implemented yet.

fzoric8, 2021-08-18:
@AndyZe Oh, I see, you're right, thank you for the info about `status` topic, somehow I completely missed that. 

Thank you very much for info about damped least squares. I'll look it up and maybe even try something. 

I've seen a few alghoritms for singularity handling, but nothing implemented or open-sourced. I'll have to dig deeper :) 

Thank you for your help! :+1:

closed 2021-08-18 by fzoric8

nbbrooks, 2021-08-18:
@fzoric8 Take a look at https://github.com/ros-planning/moveit2/pull/620 - I recommend using plotjuggler or rqt to stream the condition and status values while you are Cartesian servoing around.

Note that the status and condition topics are only published to when a non-zero Cartesian input command is being provided. e.g. if you let go of the joystick it won't tell you if the arm is currently at rest inside the singularity thresholds