moveit/moveit #2619: melodic-devel CI fails because of moveit_resources

v4hn, 2021-04-26:
CI for `melodic-devel` is broken since multi-planner support was merged in moveit_resources.

C&P from @jschleicher [in this post](https://github.com/ros-planning/moveit/pull/2567#issuecomment-815810616):
<<<
Finally I could trace the build failure down to missing _capabilities_:

* https://github.com/ros-planning/moveit_resources/pull/47 no longer loads `pilz_industrial_motion_planner/MoveGroupSequenceAction` as `move_group/capabilities` where melodic still expects it to be
* `default_planner_request_adapters/ResolveConstraintFrames`  is since ros-planning/moveit_resources#47 required in the panda config but not available on the melodic branch

Actually this seems to be rather unrelated to the GithubAction migration and should happen on travis as well, once it is triggered for melodic. What would you think about branching melodic_devel on moveit_resources without the new features? Or should pipeline-selection and subframes be backported to melodic?

With ros-planning/moveit_resources@93cb03f3 the tests pass for me.
\>\>\>

v4hn, 2021-04-26:
We explicitly decided to keep subframes out of melodic: https://github.com/ros-planning/moveit_msgs/pull/72

pipeline-selection is very useful, but also a message change & request to update user configurations.
Thus I don't think we should backport it either.

The missing `ResolveConstraintFrames` adapter reports a `ROS_ERROR`, but can be ignored (we told users before to do so).

Basically our options are to
- fork the resources (we avoided that up to now).
- make resources compatible with the old and new pipeline schema.

I just rejected the second option for me personally because the resources should be as clean as possible and uploading a lot of parameters a second time in a parent namespace is a lot of junk on the parameter server.

I'll push a new `melodic-devel` branch for `moveit_resources` in a bit to address the problems.

rhaschke, 2021-04-26:
Although the new planning pipeline code can work with old configs, it obviously doesn't work the other way around: the old code cannot find its configuration when using the new config. I agree with Michael, that the configs should reflect best practice, i.e. reverting isn't an option. I agree to fork `moveit_resources`.

v4hn, 2021-04-26:
I went ahead and pushed [a melodic-devel branch](https://github.com/ros-planning/moveit_resources/tree/melodic-devel) for `moveit_resources`, created https://github.com/ros-planning/moveit_resources/pull/65 to drop melodic support in the `master` branch and [changed the pulled branch](https://github.com/ros-planning/moveit/pull/2567/commits/967f6f667acba2dba18b51e9a516555365776ed2) in @tylerjw's pr to get the `melodic-devel` branch in moveit to use github actions.

tylerjw, 2021-04-26:
Thank you for solving this issue.  The only thing I don't agree with is dropping melodic off the CI run for master on moveit_resources.  We still want to support users on melodic who build from source on master and therefore we should continue to test it on melodic.

rhaschke, 2021-05-06:
I think this is fixed via the new melodic releases of moveit_resources and moveit.

closed 2021-05-06 by rhaschke