moveit/moveit #2620: [Joystick Control Teleoperation] : moveit_joy node die

astroxyx, 2021-04-26:
hello everyone,I wanted to use our remote control in order to move my robot through Rviz / MoveIt.I have a node to receive the remote control data and publish the data.But when I test the Joystick Control Teleoperation,but the moveit_joy die(I didn't insert the handle,so Couldn't open joystick /dev/input/js0. is normal),but I don't konw why the moveit_joy die :

Is it a problem with my SRDF?

                 joy: [main]: joy_node: autorepeat_rate (40.000000 Hz) > 1/coalesce_interval (39.999999 Hz) does not make sense.
                   ~   Timing behavior is not well defined.
                 joy: [main]: Couldn't open joystick /dev/input/js0. Will retry every second.
          moveit_joy: [core::RobotModel::buildModel]: Loading robot model 'engineer'...
          moveit_joy: [core::JointModel* moveit::core::RobotModel::constructJointModel]: No root/virtual joint specified in SRDF
                   ~  . Assuming fixed joint
          moveit_joy: [core::LinkModel* moveit::core::RobotModel::constructLinkModel]: Link back_left_wheel has visual geometry 
                   ~  but no collision geometry. Collision geometry will be left empty. Fix your URDF file by explicitly specify
                   ~  ing collision geometry.
          moveit_joy: [core::LinkModel* moveit::core::RobotModel::constructLinkModel]: Link back_right_wheel has visual geometry
                   ~   but no collision geometry. Collision geometry will be left empty. Fix your URDF file by explicitly specif
                   ~  ying collision geometry.
          moveit_joy: [core::LinkModel* moveit::core::RobotModel::constructLinkModel]: Link front_left_wheel has visual geometry
                   ~   but no collision geometry. Collision geometry will be left empty. Fix your URDF file by explicitly specif
                   ~  ying collision geometry.
          moveit_joy: [core::LinkModel* moveit::core::RobotModel::constructLinkModel]: Link front_right_wheel has visual geometr
                   ~  y but no collision geometry. Collision geometry will be left empty. Fix your URDF file by explicitly speci
                   ~  fying collision geometry.
          moveit_joy: [initialize]: Using empty link_prefix.
          moveit_joy: engineer_arm ['/rviz/moveit/move_marker/goal_palm']
          moveit_joy: Traceback (most recent call last):
          moveit_joy:   File "/opt/ros/noetic/lib/moveit_ros_visualization/moveit_joy.py", line 42, in <module>
          moveit_joy:     app = MoveitJoy()
          moveit_joy:   File "/opt/ros/noetic/lib/python3/dist-packages/moveit_ros_visualization/moveitjoy_module.py", line 538,
                   ~   in __init__
          moveit_joy:     self.parseSRDF()
          moveit_joy:   File "/opt/ros/noetic/lib/python3/dist-packages/moveit_ros_visualization/moveitjoy_module.py", line 516,
                   ~   in parseSRDF
          moveit_joy:     for name in planning_groups.keys():
          moveit_joy: RuntimeError: dictionary changed size during iteration
          moveit_joy: moveit_joy exited with status 1
            [rosmon]: Shutting down...
                 joy: joy exited with status 0

AndyZe, 2021-04-26:
It looks like your joystick device couldn't be found:  `Couldn't open joystick /dev/input/js0. Will retry every second.`

You can navigate to the `/dev/input` folder and check what devices are available like this:

```
cd /dev/input
ls
```

Closing because this appears not to be a MoveIt issue.

closed 2021-04-26 by AndyZe

astroxyx, 2021-04-27:
thanks for your answer ,I try to insert my handle,but moveit_joy node still die,I can see the handle data in /joy topic.

               joy: [main]: joy_node: autorepeat_rate (40.000000 Hz) > 1/coalesce_interval (39.999999 Hz) does not make sense.
                   ~   Timing behavior is not well defined.
                 joy: [main]: Couldn't set gain on joystick force feedback: Bad file descriptor
                 joy: [main]: Opened joystick: /dev/input/js0 (Wireless Controller). deadzone_: 0.200000.
          moveit_joy: [core::RobotModel::buildModel]: Loading robot model 'engineer'...
          moveit_joy: [core::JointModel* moveit::core::RobotModel::constructJointModel]: No root/virtual joint specified in SRDF
                   ~  . Assuming fixed joint
          moveit_joy: [core::LinkModel* moveit::core::RobotModel::constructLinkModel]: Link back_left_wheel has visual geometry 
                   ~  but no collision geometry. Collision geometry will be left empty. Fix your URDF file by explicitly specify
                   ~  ing collision geometry.
          moveit_joy: [core::LinkModel* moveit::core::RobotModel::constructLinkModel]: Link back_right_wheel has visual geometry
                   ~   but no collision geometry. Collision geometry will be left empty. Fix your URDF file by explicitly specif
                   ~  ying collision geometry.
          moveit_joy: [core::LinkModel* moveit::core::RobotModel::constructLinkModel]: Link front_left_wheel has visual geometry
                   ~   but no collision geometry. Collision geometry will be left empty. Fix your URDF file by explicitly specif
                   ~  ying collision geometry.
          moveit_joy: [core::LinkModel* moveit::core::RobotModel::constructLinkModel]: Link front_right_wheel has visual geometr
                   ~  y but no collision geometry. Collision geometry will be left empty. Fix your URDF file by explicitly speci
                   ~  fying collision geometry.
          moveit_joy: [initialize]: Using empty link_prefix.
          moveit_joy: engineer_arm ['/rviz/moveit/move_marker/goal_palm']
          moveit_joy: Traceback (most recent call last):
          moveit_joy:   File "/opt/ros/noetic/lib/moveit_ros_visualization/moveit_joy.py", line 42, in <module>
          moveit_joy:     app = MoveitJoy()
          moveit_joy:   File "/opt/ros/noetic/lib/python3/dist-packages/moveit_ros_visualization/moveitjoy_module.py", line 538,
                   ~   in __init__
          moveit_joy:     self.parseSRDF()
          moveit_joy:   File "/opt/ros/noetic/lib/python3/dist-packages/moveit_ros_visualization/moveitjoy_module.py", line 516,
                   ~   in parseSRDF
          moveit_joy:     for name in planning_groups.keys():
          moveit_joy: RuntimeError: dictionary changed size during iteration
          moveit_joy: moveit_joy exited with status 1
            [rosmon]: Shutting down...
                 joy: joy exited with status 0

astroxyx, 2021-04-27:
Dictionary elements cannot be modified while traversing,but I check the code where cause the error,it seems to delete elements.
```
for name in planning_groups.keys():
            if len(planning_groups[name]) == 0:
                del planning_groups[name]
            else:
                print(name, planning_groups[name])
```