moveit/moveit #2623: PlanningSceneInterface not adding objects in some cases

JGioia, 2021-04-26:
### Description

I am getting consistent issues with adding objects to a PlanningSceneInterface. I have found a workaround, however I don't think it is working how it is supposed to.

### Your environment
* ROS Distro: Melodic
* OS Version: Ubuntu 18.04
* Binary: 1.0.7

### Steps to reproduce
Here's the python code that causes the issue. It may not make sense why I would do this, however this was the best I could condense what is happening across my code.

```
rospy.sleep(2)
scene2 = moveit_commander.PlanningSceneInterface()
pose = PoseStamped()
pose.header.frame_id = "base_link"
scene2.add_box("test", pose, size=(4, 4, 0.1))
rospy.sleep(2)
scene2.remove_world_object("test")
```

This fixes the issue.
```
scene = moveit_commander.PlanningSceneInterface()
rospy.sleep(2)
scene2 = moveit_commander.PlanningSceneInterface()
pose = PoseStamped()
pose.header.frame_id = "base_link"
scene2.add_box("test", pose, size=(4, 4, 0.1))
rospy.sleep(2)
scene2.remove_world_object("test")
```

This causes the same issue.
```
scene = moveit_commander.PlanningSceneInterface()
scene2 = moveit_commander.PlanningSceneInterface()
pose = PoseStamped()
pose.header.frame_id = "base_link"
scene2.add_box("test", pose, size=(4, 4, 0.1))
rospy.sleep(2)
scene2.remove_world_object("test")
```

### Expected behavior
I expect a box to be added to the planning scene in all three situations. I expect to be able to see it with the PlanningScene viewer in RViz.

### Actual behavior
Only the second block of code adds the box to the planning scene. The others do not crash or do anything to indicate that they did not work. 

### Backtrace or Console output

[Here's](https://gist.github.com/JGioia/b4d5b8be7abe21490d7bc172340282fe) the console output for the working version.
[Here's](https://gist.github.com/JGioia/06627620524afff0764d5db929f34bc8) the console output for the not working versions (it's the same for both of them).

rhaschke, 2021-04-26:
First, note the console warning:
```
ros.moveit_ros_planning.planning_scene_monitor: Unable to transform object from frame 'world' to planning frame 'base_link' (Could not find a connection between 'base_link' and 'world' because they are not part of the same tree.Tf has two or more unconnected trees.)
```
indicating that you use the wrong `frame_id`. 

Given your code, adding and immediately removing the object again, I wouldn't expect to see an object popping up in the scene!

The problem with the world objects not showing up is usually a timing problem: Messages are sent via topics in an asynchronous fashion. If you quit your python program too early (not waiting and thus giving ROS a chance to deliver the message), the message will just be lost. However, you do already have sleeps in there. 
To get more reliable behavior we added a synchronous (service) interface as well:
https://github.com/ros-planning/moveit/blob/efd20cc440cff9a20b481fe9394c21310365a4a2/moveit_commander/src/moveit_commander/planning_scene_interface.py#L60

Just initialize your PS like this:
```python
scene = moveit_commander.PlanningSceneInterface(synchronous=True)
```

JGioia, 2021-04-27:
Thanks for the help! Adding `synchronous=True` fixed my problem. It makes a lot more sense now why I was having these issues. Do you know if I missed something when learning about this that could have helped me avoid my problem? I feel like this could be a pretty common issue for beginners. Also can I ask why `synchronous=True` isn't the default behavior? It seems more useful for adding and removing objects, but I'm not super familiar MoveIt so I'm probably missing something.

closed 2021-04-27 by JGioia

rhaschke, 2021-04-27:
I guess, we kept the asynchronous behaviour as the default, to not risk breaking existing setups. However, you are right, we might rethink this decision. @v4hn and @felixvd what do you think?
We highly welcome any improvements to the docs, e.g. in https://github.com/ros-planning/moveit_tutorials.

felixvd, 2021-04-27:
I think it's a good idea, I use it synchronously as well.

v4hn, 2021-04-27:
Synchronous is mostly not the default for historical reasons, but there's one more aspect to it.

I implemented the apply* interfaces (the synchronous version) because of the others are not reliable.
But they only work with the `move_group` node and not with custom `PlanningSceneMonitor` instances.
Because a number of maintainers advertised for using the C++ classes (and thus PSM) directly, last time we discussed this, it was enough reason to not make it the default.

For my own projects I haven't used the asynchronous interfaces in a long time and I approve anyone who wants to change the default in `master`.