moveit/moveit #2624: Initial sync of PlanningSceneDisplay missing

simonschmeisser, 2021-04-27:
### Description
When I start my software it misses the initial synchronization between the PlanningSceneMonitor and PlanningSceneDisplay, best visible in the RobotState being all zero. This repeats when I add  another PlanningSceneDisplay in a separate rviz.

[image]

I just switched from `melodic-devel` to `master` so maybe I need to migrate something?

### Your environment
* ROS Distro: Melodic
* OS Version: Ubuntu 18.04
* Source or Binary build?
* If binary, which release version? current ROS melodic for base system
* If source, which branch? MoveIt  master (+ octomap branch #2596 ) , rviz & co current master/noetic-devel

### Backtrace or Console output

When toggling on the PlanningSceneDisplay I get the following including an error about an empty JointStateMessage

```
[ INFO] [1619533995.786134857]: Stopping planning scene monitor
[ INFO] [1619533999.020052495]: Loading robot model 'world'...                                                                                                                                                                                                                
[ INFO] [1619533999.067777623]: MoveItOPWKinematicsPlugin initializing                                                                                                                                                                                                        
[ INFO] [1619533999.067853756]: Dimension planning group 'manipulator': 6. Active Joints Models: 6. Mimic Joint Models: 0
[ INFO] [1619533999.067917364]: Getting kinematic parameters from parameter server.
[ INFO] [1619533999.073403590]: Loaded parameters for ik solver:
Distances: [0 -0.05 0 0.35 0.31 0.335 0.085]
Offsets = [0 0 0 0 0 0 ]
Sign_corrections = [1 1 1 1 1 1 ]
[ INFO] [1619533999.116072269]: Starting planning scene monitor
[ INFO] [1619533999.119336132]: Listening to '/move_group/monitored_planning_scene'
[ERROR] [1619533999.126463701]: Found empty JointState message

```

rhaschke, 2021-04-27:
Hm. I cannot confirm this issue. Can you debug this in more detail, please? Did you try without #2596 as well?

simonschmeisser, 2021-04-27:
Sure, I just checked with MoveIt master and it shows the same issue. Should I be using rviz melodic-devel on melodic or is noetic-devel fine?

simonschmeisser, 2021-04-27:
So it seems nothing gets published on the PlanningScene topics /move_group/monitored_planning_scene` and `/planning_scene` because the setup is "at rest". `/joint_states` do not fluctuate due to being simulated and nothing else in the scene changes either

simonschmeisser, 2021-04-27:
Oh, found it, when I created my own MoveItcpp clone long ago I copied the getPlanningSceneService capability into my code and that is outdated now. #1424 basically. So I'll ifdef that to use `providePlanningSceneService` on noetic instead, which works

closed 2021-04-27 by simonschmeisser