moveit/moveit #2631: MoveIt Dynamic Planning Scene Update Problem During Robot Execution

panagelak, 2021-04-29:
Hello MoveIt Community!!

I have a problem with MoveIt's Planning scene not taking into consideration new updates/collision objects while the robot is executing the plan (if i update the scene before the robot plans the trajectory it takes them into consideration)

The idea of my project is to detect 3d bodypart keypoints of an operator and then cast them into MoveIt's planning scene in order for the robot to not collide with the operator as the robot execute its plan and the operator moves, in order to make an application where robot and operator can coexist.

Below we can see how the robot plans a path around an obstacle that has been added to the planning scene before we send a goal to the MoveIt action server (which calculates the plan and executes it)

[image]

Now in order to test for example a sudden movement of the operator, i have made a node that listens to the robot end-effector transform frame and when it starts to move, commands the operator to lift his hand and block the path

[image]

As we can see the robot doesnt take into consideration the updated planning scene (while the robot executes the plan) and as a result collides with the operator (even though it had a lot of time to stop). This also happens if we spawn a collision obstacle while the robot is moving.

[image]

Please note that i dont intent to plan around a moving obstacle, a cancelation of the execution when a collision obstacle intersect the plan and replanning will suffice

I could make a custom solution where i subscribe to the calculated plan and when a collision obstacle intersect to send a cancellation to the moveit action server in order to stop the robot, but this is a lot of work and i am not sure how fast it will work.

Furthermore i know that with MoveIt2 we can expect better performance with dynamic scene updates i just want to know if it's possible to stop the robot when a new collision object intersect the plan suddenly.

Thanks a lot for your feedback, i am looking forward to your inputs!!!

Here is a video of the problem

https://user-images.githubusercontent.com/9826483/116536450-9b3b4380-a8ed-11eb-8fdb-8de2068ac7af.mp4

v4hn, 2021-04-29:
Hi @panagelak , I love the project! :+1: 

This functionality should actually work out of the box if you add the human parts, i.e. spheres, as collision objects to the planning scene.
If that's what you actually do and you plan&execute the trajectories through the `MoveGroup` action, [this method](https://github.com/ros-planning/moveit/blob/master/moveit_ros/planning/plan_execution/src/plan_execution.cpp#L271) should detect the collision and abort.

Looking into this I found at least one issue, because [the ExecuteTrajectoryAction actually bypasses the `plan_execution_` and directly pushes trajectories to trajectory execution](https://github.com/ros-planning/moveit/blob/master/moveit_ros/move_group/src/default_capabilities/execute_trajectory_action_capability.cpp#L97) which does not include this monitoring.

panagelak, 2021-04-29:
Hi @v4hn ,

Thanks for the quick  response!

Perhaps i was a little unclear since i don't use the default action server that Moveit provides but a custom one that wraps the Moveit's functionality/api based on my needs 

the moveit api i use for execution is from the tutorials e.g 

#include <moveit/move_group_interface/move_group_interface.h>
#include <moveit/planning_interface/planning_interface.h>
moveit::planning_interface::MoveGroupInterfacePtr move_group_;
moveit::planning_interface::MoveGroupInterface::Plan plan_;
move_group_->execute(plan_) 
which i have put in a different thread in order to have feedback, but i guess i could have just used
move_group->asyncExecute(plan_)

i cant find from move_group (autocomplete) a function like **executeAndMonitor**  
i think this functionallity should be inside move_group_

Perhaps you can guide me on which headers to include and what object has this functionallity since i am a little confused e.g

#include <moveit/plan_execution/plan_execution.h>
plan_execution::PlanExecution obj;
obj.executeAndMonitort(plan, true) 
(but i cant use the plan i use for move_group it wants an ExecutableMotionPlan)

Thanks a lot for your feedback!

v4hn, 2021-04-29:
The `MoveGroupInterface` c++ class is actually just a wrapper for simpler access to the ROS actions.
We provide it because they have quite a few parameters and only support ROS messages, but you are using them nonetheless.

Especially the `MoveGroupInterface::execute` method [is just a wrapper calling the `ExecuteTrajectory` action](https://github.com/ros-planning/moveit/blob/master/moveit_ros/planning_interface/move_group_interface/src/move_group_interface.cpp#L875).
So you basically ran into the exact problem I mentioned above:

Because the ExecuteTrajectory action capability bypasses the `plan_execution_` interfaces, it does not actually monitor the execution (and thus does not abort if a collision will occur).

If you use the `MoveGroupInterface::move` method (where you never see the trajectory that will execute because you only specify the planning goal) the same scenario should work unless there is another bug somewhere else...

Your options are the following:
- implement monitoring of collisions yourself and abort from the outside by preempting the action (basically what you described above)
- rework the ExecuteTrajectory action to use `plan_execution_`. I did not look into whether that's possible while keeping the current `ExecuteTrajectoryAction` interface
- add a new [move_group capability](https://github.com/ros-planning/moveit/tree/master/moveit_ros/move_group/src/default_capabilities) (documentation [here](https://moveit.ros.org/documentation/plugins/#movegroupcapability) and [here](https://moveit.ros.org/documentation/concepts/)) based on a mixture of `execute_trajectory_action_capability` and `move_action_capability`, called something like `execute_monitored_trajectory_action_capability`

We would welcome the second or third option upstream.

panagelak, 2021-04-29:
Thanks for your feedback @v4hn 

I can understand better now how things work

Indeed after setting the pose target + allowRepllaning(true)  and without planning using the 
move_group->move() it checks the new collision object , stops and replans around it

although i can still plan in order to visualize the trajectory with the current snapshot of the planning scene and abort my action server if it doesn't find a plan at the start

I will make some research with the other methods you described me or
(possibly using the MoveitCpp directly in order to avoid nested action servers?)

Thanks again!!

You can close this issue if you want

Best Regards!

v4hn, 2021-04-29:
You are welcome. Thank you for getting in touch and pointing to a serious problem.

> Indeed after setting the pose target + allowRepllaning(true)  and without planning using the
> move_group->move() it checks the new collision object , stops and replans around it

That's good to hear.

> although i can still plan in order to visualize the trajectory with the current snapshot of the planning scene and abort my action server if it doesn't find a plan at the start

Yes, but the trajectory you will see in your own plan is not the trajectory the robot will move along if you use `move()`.

> possibly using the MoveItCpp directly in order to avoid nested action servers?

Whether that makes sense depends largely on your exact use-case and proficiency in MoveIt.
However, I just discovered that MoveItCpp does not even instantiate `PlanExecution` for you.
That's a serious shortcoming as well!
Of course, you can just instantiate it yourself with the classes provided in MoveItCpp.

> You can close this issue if you want

I pointed out two issues here that should be resolved before this is closed:

- MoveItCpp should provide a normal `PlanExecution` object as well, which should be encouraged over the normal `TrajectoryExecutionManager`. Its `executeAndMonitor` method provides the collision monitoring
- The `ExecuteTrajectory` action should definitely make use of the monitoring. I consider this shortcoming a serious safety issue.

We might actually want to consider monitoring in the TEM, though I'm not convinced that makes the code any better.