moveit/moveit #2636: flaky tests for MoveGroupInterface/PlanningSceneInterface

v4hn, 2021-04-30:
As observed [in multiple PRs](https://github.com/ros-planning/moveit/pull/2630#issuecomment-829944030), a test in `moveit_ros_planning` is flaky.

Looking further into it, it's not quite so new, but we haven't seen it fail before.

[These EXPECTs are all intrinsically flaky](https://github.com/ros-planning/moveit/blob/master/moveit_ros/planning_interface/test/move_group_interface_cpp_test.cpp#L245-L255) because they use the asynchronous publishers to update the planning scene, but afterwards use the synchronous service to check whether the messages were processed successfully.
Especially because there is no sleep after publishing there are no guarantees whatsoever.

Basically we have these options:
- remove/disable the tests
- add .5 second sleeps will probably work (though it's not a sound solution)
- add a PlanningSceneMonitor and wait for the updates through it.

rhaschke, 2021-04-30:
I was hesitant to remove the tests, but actually, this test isn't about the PlanningScene interface! It should check for planning with PathConstraints (only). I hope we do have PlanningScene tests somewhere else and thus agree to just remove the checks here.

v4hn, 2021-04-30:
I don't agree though! We don't have other tests for them! You are right though in that they should be a separate test.
I'm working on a solution using a PSM.

PR #2640 (2021-05-03): planning_interface: synchronize async interfaces in test
It's always messy to synchronize these interfaces, so I don't use them in my own projects.
However, these are the only tests we have for the async method calls at the moment, so we should make them run ~~reliably~~ as reliably as possible.
There is no sense in running these updates asynchronously for testing the interfaces, so I added a (somewhat cumbersome) feedback channel to wait for the triggered updates.
I'm not aware of a better solution for this that does not consist of repeatedly polling the service.

Hopefully this closes #2636  [... cut]
closed 2021-05-03 by rhaschke