moveit/moveit #2641: move_group.plan() does not show planned path in rviz anymore

UmutUz, 2021-05-02:
When I call the move_group.plan() function usually I would see the plan in rviz the same way I would hit plan in the MotionPlanner plugin. Then I would call execute on the move_group and I can still execute the path. Well, it somehow doesn't work anymore. I didn't do any updates or changes to the code either so I am beyond confused.

ROS Melodic
Ubuntu 18.04
```

    def plan_path_from_joint_goal(self, robot, goal_joint_goal, movegroup):
        
        movegroup.set_start_state(self.robot.get_current_state())
        movegroup.set_joint_value_target(goal_joint_goal)
        motion_plan = movegroup.plan(goal_joint_goal)
        return motion_plan

    planning_ik_right = self.plan_path_from_joint_goal(self.robot,ik_resp_right, self.move_group)
    planning_ik_left = self.plan_path_from_joint_goal(self.robot2,ik_resp_left, self.move_group_left)
```

Should plan the motion plan and display it in rviz before I execute it. I can still execute the plan but I have no idea if the path is any good.

felixvd, 2021-05-02:
This is due to #2578 . The topics are not published to `move_group/display_planned_path` at the moment. As a workaround, you can click on the `display_planned_path` topic in the Rviz displays and select the one of your planner (for OMPL, it should be `/move_group/planning_pipelines/ompl/display_planned_path`, but it should appear in the dropdown list once you have planned a path).

I will close this since the problem is tracked already.

closed 2021-05-02 by felixvd

UmutUz, 2021-05-04:
Okay, so I got to try your temporary "fix" and it didn't work, so I just use the "Robot Visual" tick for now which does pretty much the same as showing the plan except it's really slow but I can't really complain, as long as it works. Thanks though!