moveit/moveit #2644: PointCloudOctomapUpdater fails to update the octomap in realtime.

PatrickVibild, 2021-05-03:
### Description

Delay on octomap generated by PointCloudOctomapUpdater.

### Your environment
* ROS Distro: Melodic
* OS Version: Ubuntu 18.04

### Steps to reproduce
Sensor manager.
```
<launch>

  <param name="octomap_frame" type="string" value="odom_combined" />
  <param name="octomap_resolution" type="double" value="0.06" />
  <param name="octomap_latch" type="bool" value="false"/>

  <param name="octomap_sensor_model/min" value="0.99"/>
  <param name="octomap_sensor_model/max" value="0.01"/>

  <param name="octomap_sensor_model/hit" type="double" value="0.99"/>
  <param name="octomap_sensor_model/miss" type="double" value="0.01"/>

  <param name="max_range" type="double" value="5.0" />

  <!-- Load the robot specific sensor manager; this sets the moveit_sensor_manager ROS parameter -->
  <arg name="moveit_sensor_manager" default="edo" />
  <include file="$(find edo_moveit_gripper)/launch/$(arg moveit_sensor_manager)_moveit_sensor_manager.launch.xml" />

</launch>
```

PointCloudOctomapUpdater declaration yaml.

```
sensors:
  - sensor_plugin: occupancy_map_monitor/PointCloudOctomapUpdater
    point_cloud_topic: /camera_side/depth/color/points
    queue_size: 5
    max_range: 3.0
    point_subsample: 1
    shadow_threshold: 0.2
    padding_scale: 4.0
    padding_offset: 0.03
    max_update_rate: 5
    filtered_cloud_topic: filtered_cloud
    min_range: 0.3

```

### Expected behaviour
Reading documentation changing the parameters hit/miss and min/max of the sendor_model should make a dynamic octomap.

### Actual behaviour
No changes when modifying the octomap server parameters in the sensor manager. The generated map is not responsive.

v4hn, 2021-05-03:
I'm not sure you can call this a bug.
The PointCloudOctomapUpdater is definitely very slow, though better hardware helps a bit.

If your point clouds originate from an RGB-D camera, you can accelerate a lot using the [DepthImageOctomapUpdater](https://github.com/ros-planning/moveit/tree/master/moveit_ros/perception/depth_image_octomap_updater) instead.

Integrating a project like [GPU-Voxels](https://gpu-voxels.org/) for faster voxelgrid handling could be very useful, but has not been done ttbomk.

PatrickVibild, 2021-05-03:
Thanks, I changed the sensor plugin and became way more reactive.

I was wondering if there is anyway defined what are the parameters for each sensor. The documentation is completely empty regarding each of the yaml parameters.

```
        image_topic: This specifies the topic to listen on for a depth image.
        queue_size: he number of images to queue up
        near_clipping_plane_distance:
        far_clipping_plane_distance:
        shadow_threshold:
        padding_offset: The size of the padding (in cm).
        padding_scale:
        filtered_cloud_topic: The topic on which the filtered cloud will be published (mainly for debugging). The filtering cloud is the resultant cloud after self-filtering has been performed.
```

v4hn, 2021-05-03:
I'm not sure where you copied that block, but it's somewhat outdated (and wrong).
Here's the current version: https://ros-planning.github.io/moveit_tutorials/doc/perception_pipeline/perception_pipeline_tutorial.html
It's not much better though.

If you have suggestions for further improvements, please create a corresponding pull-request :-)

simonschmeisser, 2021-05-03:
@v4hn do you know what makes the `DepthImageOctomapUpdater` faster? Is it parallelized? Or is there a fundamental reason for it to be more efficient?

v4hn, 2021-05-03:
Aside from the fact that point clouds are *huge* compared to their respective depth images (and thus there is significantly less overhead in ROS communication), the [`DepthImageOctomapUpdate` is based on the OpenGL-accelerated `mesh_filter`](https://github.com/ros-planning/moveit/tree/master/moveit_ros/perception/mesh_filter), but [`PointCloudOctomapUpdater` uses the `geometric_shape`-based `point_containment_filter`](https://github.com/ros-planning/moveit/tree/master/moveit_ros/perception/point_containment_filter)

PatrickVibild, 2021-05-04:
@v4hn Thanks a lot for the information, I can see that this was not clearly a bug and I miss used the bug.

Before I close this, could you suggest me any place to get any support community about moveit? tried several times in stack overflow and ros answers without getting any answers there. 

I managed to make my dynamic on the octomap. Hardware is not an issue, running it on a server with a 12 cores Xeon and 2080TI. My issue now is that the robotic arm is colliding with the shadow that it projects when it moves. So when I move the arm it stops every 2s due to detecting a collision. I increased the padding for the robot, but then I notices that it will physically collide with objects that are close to the robot when starting a joint movement.

Again dont want to miss use the bug section will close it, but never felt to lucky to have responses to my moveit questions.

v4hn, 2021-05-04:
No problem.

> Before I close this, could you suggest me any place to get any support community about moveit?
> tried several times in stack overflow and ros answers without getting any answers there.

I don't know of anyone who replies on stackoverflow. We always refer users to http://answers.ros.org
If you posted there but didn't get an answer it's likely because nobody took the time.
You might for example raise awareness for a question there by joining [MoveIt's Discord server](https://discord.com/invite/RrySut8).
Also see [support](https://moveit.ros.org/support/).
Please try to make sure anything you discuss in chats ends up online somewhere (usually ros answers), so we don't get the same questions over and over again.

> I managed to make my dynamic on the octomap.

Congrats.

> My issue now is that the robotic arm is colliding with the shadow that it projects when it moves.
> So when I move the arm it stops every 2s due to detecting a collision.

I expect you mean points on the arm that are not correctly self-filtered.
That can happen for a few reasons:
- the calibration is not accurate enough
- the padding (offset + scale) does not correctly include sensor artifacts around the robot
- the camera acquisition time in the received sensor messages is not correct and the robot is actually in a different position than where it should be according to `/joint_states`