moveit/moveit #2648: file conflict when upgrading moveit_ros_planning

christian-rauch, 2021-05-05:
### Description

Updating `moveit_ros_planning` (a.k.a. `ros-noetic-moveit-ros-planning`) from 1.1.0 to 1.1.2 breaks apt.

### Your environment
- Ubuntu 20.04, Noetic
- `ros-noetic-moveit-ros-planning` `1.1.0-1focal.20210302.142257`

### Steps to reproduce
I am not sure if this is reproducible.
1. install `ros-noetic-moveit-ros-planning` and `ros-noetic-moveit-ros-planning-interface` version `1.1.0-1focal.20210302`
2. `sudo apt upgrade` to version `1.1.2-1focal.20210424`

### Expected behaviour
Upgrade without issues.

### Actual behaviour
apt fails and interrupts all system updates

### Backtrace or Console output
```
Unpacking ros-noetic-moveit-ros-planning (1.1.2-1focal.20210424.003930) over (1.1.0-1focal.20210302.142257) ...
dpkg: error processing archive /tmp/apt-dpkg-install-Mi1pqr/127-ros-noetic-moveit-ros-planning_1.1.2-1focal.20210424.003930_amd64.deb (--unpack):
 trying to overwrite '/opt/ros/noetic/include/moveit/moveit_cpp/moveit_cpp.h', which is also in package ros-noetic-moveit-ros-planning-interface 1.1.0-1focal.20210302.150335
dpkg-deb: error: paste subprocess was killed by signal (Broken pipe)
```

This is probably because file `/opt/ros/noetic/include/moveit/moveit_cpp/moveit_cpp.h` moved between packages. But if packages are not upgraded in the correct order, this will lead to conflicts. Unfortunately, this will interrupt all system updates until resolved manually.

MoveIt has the tendency to break API/ABI during stable releases, but please don't break the system update.

gavanderhoorn, 2021-05-05:
Could it be your package repository / mirror is in an inconsistent state?

`packages.ros.org` does not have `ros-noetic-moveit-ros-planning-interface` with version `1.1.0`: [packages.ros.org/ros/ubuntu/pool/main/r/ros-noetic-moveit-ros-planning-interface](http://packages.ros.org/ros/ubuntu/pool/main/r/ros-noetic-moveit-ros-planning-interface/).

```
    ros-noetic-moveit-ros-planning-interface-dbgsym_1.1.2-1buster.20210424.075542_amd64.deb | 2021-04-24 08:01 | 17M |  
  | ros-noetic-moveit-ros-planning-interface-dbgsym_1.1.2-1buster.20210426.092354_arm64.deb | 2021-04-26 10:09 | 17M |  
  | ros-noetic-moveit-ros-planning-interface-dbgsym_1.1.2-1focal.20210424.015512_armhf.deb | 2021-04-24 02:03 | 18M |  
  | ros-noetic-moveit-ros-planning-interface-dbgsym_1.1.2-1focal.20210424.063613_amd64.deb | 2021-04-24 06:45 | 18M |  
  | ros-noetic-moveit-ros-planning-interface-dbgsym_1.1.2-1focal.20210425.180410_arm64.deb | 2021-04-25 18:40 | 18M |  
  | ros-noetic-moveit-ros-planning-interface_1.1.2-1buster.20210424.075542_amd64.deb | 2021-04-24 08:01 | 497K |  
  | ros-noetic-moveit-ros-planning-interface_1.1.2-1buster.20210426.092354_arm64.deb | 2021-04-26 10:09 | 460K |  
  | ros-noetic-moveit-ros-planning-interface_1.1.2-1buster.debian.tar.xz | 2021-04-14 20:13 | 7.6K |  
  | ros-noetic-moveit-ros-planning-interface_1.1.2-1buster.dsc | 2021-04-14 20:13 | 1.7K |  
  | ros-noetic-moveit-ros-planning-interface_1.1.2-1focal.20210424.015512_armhf.deb | 2021-04-24 02:03 | 455K |  
  | ros-noetic-moveit-ros-planning-interface_1.1.2-1focal.20210424.063613_amd64.deb | 2021-04-24 06:45 | 522K |  
  | ros-noetic-moveit-ros-planning-interface_1.1.2-1focal.20210425.180410_arm64.deb | 2021-04-25 18:40 | 481K |  
  | ros-noetic-moveit-ros-planning-interface_1.1.2-1focal.debian.tar.xz | 2021-04-14 20:09 | 7.6K |  
  | ros-noetic-moveit-ros-planning-interface_1.1.2-1focal.dsc | 2021-04-14 20:09 | 1.7K |  
  | ros-noetic-moveit-ros-planning-interface_1.1.2.orig.tar.gz | 2021-04-14 20:09 | 63K |  
```

christian-rauch, 2021-05-05:
I am using the official repo:
```
$ cat /etc/apt/sources.list.d/ros1-latest.list
deb http://packages.ros.org/ros/ubuntu focal main
```

The link you provided only has a single version (1.1.2) but I am pretty sure there were earlier versions before this. Are packages with older versions purged?

rhaschke, 2021-05-05:
`moveit_cpp.h` moved from package `ros_planning_interface` into `ros_planning`. However, the old `ros_planning_interface` package isn't purged before the new `ros_planning` is installed, because the former depends on the latter.
I'm afraid you need to explicitly purge the old `moveit_*` packages _before_ (re)installing the new ones. 
If you see another option, I'm curious to learn about it.

gavanderhoorn, 2021-05-05:
@christian-rauch wrote:
> Are packages with older versions purged?

The repository infrastructure used by OR for the ROS package repositories does not support more than a single version of a package at a time.

So AFAIK: yes, older versions are purged.

christian-rauch, 2021-05-05:
> If you see another option, I'm curious to learn about it.

You can declare conflicts between packages and versions (https://www.debian.org/doc/debian-policy/ch-relationships.html) so that apt figures out the correct order in which to update packages. But I think bloom is not supporting this.

Another option would be to not break these things in the first place :-) If you move files between packages, there is always the chance to break apt. MoveIt broke API/ABI during a stable release before, but breaking apt is especially annoying since it blocks all system updates and may leave it in an inconsistent state.

My opinion is that these breaking changes should not be done within a stable ROS release. They should be done before a new ROS distribution is released because then packages will have different names and there will be no conflicts.

v4hn, 2021-05-10:
> MoveIt has the tendency to break API/ABI during stable releases, but please don't break the system update.

We don't have the tendency, we *explicitly* bump sonames because any patch to MoveIt is really likely to break ABI and we still provide backports and releases in existing ROS distributions.

I wholeheartedly agree though! This is a serious problem and it should not have happened!
@henningkayser obviously did not foresee this issue.
Aside from putting blame on the entire team for letting it happen, I don't see a way to fix this though if bloom does not provide any way to add this deb package information during the release.
If we move the files back, users who went through your reported issue will be affected *again*.
The only potential fix I could imagine would be to have a quick followup release (we actually have 1.1.3 pending in shadow-fixed already by the way!) and manually add the conflict with `ros-noetic-moveit-ros-planning` version 1.1.0 to the deb file after bloom generated the files. I have no idea whether OSRF would support that process or who would have to act to get it done.
Do you see any alternative?

gavanderhoorn, 2021-05-10:
You can actually declare conflicts between packages in the manifest: [REP 149: Package Manifest Format Three Specification: Dependency tags: <conflict> (multiple)](https://ros.org/reps/rep-0149.html#conflict-multiple) (same for `replace`).

IIUC, that could have been used here.

christian-rauch, 2021-05-10:
> We don't have the tendency, we _explicitly_ bump sonames because any patch to MoveIt is really likely to break ABI and we still provide backports and releases in existing ROS distributions.

This does only prevent linking the wrong library version. It will still break API which means users have to change their code within a ROS release to adapt to all API changes. And this is what should not happen. If Ubuntu would replace Eigen3 with a hypothetical Eigen4 that changes the API / function signatures and pushes this to their users, they also wouldn't be happy to update their code.

Sorry if this sounds like a negative rant, but I think breaking API during a stable release is a serious issue as it will always mean users have to spend additional time on "fixing" their code after some dependency decided to change API. This is ok between ROS distributions, because the user can decide when to switch and spend time on adapting their code. But just changing the API during a release and pushing the change to users is not ok.

v4hn, 2021-05-12:
Let's keep this open until the fix is in the ROS package repository.

PR #2659 (2021-05-11): declare conflict after moving headers
... that install to the same location.

Following the suggestion from https://github.com/ros-planning/moveit/issues/2648#issuecomment-836647351

Fixup for eb3bf6fdb6ece5324e3d1a81fd5d2b2197804710

Fixes #2648.

### Description

Please explain the changes you made, including a reference to the related issue if applicable

### Checklist
- [ ] **Required by CI**: Code is auto formatted using [clang-format](http://moveit.ros.org/documentation/contributing/code)
- [ ] Extend the tutorials / documentation [reference](http://moveit.ros.org/documentation/contribut [... cut]
commit 01e594ea (2021-05-11): declare conflict after moving headers (#2659)

... that install to the same location.

Fixup for eb3bf6fdb6ece5324e3d1a81fd5d2b2197804710

Fixes #2648.