moveit/moveit #2649: auto_create_ikfast_moveit_plugin.sh fails due to sympy0.7.1 dependency not available from pip any more.

ags-dy, 2021-05-06:
### Description

I have used auto_create_ikfast_moveit_plugin.sh script in this repo to generate ikfast plugins in the past.  When I tried to run the same code today it failed during the build of the docker images because it failed to pip install sympy 0.7.1.  I assume this is due to further depreciation of python2. 

### Your environment
* ROS Distro: [Noetic]
* OS Version: e.g. Ubuntu 20.04
* Binary build 1.1.2-1focal.20210424.005508

### Steps to reproduce
```bash
$rosrun moveit_kinematics auto_create_ikfast_moveit_plugin.sh --iktype Transform6D panda_arm.urdf panda panda_link0 
```

### Expected behaviour
The ikfast docker image should install sympy

### Actual behaviour
pip install of sympy fails

### Backtrace or Console output
```bash
$rosrun moveit_kinematics auto_create_ikfast_moveit_plugin.sh --iktype Transform6D panda_arm.urdf panda panda_link0 panda_link8 
Converting urdf to Collada
Building docker image
Sending build context to Docker daemon   2.56kB
Step 1/3 : FROM personalrobotics/ros-openrave
 ---> 3470beb7a703
Step 2/3 : RUN apt-key adv --keyserver 'hkp://keyserver.ubuntu.com:80' --recv-key C1CF6E31E6BADE8868B172B4F42ED6FBAB17C654 &&     apt-key del 421C365BD9FF1F717815A3895523BAEEB01FA116 &&     apt-get update &&     apt-get install -y --no-install-recommends python-pip build-essential liblapack-dev ros-indigo-collada-urdf &&     apt-get clean && rm -rf /var/lib/apt/lists/*
 ---> Using cache
 ---> 691f6b13328c
Step 3/3 : RUN pip install sympy==0.7.1
 ---> Running in cd6fe1646158
Downloading/unpacking sympy==0.7.1
  Cannot fetch index base URL https://pypi.python.org/simple/
  Could not find any downloads that satisfy the requirement sympy==0.7.1
Cleaning up...
No distributions at all found for sympy==0.7.1
Storing debug log for failure in /root/.pip/pip.log
The command '/bin/sh -c pip install sympy==0.7.1' returned a non-zero code: 1
```

Thanks

ags-dy, 2021-05-06:
In the script: https://github.com/ros-planning/moveit/blob/master/moveit_kinematics/ikfast_kinematics_plugin/scripts/auto_create_ikfast_moveit_plugin.sh#L100

I replaced:
```bash
RUN pip install sympy==0.7.1
````
with:
```bash
RUN pip install git+https://github.com/sympy/sympy.git@sympy-0.7.1
```
and that appears to fix it.

gavanderhoorn, 2021-05-06:
The problem is not Python 2 deprecation, it's most likely the ancient version of `pip` used in the `personalrobotics/openrave` image.

`sympy` at version `0.7.1` is still available, even with Python 2:

```bash
docker run -it --rm ubuntu:bionic
apt update && apt install --no-install-recommends -y python-pip
pip install -U pip wheel setuptools
pip install sympy==0.7.1
```

ags-dy, 2021-05-06:
When I add that line to the script, it does not solve the problem:

```bash
Building docker image
Sending build context to Docker daemon  25.37MB
Step 1/4 : FROM personalrobotics/ros-openrave
 ---> 3470beb7a703
Step 2/4 : RUN apt-key adv --keyserver 'hkp://keyserver.ubuntu.com:80' --recv-key C1CF6E31E6BADE8868B172B4F42ED6FBAB17C654 &&     apt-key del 421C365BD9FF1F717815A3895523BAEEB01FA116 &&     apt-get update &&     apt-get install -y --no-install-recommends python-pip build-essential liblapack-dev ros-indigo-collada-urdf &&     apt-get clean && rm -rf /var/lib/apt/lists/*
 ---> Using cache
 ---> 691f6b13328c
Step 3/4 : RUN pip install -U pip wheel setuptools
 ---> Running in 03388423e747
Cannot fetch index base URL https://pypi.python.org/simple/
Could not find any downloads that satisfy the requirement pip in /usr/lib/python2.7/dist-packages
Downloading/unpacking pip
Cleaning up...
No distributions at all found for pip in /usr/lib/python2.7/dist-packages
Storing debug log for failure in /root/.pip/pip.log
The command '/bin/sh -c pip install -U pip wheel setuptools' returned a non-zero code: 1
```

gavanderhoorn, 2021-05-06:
My comment was not intended to solve your problem. I just wanted to highlight the fact that it's more likely the outdated `pip`, not the fact that Sympy is no longer available on Pypi or similar.

ags-dy, 2021-05-06:
By the problem, I meant that it appears that pip is not able to install packages from pypi (and that included updating pip itself)

This also fails:
```bash
run python -m pip install –upgrade pip
```
In term of fixing the problem, I have solved it sufficiently for my use case.  Just including here in case anyone else has the problem.

gavanderhoorn, 2021-05-06:
As a work-around for the outdated `pip` in the `personalrobotics/openrave` image this is probably ok.

Otherwise the image would have to be recreated, which will be much more work.

closed 2021-05-06 by rhaschke