moveit/moveit #2653: Failing CI unit test: pilz_industrial_motion_planner/test/integrationtest_get_solver_tip_frame.test

rhaschke, 2021-05-06:
https://github.com/ubi-agni/moveit/runs/2521124797
```
Full test results for 'build/pilz_industrial_motion_planner/test_results/pilz_industrial_motion_planner/rostest-test_integrationtest_get_solver_tip_frame.xml'
-------------------------------------------------
<?xml version="1.0" encoding="utf-8"?>
<testsuite errors="0" failures="1" name="unittest.suite.TestSuite" tests="1" time="1.418"><testcase classname="rostest.runner.RosTest" name="testintegrationtest_get_solver_tip_frame" time="1.4178"><failure type="AssertionError">test [integrationtest_get_solver_tip_frame] did not generate test results
  File "/usr/lib/python2.7/unittest/case.py", line 329, in run
    testMethod()
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rostest/runner.py", line 164, in fn
    self.assert_(os.path.isfile(test_file), "test [%s] did not generate test results"%test_name)
  File "/usr/lib/python2.7/unittest/case.py", line 422, in assertTrue
    raise self.failureException(msg)
</failure></testcase><system-out>&lt;![CDATA[
[ROSTEST]setup[/home/runner/work/moveit/moveit/.work/target_ws/src/moveit/moveit_planners/pilz_industrial_motion_planner/test/integrationtest_get_solver_tip_frame.test] run_id[49b9cdea-ae9e-11eb-bc73-0242ac110002] starting
&amp;#x001B;[1mstarted roslaunch server http://9cd72f10c291:39871/&amp;#x001B;[0m

SUMMARY
========

PARAMETERS
 * /robot_description: &lt;?xml version="1....
 * /robot_description_kinematics/manipulator/kinematics_solver: prbt_manipulator/...
 * /robot_description_kinematics/manipulator/kinematics_solver_search_resolution: 0.005
 * /robot_description_kinematics/manipulator/kinematics_solver_timeout: 0.005
 * /robot_description_kinematics/manipulator/link_prefix: prbt_
 * /robot_description_planning/cartesian_limits/max_rot_vel: 1.57
 * /robot_description_planning/cartesian_limits/max_trans_acc: 2.25
 * /robot_description_planning/cartesian_limits/max_trans_dec: -5
 * /robot_description_planning/cartesian_limits/max_trans_vel: 1
 * /robot_description_planning/joint_limits/prbt_gripper_finger_left_joint/has_acceleration_limits: True
 * /robot_description_planning/joint_limits/prbt_gripper_finger_left_joint/has_velocity_limits: True
 * /robot_description_planning/joint_limits/prbt_gripper_finger_left_joint/max_acceleration: 0.328
 * /robot_description_planning/joint_limits/prbt_gripper_finger_left_joint/max_velocity: 0.082
 * /robot_description_planning/joint_limits/prbt_joint_1/has_acceleration_limits: True
 * /robot_description_planning/joint_limits/prbt_joint_1/has_velocity_limits: True
 * /robot_description_planning/joint_limits/prbt_joint_1/max_acceleration: 3.49
 * /robot_description_planning/joint_limits/prbt_joint_1/max_velocity: 1.57
 * /robot_description_planning/joint_limits/prbt_joint_2/has_acceleration_limits: True
 * /robot_description_planning/joint_limits/prbt_joint_2/has_velocity_limits: True
 * /robot_description_planning/joint_limits/prbt_joint_2/max_acceleration: 3.49
 * /robot_description_planning/joint_limits/prbt_joint_2/max_velocity: 1.57
 * /robot_description_planning/joint_limits/prbt_joint_3/has_acceleration_limits: True
 * /robot_description_planning/joint_limits/prbt_joint_3/has_velocity_limits: True
 * /robot_description_planning/joint_limits/prbt_joint_3/max_acceleration: 3.49
 * /robot_description_planning/joint_limits/prbt_joint_3/max_velocity: 1.57
 * /robot_description_planning/joint_limits/prbt_joint_4/has_acceleration_limits: True
 * /robot_description_planning/joint_limits/prbt_joint_4/has_velocity_limits: True
 * /robot_description_planning/joint_limits/prbt_joint_4/max_acceleration: 3.49
 * /robot_description_planning/joint_limits/prbt_joint_4/max_velocity: 1.57
 * /robot_description_planning/joint_limits/prbt_joint_5/has_acceleration_limits: True
 * /robot_description_planning/joint_limits/prbt_joint_5/has_velocity_limits: True
 * /robot_description_planning/joint_limits/prbt_joint_5/max_acceleration: 3.49
 * /robot_description_planning/joint_limits/prbt_joint_5/max_velocity: 1.57
 * /robot_description_planning/joint_limits/prbt_joint_6/has_acceleration_limits: True
 * /robot_description_planning/joint_limits/prbt_joint_6/has_velocity_limits: True
 * /robot_description_planning/joint_limits/prbt_joint_6/max_acceleration: 3.49
 * /robot_description_planning/joint_limits/prbt_joint_6/max_velocity: 1.57
 * /robot_description_semantic: &lt;?xml version="1....
 * /rosdistro: melodic
 * /rosversion: 1.14.11

NODES

[ROSTEST]setup[/home/runner/work/moveit/moveit/.work/target_ws/src/moveit/moveit_planners/pilz_industrial_motion_planner/test/integrationtest_get_solver_tip_frame.test] run_id[49b9cdea-ae9e-11eb-bc73-0242ac110002] done
[ROSTEST]Running test [integrationtest_get_solver_tip_frame]
&amp;#x001B;[1mROS_MASTER_URI=http://9cd72f10c291:45219/&amp;#x001B;[0m
&amp;#x001B;[1msetting /run_id to 49b9cdea-ae9e-11eb-bc73-0242ac110002&amp;#x001B;[0m
[ROSTEST]running test integrationtest_get_solver_tip_frame
&amp;#x001B;[1mprocess[integrationtest_get_solver_tip_frame-1]: started with pid [34033]&amp;#x001B;[0m
[ROSTEST]test [integrationtest_get_solver_tip_frame] finished
[ROSTEST]tearDown[/home/runner/work/moveit/moveit/.work/target_ws/src/moveit/moveit_planners/pilz_industrial_motion_planner/test/integrationtest_get_solver_tip_frame.test]
shutting down processing monitor...
... shutting down processing monitor complete
[ROSTEST]rostest teardown /home/runner/work/moveit/moveit/.work/target_ws/src/moveit/moveit_planners/pilz_industrial_motion_planner/test/integrationtest_get_solver_tip_frame.test complete

]]&gt;</system-out><system-err>&lt;![CDATA[
&amp;#x001B;[31m[integrationtest_get_solver_tip_frame-1] process has died [pid 34033, exit code -11, cmd /home/runner/work/moveit/moveit/.work/target_ws/devel/.private/pilz_industrial_motion_planner/lib/pilz_industrial_motion_planner/integrationtest_get_solver_tip_frame --gtest_output=xml:/home/runner/work/moveit/moveit/.work/target_ws/build/pilz_industrial_motion_planner/test_results/pilz_industrial_motion_planner/rosunit-integrationtest_get_solver_tip_frame.xml __name:=integrationtest_get_solver_tip_frame __log:=/root/.ros/log/49b9cdea-ae9e-11eb-bc73-0242ac110002/integrationtest_get_solver_tip_frame-1.log].
log file: /root/.ros/log/49b9cdea-ae9e-11eb-bc73-0242ac110002/integrationtest_get_solver_tip_frame-1*.log&amp;#x001B;[0m

]]&gt;</system-err></testsuite>
```

rhaschke, 2021-05-07:
I found the culprit: the planning plugin links against old libraries in /opt/ros instead of using the versions from the workspace it was compiled for:
```bash
ldd /root/upstream_ws/install/lib/libprbt_manipulator_moveit_ikfast_plugin.so | grep libmoveit
	libmoveit_kinematics_base.so.1.0.7 => /opt/ros/melodic/lib/libmoveit_kinematics_base.so.1.0.7 (0x00007f8e0c040000)
	libmoveit_robot_model.so.1.0.7 => /opt/ros/melodic/lib/libmoveit_robot_model.so.1.0.7 (0x00007f8e0bde2000)
	libmoveit_robot_state.so.1.0.7 => /opt/ros/melodic/lib/libmoveit_robot_state.so.1.0.7 (0x00007f8e0bb7a000)
	libmoveit_profiler.so.1.0.7 => /opt/ros/melodic/lib/libmoveit_profiler.so.1.0.7 (0x00007f8e0996f000)
	libmoveit_exceptions.so.1.0.7 => /opt/ros/melodic/lib/libmoveit_exceptions.so.1.0.7 (0x00007f8e0976b000)
	libmoveit_transforms.so.1.0.7 => /opt/ros/melodic/lib/libmoveit_transforms.so.1.0.7 (0x00007f8e09112000)
```
We have a classical chicken-egg-problem here: While moveit requires moveit_resources as a (test) dependency, moveit_resources requires moveit as a (build) depencency! In my point of view, moveit_resources shouldn't include ikfast solvers, which pull in the dependency to moveit_core!
In the past, this worked for us because both, moveit and moveit_resources were built in the same workspace. However, the build farm and the prerelease test should have issues with this config as well (the former times out after 2h)...
@jschleicher @tylerjw

tylerjw, 2021-05-08:
How should we fix this, could moveit_config be modified in someway to not depend on moveit_core?  Where should the ikfast solvers be?

rhaschke, 2021-05-09:
As we need moveit_resources for testing in moveit, moveit_resources should be independent of moveit.
Thus, the ikfast solver should be replaced with another solver.

jschleicher, 2021-05-12:
We had concerns, that KDL would give unreproducible IK solutions resulting in flaky tests, especially for LINear movements. You'd have to just try it out.

rhaschke, 2021-05-23:
I ran into this issue today again when trying to [release for Melodic](https://github.com/ros-planning/moveit/runs/2650936472)!
@jschleicher, please prepare a fixing PR asap. `moveit_resources` must not depend on any `moveit` package!
Otherwise, the build farm will run into circular dependency issues as well.

rhaschke, 2021-05-24:
Unfortunately, the KDL solver cannot find solutions for many of the test cases (see [here](https://github.com/ros-planning/moveit/runs/2652505168?check_suite_focus=true)).
To be honest, the cleanest solution would be to split off the pilz planner from the main MoveIt repo. That's the only way to cleanly resolve the circular dependency. I think the released binary package `ros-noetic-moveit-resources-prbt-ikfast-manipulator-plugin` is broken as well, because it refers to outdated libraries:

```
$ ldd  libprbt_manipulator_moveit_ikfast_plugin.so
	linux-vdso.so.1 (0x00007ffcea3f4000)
	libmoveit_kinematics_base.so.1.1.4 => not found
	libmoveit_robot_model.so.1.1.4 => not found
	libmoveit_robot_state.so.1.1.4 => not found
```

v4hn, 2021-05-24:
> We have a classical chicken-egg-problem here: While moveit requires moveit_resources as a (test) dependency, moveit_resources requires moveit as a (build) depencency! In my point of view, moveit_resources shouldn't include ikfast solvers, which pull in the dependency to moveit_core!

To be honest I'm not happy with the state of test-dependencies anyway. It's tedious and error-prone if one can't assume all of standard moveit/moveit_resources to be available at test time in moveit tests.
Isn't it a fault in build handling if we can't have the tests in package `<a>` depend on compiled code in package `<b>` that depends on `<a>`? I would argue this is a needed feature and exactly what we do in MoveIt already.
I'm not sure whether, how, or where this is actually broken, but maybe we can fix the issue there somehow?
Alternatively we could make the corresponding tests optional and not specify the dependency in the `package.xml` to hide them for release issues. I guess in our CI we could still force all resource-based tests via a cmake flag, e.g., via `-DMOVEIT_RESOURCES_TESTS=ON`. Surely quite ugly and only an option if we don't have alternatives.

If we desperately want to get rid of the compile-time dependency we could add the prbt solver to `moveit_kinematics`, though I'm not happy with committing auto-generated code there. Ikfast is somewhat of a special case though.

On a package level the same issue exists in moveit packages already though! [`moveit_core` test-depends on `moveit_resources_panda_moveit_config`](https://github.com/ros-planning/moveit/blob/master/moveit_core/package.xml#L57) because we want to be able to construct a `RobotModel` from it. `moveit_resources_panda_moveit_config` specifies KDL (from `moveit_kinematics`) as its IK solver. The corresponding entry in the `package.xml` [is disabled](https://github.com/ros-planning/moveit_resources/blob/master/panda_moveit_config/package.xml#L24-L30) to ignore the circular dependency. The current tests in `moveit_core` actually setup a `RobotModel` with incomplete functionality because we never instantiate the IK solver to skirt the issue.

All of that aside I don't think it's actually chicken-and-egg in the particular case of the Pilz code. It only sounds like one because you refer to the meta-package `moveit_resources`. [`moveit_core` does not even test-depend on the `moveit_resources_prbt_*` packages](https://github.com/ros-planning/moveit/blob/master/moveit_core/package.xml#L57-L58), so they are (on a package level) free to link to `moveit_core`.

rhaschke, 2021-05-24:
It `is` a chicken-and-egg issue, because `moveit_resources` - in order to build - requires moveit_core and moveit_kinematics to be present. As `moveit` and `moveit_resources` are independent repositories which are built independently on the build farm, `moveit_resources` will be built based on an old version of `moveit`. Only later, `moveit` will be built, which uses the generated IKFast solver (linking against an _old_ `moveit_core` in `/opt/ros`) together with a _new_ `moveit_core`, causing the failure (in situations where ABI / API has changed).

rhaschke, 2021-05-24:
> It's tedious and error-prone if one can't assume all of standard moveit / ~~moveit_resources~~ to be available at test time in moveit tests.

Just to clarify: Usually, you can rely on all of moveit to be available for tests, because tests are built/ran after the libs.

v4hn, 2021-05-25:
You are talking in terms of git repositories and not packages.
Looking only at `package.xml` files I see no issue (which is why it works if `moveit_core` and `moveit_resources` are part of the same workspace).

The buildfarm (according to my meager knowledge of it) builds each package - independent of the upstream repository - in isolation and only considers the `package.xml` entries.
Now the problem you describe is that the buildfarm cannot handle the case I outlined above, namely it expects all tests of `<a>` to be able to run without code from another repository with package `<b>` that has to be build based on `<a>`.
If `<b>` clearly specifies a compile-time dependency on `<a>` and the buildfarm does not rebuild `<b>` before running `<a>`'s tests (which explicitly require `<b>`), then the buildfarm is broken in that regard and should either rebuild `<b>` before testing or deny the possibility altogether stating they don't do that. I would expect they do not break because it works for some cases where developers carefully split things up. :/

For our CI setup I also don't see a problem with forcing both packages to be present in `target_ws` if we want to be able to run tests.

One other alternative to moving the plugin to `moveit_kinematics` the build-farm might be able to handle would be to turn `moveit_resources` into a submodule of the moveit repository.
Only the build farm and users who want to run tests would have to initialize submodules. Just putting out some ideas on how this might be resolved...

jschleicher, 2021-05-25:
Another possibility would be to start ABI compatibility on the `kinematics_base` library. Imho one large weakness of the ROS ecosystem is to neglect ABI-compatibility at the first hand and rebuild everything, if one part changes. The tools within Ubuntu / Debian allow version ranges and imho bloom and the ROS buildfarm should make use of it to some extent... What do you think?

rhaschke, 2021-05-25:
> You are talking in terms of git repositories and not packages. Looking only at `package.xml` files I see no issue (which is why it works if `moveit_core` and `moveit_resources` are part of the same workspace). 

Exactly.

> The build farm builds each package - independent of the upstream repository - in isolation and only considers the `package.xml` entries. 

You are right: to build binary releases, packages are built one by one - independently of the repo structure. Actually I wasn't aware of that anymore. However, the dev builds as well as the pre-release tests are performed _per repo_ and those fail:
* https://build.ros.org/job/Mdev__moveit__ubuntu_bionic_amd64
* https://github.com/ros-planning/moveit/runs/2652226284?check_suite_focus=true

So, I am referring to per-repo builds.

> For our CI setup I also don't see a problem with forcing both packages to be present in `target_ws`.

That's the workaround I used for now.

> Another possibility would be to compy to  ABI compatibility and make use of corresponding tools in the Ubuntu tool chain.

IMHO this is only a theoretical solution. We didn't manage ABI/API compatibility in the past for several reasons (on purpose, by accident, ...). It is hopeless to require this for all the thousands of ROS packages out there.
By the way, to circumvent the ABI compatibility issue we introduced SONAMES in the past...

rhaschke, 2021-05-27:
We discussed this issue in today's maintainer meeting. The only clean solution, we came up with, would be to factor out the Pilz planner (and corresponding moveit_resources) into its own repo. This will have a few downsides:
* It will become slightly harder to maintain consistency across both repositories
  However, including the pilz planner into the downstream workspace for MoveIt's GHA, we can mitigate this.
* We need an extra release step for this separate repo - but only if there are actual changes.

What do you think @jschleicher, @ct2034?

v4hn, 2021-05-27:
Now I feel even worse for skipping the meeting for reasons of private life.
The compatible time slots between Europe and America are far from compatible with me at the moment...

Why would such a split be any cleaner than maintaining the plugin in moveit_kinematics?
The PRBT moveit config is part of moveit_resources and thus an officially released with MoveIt.
I would argue that's perfectly enough reason to include a small ik plugin for it in the main repository (instead of splintering off parts again).

If we start discussing splitting parts out, splitting off `moveit_core` would solve the problem as well, but we would not discuss this.
It makes a lot of sense to include the test coverage of the pilz planner repository in the main repository.

Sorry for disputing further.

rhaschke, 2021-05-27:
We shortly discussed moving the IK plugin into moveit_kinematics. However, this will tear apart the PRBT config even more.
As far as I know the Pilz robot is only used for testing the Pilz planner...
By the way, we won't loose any test coverage splitting off the Pilz planner. We would still keep it in the downstream workspace.

v4hn, 2021-05-27:
>  As far as I know the Pilz robot is only used for testing the Pilz planner...

It's a released package in moveit_resources and everyone is free to use these files to setup MoveIt examples/tests.
If we would not want that, we should not have a released moveit_resources package at all and only use it for internal testing.
For example I could well-imagine the new GSoC project to use this robot (as well as the others in our resources) to benchmark MoveIt's performance in different scenes.

> We would still keep it in the downstream workspace.

That's a good point, true. Though of course users would have to add a third repository to their workspaces to get these tests.
So the main point against splitting off *any* part of the repository is that it would fracture source locations again after a lot of effort to consolidate them. As I believe that effort was well-justified, I still prefer any other way.

> However, this will tear apart the PRBT config even more

I would not say that is a bad thing. After all you argued against source code in moveit_resources.
To clarify: Do you also want to remove the configs from `moveit_resources` again, or are you only talking about splitting planner+tests+ikplugin out because the tests need to build the ik plugin?
Without the potentially new repository in the workspace the config in `moveit_resources` would be broken for IK requests.

rhaschke, 2021-05-27:
I didn't argue against source code in moveit_resources in general, but against a circular dependency introduced by that code.
For your clarification: I suggest moving all prbt packages from moveit_resources and moveit_planners into their own repository. All these packages belong together. I agree that just splitting off the ikplugin doesn't make sense, because that would leave moveit_resources in an incomplete state.

v4hn, 2021-05-28:
> For your clarification: I suggest moving all prbt packages from moveit_resources and moveit_planners into their own repository. 

I do not believe this is a reasonable way to go when the "only" problem is a dependency because of tests.
I will work on a pull-request to merge the IK plugin into `moveit_kinematics` instead as soon as I can.

It would be even better if we were to combine this with [our nice ikfast-generator tests](https://github.com/ros-planning/moveit/blob/master/moveit_kinematics/test/test_ikfast_plugins.sh), but we should also have the package around if users without the appropriate ubuntu/docker setup want to run the tests.

rhaschke, 2021-05-28:
It's not only a test problem: The issue here is actually the root cause for #2688/#2689, which you filed yourself.

> I will work on a pull-request to merge the IK plugin into moveit_kinematics instead as soon as I can.

Sorry, but moving the very _specific_ and _auto-generated_ IK plugin into `moveit_kinematics`, which is a _generic_ package providing _abstract plugin interfaces_, isn't an acceptable solution to me. You shouldn't spend time on this before we haven't agreed on the way to go.

v4hn, 2021-05-28:
> Sorry, but moving the very specific and auto-generated IK plugin into moveit_kinematics, which is a generic package providing abstract plugin interfaces, isn't an acceptable solution to me.

It would be better to auto-generate the plugin each time. But I will not force users to do that and ikfast plugin generation is somewhat special in terms of required software dependencies. So we have to provide the generated sources as well.
All alternatives to merging it into this repository will result in the same dependency issues, so I see no alternative to merging it here.
I created two initial pull-requests to implement this.

Splitting off an entire planner that is meant to provide **fundamental** support for many potential users because the tests require a downstream package to be compiled does not make any sense to me and creates unnecessary overhead. I believe both our positions here are clear.

The only alternative I see right now is to add a new folder/package like `moveit_test_utils` to the repository root and move the plugin there. We could also move [the `robot_model_test_utils`](https://github.com/ros-planning/moveit/blob/master/moveit_core/utils/include/moveit/utils/robot_model_test_utils.h) there in that case.

> The issue here is actually the root cause for #2688/#2689, which you filed yourself.

Just to mention it, [there were some other useless offenders in moveit_resources prbt_support](https://github.com/ros-planning/moveit_resources/pull/79/commits/5fb2f2c72ab3e14f8458967008dd0b26384b3560#diff-63e10989baf60da8301988bc5fccb0972e51591866b80a0a955dc722c3f4b418L34-L35) as well that I also removed in the PR.

rhaschke, 2021-06-07:
I had a discussion with @jschleicher and @ct2034. They said that Pilz will not produce their robot anymore and they had to reduce efforts spent on MoveIt - unfortunately. Given that (new) situation, we agreed that the best solution would be to replace the Pilz robot with another robot, e.g. Panda for the unit tests of the Pilz planner. 
Their test infrastructure, namely utilities to load named via points, will be valuable for MoveIt in general and should be integrated into MoveIt as part of `moveit_test_utils`. (@jschleicher: could you please add a package reference? I wasn't able to find it immediately.)
However, this will be some tedious work requiring a volunteer to jump on. We agreed to discuss the topic in the next maintainer meeting in a larger round again.

v4hn, 2021-06-07:
Ouch. Under these circumstances I very much agree though it's yet another big open workpackage.

ct2034, 2021-06-07:
Thanks for the nice discussion, @rhaschke. Indeed, we are going forward with our PRBT. I think therefore it would also make sense to not use it in test anymore. 
Our test infrastructure, you are mentioning can be found here: https://github.com/ros-planning/moveit/tree/master/moveit_planners/pilz_industrial_motion_planner_testutils

v4hn, 2021-06-08:
@ct2034 wrote
> Indeed, we are going forward with our PRBT

I believe you meant to say *not*? Otherwise I'm confused now...

ct2034, 2021-06-08:
I fixed it. Sorry for the confusion

v4hn, 2021-06-26:
During the last m3 (MoveIt Maintainer Meeting? :smiley:), we briefly discussed an important aspect in this overall debate: We don't provide another robot configuration in `moveit_resources` that uses an analytical IK solver.
However, in order to get deterministic results with `LIN` planner tests in pilz (which we definitely want to keep imo!) we actually need such a configuration around. All planners in `moveit_kinematics` might fail nondeterministically.

But this basically brings us back to https://github.com/ros-planning/moveit/pull/2697, because the corresponding plugin needs to be compiled in the MoveIt code-base.
I agree that the generated IKfast plugin is not the best choice for this, though it's the simplest to work around the current issues.
I'm still open to move the plugin into a new package `moveit_testing_utils` that is not pulled-in through the moveit meta package to reduce user annoyance. Though of course the corresponding moveit_resources configuration would fail without the plugin available, so we probably want to pull it in for everyone.

An alternative that might be neater would be to integrate a robot with ortho-parallel-base-spherical-wrist configuration, pull in https://github.com/Jmeyer1292/opw_kinematics as a dependency to moveit_kinematics and integrate [@JeroenDM's plugin](https://github.com/JeroenDM/moveit_opw_kinematics_plugin) upstream. 
I expect @JeroenDM would be happy to see it there :cat2: 
That approach would give us a "general-purpose" analytical solver in moveit which could be used in one of the resources configurations.
It's also work someone has to do to migrate the tests though and we need to decide on yet another robot to import into the resources (Personally I believe the more the merrier as long as each config adds a unique aspect). 

What do you think @rhaschke ?

gavanderhoorn, 2021-06-27:
> An alternative that might be neater would be to integrate a robot with ortho-parallel-base-spherical-wrist configuration

we already have one: the M-10iA copied from `ros-industrial/fanuc`.

Upstream there already has [the necessary parameters](https://github.com/ros-industrial/fanuc/blob/1fe02c8a3e6f00f31861e037e66cf4159586f0e8/fanuc_m10ia_support/config/opw_parameters_m10ia.yaml) to use `opw_kinematics`.

rhaschke, 2021-06-27:
I like this idea to pull an analytical solver into the tests.

v4hn, 2021-06-27:
> we already have one: the M-10iA copied from `ros-industrial/fanuc`.

In that case, I obviously made a fool of myself.
I even looked at the robot model before, but not at the wrist rotation axes because the associated meshes are asymetric.
Assuming we leave the panda with a KDL solver, I see no reason not to migrate our fanuc config to opw.

> Upstream there already has [the necessary parameters](https://github.com/ros-industrial/fanuc/blob/1fe02c8a3e6f00f31861e037e66cf4159586f0e8/fanuc_m10ia_support/config/opw_parameters_m10ia.yaml) to use `opw_kinematics`.

That's pretty much ideal then!
Let's see whether @JeroenDM likes the idea.

JeroenDM, 2021-06-29:
>  and integrate @JeroenDM's plugin upstream.

I'm not what this means exactly.

If this means I need to finish the noetic release I will do my best, but I'm having difficulty finding time to work on it at the moment (as @gavanderhoorn probably has noticed...).

But I certainly am a big fan of using a deterministic IK solver in the tests. And for the supported robots, opw_kinematics is easier much to use than ikfast I think.

v4hn, 2021-06-29:
> I'm not what this means exactly.

It means that you would merge the source code of your repository into the main MoveIt repository
and the MoveIt maintainer doing the releases would release it together with the rest. :-)

JeroenDM, 2021-06-29:
Ah, in that case, I like it even more :)

v4hn, 2021-06-30:
@rhaschke I'm somewhat confused right now. I believe you mentioned multiple times that the prerelease tests are currently broken because of this dependency issue. But I started a GHA prerelease test for this repo a few days ago (because apparently nobody did that before...) and [it succeeded](https://github.com/ros-planning/moveit/actions/runs/974958370).

Is there an issue with the GHA or is the discussed issue magically gone?

rhaschke, 2021-06-30:
There is only an issue if we changed ABI between MoveIt releases.

v4hn, 2021-06-30:
True, because the plugin in `moveit_resources` linked the system installation of MoveIt and will still work with the new release... :facepalm: 
The issue is *still there* because `moveit_resources` pulls in the whole MoveIt stack for building the package when it shouldn't, but because there is no explicit test for that, CI will still succeed.

The discussion has been going on for too long, sorry to waste your time.

rhaschke, 2021-11-05:
Closing. Replaced by #2947.

closed 2021-11-05 by rhaschke