moveit/moveit #2655: IKfast failure due to importing scipy.optimize.leastsq

ndehio, 2021-05-07:
### Description
Hi, I am having trouble with applying the IKfast feature. The failure is related to importing `scipy.optimize.leastsq`. I created a fresh virtual machine ubuntu20 to be on the safe side.

### Your environment
* ROS Distro: [Noetic]
* OS Version: Ubuntu 20.04 Virtual Machine
* [commit version](https://github.com/ros-planning/moveit/commit/6e00ff16654d46b11d1e9b88b339b4266959394a) (-> contains the latest merge related to [this issue](https://github.com/ros-planning/moveit/issues/2649))

### Steps to reproduce
```
sudo apt-get update 
sudo apt install git
sudo apt-get install docker.io
sudo service docker start
sudo usermod -a -G docker $USER
#=> REBOOT

#following http://wiki.ros.org/noetic/Installation/Ubuntu
sudo sh -c 'echo "deb http://packages.ros.org/ros/ubuntu $(lsb_release -sc) main" > /etc/apt/sources.list.d/ros-latest.list'
sudo apt-key adv --keyserver 'hkp://keyserver.ubuntu.com:80' --recv-key C1CF6E31E6BADE8868B172B4F42ED6FBAB17C654
sudo apt update
sudo apt install ros-noetic-desktop-full
sudo apt-get install ros-noetic-franka-description
source /opt/ros/noetic/setup.bash
sudo apt install python3-rosdep python3-rosinstall python3-rosinstall-generator python3-wstool build-essential
sudo rosdep init
rosdep update

git clone https://github.com/ros-planning/moveit.git
cd moveit/moveit_kinematics/ikfast_kinematics_plugin/scripts
# copy the `panda_default.urdf` file here
export MYROBOT_NAME="panda_default"
export PLANNING_GROUP="panda_default"
export BASE_LINK="0"
export EEF_LINK="10"
export FREE_INDEX="1"
./auto_create_ikfast_moveit_plugin.sh --iktype Transform6D --name $MYROBOT_NAME $MYROBOT_NAME.urdf $PLANNING_GROUP $BASE_LINK $EEF_LINK
```

### Expected behaviour
The ikfast should generate a cpp file.

### Actual behaviour
The script `auto_create_ikfast_moveit_plugin.sh` fails with the message:
`could not import scipy.optimize.leastsq`

### Backtrace or Console output
```
ubuntu@ubuntu2004:~/moveit/moveit_kinematics/ikfast_kinematics_plugin/scripts$ ./auto_create_ikfast_moveit_plugin.sh --iktype Transform6D --name $MYROBOT_NAME $MYROBOT_NAME.urdf $PLANNING_GROUP $BASE_LINK $EEF_LINK
Converting urdf to Collada

Document successfully written to /tmp/ikfast.vGP6zF/robot.dae
Building docker image
Sending build context to Docker daemon  25.38MB
Step 1/3 : FROM personalrobotics/ros-openrave
 ---> 3470beb7a703
Step 2/3 : RUN apt-key adv --keyserver 'hkp://keyserver.ubuntu.com:80' --recv-key C1CF6E31E6BADE8868B172B4F42ED6FBAB17C654 &&     apt-key del 421C365BD9FF1F717815A3895523BAEEB01FA116 &&     apt-get update &&     apt-get install -y --no-install-recommends python-pip build-essential liblapack-dev ros-indigo-collada-urdf &&     apt-get clean && rm -rf /var/lib/apt/lists/*
 ---> Using cache
 ---> e9148302257a
Step 3/3 : RUN pip install git+https://github.com/sympy/sympy.git@sympy-0.7.1
 ---> Using cache
 ---> 994592d60251
Successfully built 994592d60251
Successfully tagged fixed-openrave:latest
Successfully built docker image.
Running openrave0.9.py --database inversekinematics --robot=/input/wrapper.xml --iktype=Transform6D --iktests=1000
could not import scipy.optimize.leastsq
2021-05-07 08:02:23,677 openrave [WARN] [plugindatabase.h:577 InterfaceBasePtr OpenRAVE::RaveDatabase::Create] Failed to create name fcl_, interface collisionchecker
2021-05-07 08:02:25,611 openrave [WARN] [xmlreaders-core.cpp:2504 ManipulatorXMLReader::endElement] Failed to find manipulator base 0
2021-05-07 08:02:25,621 openrave [WARN] [xmlreaders-core.cpp:2497 ManipulatorXMLReader::endElement] Failed to find manipulator end effector 10
2021-05-07 08:02:25,622 openrave [WARN] [kinbody.cpp:1515 KinBody::SetDOFValues] dof 3 value is not in limits 0.000000e+00<-6.980000e-02
2021-05-07 08:02:25,622 openrave [WARN] [robotmanipulator.cpp:1270 RobotBase::Manipulator::_ComputeInternalInformation] manipulator panda_default has undefined base and end effector links 0, 10
2021-05-07 08:02:25,624 openrave [WARN] [environment-core.h:485 Load] load failed on file /input/wrapper.xml
2021-05-07 08:02:25,625 openrave [WARN] [environment-core.h:2421 _SimulationThread] timeout of UpdatePublishedBodies
Traceback (most recent call last):
  File "/usr/bin/openrave0.9.py", line 126, in <module>
    database.run(args=args)
  File "/usr/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/databases/inversekinematics.py", line 1123, in run
    InverseKinematicsModel.RunFromParser(*args,**kwargs)
  File "/usr/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/databases/inversekinematics.py", line 1088, in RunFromParser
    model = DatabaseGenerator.RunFromParser(Model=Model,parser=parser,robotatts=robotatts,args=args,**kwargs)
  File "/usr/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/databases/__init__.py", line 238, in RunFromParser
    options,model = DatabaseGenerator.InitializeFromParser(Model,parser,orgenv,args,robotatts,defaultviewer,allowkinbody)
  File "/usr/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/databases/__init__.py", line 219, in InitializeFromParser
    model = Model(robot=robot)
  File "/usr/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/databases/inversekinematics.py", line 1084, in <lambda>
    Model = lambda robot: InverseKinematicsModel(robot=robot,iktype=iktype,forceikfast=True)
  File "/usr/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/databases/inversekinematics.py", line 238, in __init__
    for link in robot.GetChain(manip.GetBase().GetIndex(),manip.GetEndEffector().GetIndex(),returnjoints=False)[1:]:
AttributeError: 'NoneType' object has no attribute 'GetIndex'
```

Thank you

rhaschke, 2021-05-07:
Hi Nils, this is a weird result as our [CI job successfully generates an ikfast solver](https://github.com/ros-planning/moveit/runs/2520608074), without complaining about missing `scipy.optimize.leastsq`. Actually, I suspect that the real culprit is this:
`manipulator panda_default has undefined base and end effector links 0, 10`, which later results in accessing `None`.
I don't think your link names are 0 and 10. Rather specify the correct link _names_.

ndehio, 2021-05-07:
@rhaschke thanks for the advice, you were right! Maybe it makes sense to add a better error message or stop the script execution earlier in that case, otherwise you can close the issue.

rhaschke, 2021-05-07:
We cannot influence the error reporting of OpenRAVE. We are just using this external tool to generate an IK solver.
Thus closing the issue.

closed 2021-05-07 by rhaschke