moveit/moveit #2658: MotionSequenceAction concerns

felixvd, 2021-05-07:
While working on #2657, I noticed a few things that I was wondering about. Maybe @jschleicher, @ct2034 could shed some light.

1. As far as I understand, [this line in appendWithStrictTimeIncrease](https://github.com/ros-planning/moveit/blob/master/moveit_planners/pilz_industrial_motion_planner/src/plan_components_builder.cpp#L61) combines multiple trajectories into one. However, I don't understand the second condition in that if-block. If the start and end of the two trajectories differ, they are both added to the trajectory at exactly the same time, which is A) not physically feasible and B) a problem for the trajectory_controller, which complains about the two waypoints at the same point in time (waypoints not strictly increasing in time). What is the reasoning here? Am I misunderstanding something?

2. If the planning pipeline is not the `pilz_industrial_motion_planner`, the accelerations are generally not 0 at the end of the trajectory (for TOTG and IPTP at least). This was probably not an issue when this code was written because it's meant for the pilz planner, but the error occurs systematically for other planning pipelines. Should something be done about this?

Thanks!

martiniil, 2021-05-18:
About 1.
I assume that the trajectories are constructed in such a way that the second starts where the first ends. So if the waypoints differ, the one that is appended should have a non-zero duration_from_start and the problem you describe is not present. Actually, the lines below (for-loop) were introduced in order to circumvent exactly this problem.

martiniil, 2021-05-18:
About 2.
At first glance I think the acceleration could be neglected here. However we have the same check here: [trajectory_blender_transition_window.cpp#L164](https://github.com/ros-planning/moveit/blob/3361b2d1b6b2feabc2d3e93c75653f5a00e87fa4/moveit_planners/pilz_industrial_motion_planner/src/trajectory_blender_transition_window.cpp#L164) which is called via `PlanComponentsBuilder::blend()`
and I am not sure how this would affect the blending algorithm...

martiniil, 2021-05-18:
> About 1.
> I assume that the trajectories are constructed in such a way that the second starts where the first ends. So if the waypoints differ, the one that is appended should have a non-zero [...]

Of course, we cannot suspect that these assumptions hold for any planning pipeline. But I would remove the bug-label since the `PlanComponentsBuilder` was only intended for the pilz planner.