moveit/moveit #2666: Shadow hand pending for my command after several calls.

lianghongzhuo, 2021-05-14:
### Description

Overview of your issue here.

### Your environment
* ROS Distro: [Melodic]
* OS Version: Ubuntu 18.04
* Source or Binary build?
* If source, master

### Steps to reproduce
```
self.hand_commander = MoveGroupCommander("hand")
self.hand_commander.set_start_state_to_current_state()
self.hand_commander.set_joint_value_target(hand_joint_positions)
self.hand_commander.go(wait=True)
```

### Expected behaviour
Go to the target joint position

### Actual behaviour
Code stuck at `self.hand_commander.go(wait=True)`
### Backtrace or Console output
No error msg

My guess is this behaviour is related to https://github.com/ros-planning/moveit/pull/2663 which introduce a feature that `The joint states of `passive` joints must be published in ROS and the CurrentStateMonitor will now wait for them as well. ` If so, please provide a possible fix for a robot such as shadow hand.

lianghongzhuo, 2021-05-14:
Update,
after check out to v1.1.4 the bug still exists, so not related to https://github.com/ros-planning/moveit/pull/2663

 after check out to commit https://github.com/ros-planning/moveit/commit/cdda16fdee21f05d6a6209d59e08cc6086d9efb6
The bug disappears.
So I bet it related with https://github.com/ros-planning/moveit/pull/2596

lianghongzhuo, 2021-05-14:
to reproduce: https://github.com/ros-planning/moveit/pull/2596#issuecomment-841325475

simonschmeisser, 2021-05-18:
I'm testing what you posted in the PR and can indeed reproduce a dead lock. Now I attached gdb to the `move_group` binary and can see that two threads are doing something with a lock:

thread # 6 processes an new point cloud and wants a write lock on the octomap
```
1   futex_wait_cancelable                                                                                                                                                                                                                                                                                                                                                     futex-internal.h               88  0x7fec6eb7dad3 
2   __pthread_cond_wait_common                                                                                                                                                                                                                                                                                                                                                pthread_cond_wait.c            502 0x7fec6eb7dad3 
3   __pthread_cond_wait                                                                                                                                                                                                                                                                                                                                                       pthread_cond_wait.c            655 0x7fec6eb7dad3 
4   boost::condition_variable::wait                                                                                                                                                                                                                                                                                                                                           condition_variable.hpp         81  0x7fec6f022b7d 
5   boost::shared_mutex::lock                                                                                                                                                                                                                                                                                                                                                 shared_mutex.hpp               294 0x7fec6f022b7d 
6   collision_detection::OccMapTree::lockWrite                                                                                                                                                                                                                                                                                                                                occupancy_map.h                77  0x7fec506beb1c 
7   occupancy_map_monitor::PointCloudOctomapUpdater::cloudMsgCallback                                                                                                                                                                                                                                                                                                         pointcloud_octomap_updater.cpp 328 0x7fec506beb1c 
8   boost::function1<void, boost::shared_ptr<sensor_msgs::PointCloud2_<std::allocator<void>> const> const&>::operator()                                                                                                                                                                                                                                                       function_template.hpp          759 0x7fec506ca96e 
9   boost::detail::function::void_function_obj_invoker1<boost::function<void (boost::shared_ptr<sensor_msgs::PointCloud2_<std::allocator<void>> const> const&)>, void, boost::shared_ptr<sensor_msgs::PointCloud2_<std::allocator<void>> const>>::invoke(boost::detail::function::function_buffer&, boost::shared_ptr<sensor_msgs::PointCloud2_<std::allocator<void>> const>) function_template.hpp          159 0x7fec506ca96e 
10  boost::function1<void, boost::shared_ptr<sensor_msgs::PointCloud2_<std::allocator<void>> const>>::operator()                                                                                                                                                                                                                                                              function_template.hpp          759 0x7fec506d2990 
11  message_filters::CallbackHelper1T<boost::shared_ptr<sensor_msgs::PointCloud2_<std::allocator<void>> const> const&, sensor_msgs::PointCloud2_<std::allocator<void>>>::call                                                                                                                                                                                                 signal1.h                      76  0x7fec506d2990 
12  message_filters::Signal1<sensor_msgs::PointCloud2_<std::allocator<void>>>::call                                                                                                                                                                                                                                                                                           signal1.h                      119 0x7fec506cf1bf 
13  message_filters::SimpleFilter<sensor_msgs::PointCloud2_<std::allocator<void>>>::signalMessage                                                                                                                                     [... cut ...]
55 clone                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                 clone.S                              95  0x7fec6c9d771f 

```

I do not understand yet why they are both waiting and why there is a dead lock unfortunately.

closed 2021-05-23 by rhaschke

aditya2592, 2022-05-15:
Hi, we are seeing a similar bug in moveit2. Was #2683 copied to moveit2 or is not needed there?

PR #2683 (2021-05-23): Revert "Lock the octomap/octree while collision checking (#2596)"
This reverts commit 0935d681b0141108e9695db30107184996963be7.

It introduced a deadlock in the standard MoveGroup planning pipeline.
Octomap is already locked in `PlanningSceneMonitor::lockSceneRead()` [before planning in the capability](https://github.com/ros-planning/moveit/blob/master/moveit_ros/move_group/src/default_capabilities/move_action_capability.cpp#L222).
So the recursive locking calls (even in a different thread because OMPL spawns [a separate goal sampling thread](https://github.com/ros-planning/moveit/blob/master/ [... cut]